letscontrolit/ESPEasy #1622: Command LongPuls does not work correctly

khbraemer, 2018-07-31:
There is a problem with the command LongPulse on Espeasy release mega 20180708 normal esp8266_4096 and after. An easy test is to switch on the wifi led on for a defind time:
Command : LongPulse,02,0,5. The last correct running release is espeasy mega 20180625 normal esp8266_ 4096.bin

TD-er, 2018-07-31:
Are you sure no later version is working?
I fixed this about a week ago in this PR: https://github.com/letscontrolit/ESPEasy/pull/1594
I tested it with the ms version of that timer, but I see no reason why it should not work on the normal LongPulse command.

There is another PR of mine ready to be merged, but I also made a test build for it here: https://github.com/letscontrolit/ESPEasy/pull/1618#issuecomment-408997697
Can you test to see if it is still broken?

khbraemer, 2018-07-31:
Hello. i check the in Tools/ Command the Command LongPulse,02,0,5 on the Command line with the latest Release mega 20180723.
The command output shows:
{
"log": "GPIO 2 Pulse set for 5 se",
"plugin": 1,
"pin": 2,
"mode": "output",
"state": 0
}
The LED switched on but not off after 5 seconds. In the LOG i see the following Message:
SW : GPIO 2 Pulse set for 5 sec
but the LED does not switched off
I also check the LongPulse_ms,2,0,2000,. The same Problem, the LED switched not off after 2000ms.

Log Content:
SW : GPIO 2 Pulse set for 2000 msec

TD-er, 2018-07-31:
You have an extra ',' at the end of your quoted text.

Also make sure a switch plugin is active (no matter to what pin assigned) and set the switch as one of the first few task items in the task list (device tab in the web interface)

khbraemer, 2018-07-31:
hello, if i define an Input Port as switch with name test#test and a rule1 with following content:
on test#test=1 do
TimerSet,1,2
LongPulse,2,0,5
endon
if i set the port test#test to 1 then the longpulse works fine.
The log output shows this:
1842402: EVENT: Clock#Time=Tue,14:20
1844931: SW : GPIO 0 Set to 1
1845004: SW : Switch state 1 Output value 1
1845006: EVENT: test#test=1.00
1845012: ACT : TimerSet,1,2
1845021: Command: timerset
1845023: ACT : LongPulse,2,0,5
1845025: SW : GPIO 2 Pulse set for 5 sec
1847402: EVENT: Rules#Timer=1
My problem is that i use ESPEasy with Fhem.
The command to set a port for certain  time is:
define MyPort1_51_1_Roll_WZ_links notify RelayGPIO5_51WZ_Ost_auf:1 "wget -q -O - 'http://192.168.178.51/control?cmd=LongPulse,05,1,30'"
In older ESPEasy Releases this works  fine

LongPulse_ms,2,0,2000,.  This was an input error during typing this text.

v-a-d-e-r, 2018-07-31:
Also no problem here. LongPulse,2,0,5 -> LED turns on for 5 sec and then off. Checkt with current git version....

v-a-d-e-r, 2018-07-31:
@TD-er: Another thing: You have also a Eastron SDM energy meter? Can you run it with hardware serial? My ends in a boot loop! Software serial works fine. :-/

TD-er, 2018-07-31:
So the issue is with the command over http as a GET operation?
Any other command via http GET does work?

TD-er, 2018-07-31:
@ eastron meter. I have one including all needed things to connect it, but haven't gotten to it.
Maybe you can add an issue about it?
Is it already included in the mega branch?

TD-er, 2018-08-01:
@khbraemer Thanks for the update.
This is one of the main reasons why I want to strip this functionality from the P001_Switch plugin.
These functions (play a tune, operate on GPIO pins not assigned to the plugin, operate a servo, etc) do not belong to this plugin.

So (sadly) it is working (partly) as designed, but not as expected.
If it worked 'as designed', then the LED would not switch on initially when no switch plugin was defined and set enabled.

Grovkillen, 2019-01-02:
https://www.letscontrolit.com/forum/viewtopic.php?t=6225

We need to have this sorted for the frozen version.

thymjan, 2019-02-11:
ESP_Easy_mega-20190121_test_ESP8266_4096.bin

e.g.
`longpulse,4,1,5` doesn't work correct. GPIO 4 is set to high but is not reset to low again after 5 seconds.

but
`pulse,4,1,5` is working correctly. GPIO 4 is set to hight and reset to low after 5 miliseconds.

mneuron, 2019-02-21:
The longpulse and longpulse_ms functions are working - perhaps not as expected - when the 'switch input (p001)' plugin is enabled in the devicelist. It isn't necessary to choose a gpio in the switch requester, just set it to none.  
@TD-er it has probably to do with the 'setPluginTaskTimer' call which is ignored when the plugin is not enabled. The other plugins p009_mcp, p019_pcf8574 and other i2c extenders probably have the same behaviour.

alwashe, 2019-08-11:
Same here, I tried:
ESP_Easy_mega-20190809_normal_ESP8266_4M.bin
and
ESP_Easy_mega-20190630_normal_ESP8266_4M.bin

```LongPulse,2,0,5``` and ```LongPulse_mS,2,0,5``` are not working. The LED turns on and stays on.
```Pulse,2,0,5``` is working perfectly.

khbraemer, 2019-08-12:
> Same here, I tried:
> ESP_Easy_mega-20190809_normal_ESP8266_4M.bin
> and
> ESP_Easy_mega-20190630_normal_ESP8266_4M.bin
> 
> `LongPulse,2,0,5` and `LongPulse_mS,2,0,5` are not working. The LED turns on and stays on.
> `Pulse,2,0,5` is working perfectly.

khbraemer, 2019-08-12:
Hello, i use this command for some month and this works also fine with code ESP_Easy_mega-20190809_normal_ESP8266_4M.bin.
The device is defined as  “Switch input - Switch”  related to an GPIO.

On Clock#Time=All,%sunset+28m% do
   longpulse,4,0,25
endon

closed 2019-08-12 by khbraemer