letscontrolit/ESPEasy #1625: [OpenHAB MQTT] Crash due to yield panic

TD-er, 2018-08-03:
As reported here: https://github.com/letscontrolit/ESPEasy/issues/1574#issuecomment-410065300

Suggested fix:
Make the callback function just register handling of the MQTT as a new job in the scheduler.
This will result in all calls made during processing of the MQTT message being called from the `loop()` and thus be allowed to call `yield()` or `delay()`

> It's due to a panic reboot. Happened after a MQTT publish to my Openhab home automation system. Here's the log:

```
11514348 : NEXTION075 : code: |s,i16,sOff
11514353 : EVENT: NEXTION#idx=16.00
11514394 : ACT  : Publish /ESPEZ_Laundry/NEXTION/idx,16
11514408 : Command: publish

Panic C:\Users\Tom\AppData\Local\Arduino15\packages\esp8266\
hardware\esp8266\2.4.1\cores\esp8266\core_esp8266_main.cpp:99 __yield

ctx: cont 
sp: 3fff3e00 end: 3fff4750 offset: 01b0

>>>stack>>>
3fff3fb0:  00000001 00000001 3fff40f4 3fff2419  
3fff3fc0:  00000001 00000001 3fff899c 4025fab8  
3fff3fd0:  00000001 00000001 3fff899c 40241651  
3fff3fe0:  00000c01 00000000 00000000 40100200  
3fff3ff0:  00000000 00000010 00000000 00000000  
3fff4000:  00000000 00000000 3fff8b14 0000000f  
3fff4010:  00000000 3fff8b2c 0000000f 00000000  
3fff4020:  3fff8b44 0000000f 00000000 3fff8b5c  
3fff4030:  0000000f 00000000 3fff8b74 0000000f  
3fff4040:  00000000 00000000 6c627550 00687369  
3fff4050:  00000046 3fff4110 0000000d 402173d7  
3fff4060:  00000600 4c5f1800 00000000 4e2f7901  
3fff4070:  00000000 00000010 00000000 00000000  
3fff4080:  00000000 00000000 00000000 00000000  
3fff4090:  00000000 00000000 50003600 4c5f5a45  
3fff40a0:  646e7561 4e2f7972 49545845 692f4e4f  
3fff40b0:  00007864 00000000 00000000 00000000  
3fff40c0:  00000000 00000000 0000032d 4025ea7b  
3fff40d0:  312c7864 00000036 3fff41dc 4025eaad  
3fff40e0:  3fff41dc 00000000 3fff26ae 00000006  
3fff40f0:  00000001 3fff8afc 0000000f 00000000  
3fff4100:  00000001 00000001 00000001 40241dd5  
3fff4110:  00000c00 00000000 00000000 3fff5300  
3fff4120:  00000000 00000010 00000000 00000000  
3fff4130:  00000000 00000000 3fff8a34 0000000f  
3fff4140:  00000000 3fff8a4c 0000000f 00000000  
3fff4150:  3fff8a7c 0000000f 00000000 3fff8a94  
3fff4160:  0000000f 00000000 3fff8aac 0000000f  
3fff4170:  00000000 00000000 4e5b2c78 49545845  
3fff4180:  69234e4f 0d5d7864 6520200a 6669646e  
3fff4190:  20200a0d 5b206669 5458454e 234e4f49  
3fff41a0:  5d786469 30353d3e 2f202030 6f54202f  
3fff41b0:  20686375 6e657645 40107b10 00000000  
3fff41c0:  000003e8 00000000 3fff87fc 3fff883c  
3fff41d0:  00000000 00000000 00000000 3fff8ac4  
3fff41e0:  0000002f 00000025 3fff89d4 0000002f  
3fff41f0:  00000025 3fff899c 0000002f 00000025  
3fff4200:  3fff87e4 0000000f 00000000 3fff8964  
3fff4210:  0000002f 00000025 3fff891c 0000003f  
3fff4220:  00000025 3fff8a0c 0000001f 0000001b  
3fff4230:  00000001 00000001 00000001 00000001  
3fff4240:  3fff42d0 00000010 3fff4280 00000040  
3fff4250:  00000000 3fff2419 3fff428c 4025ebb0  
3fff4260:  3fff0031 3fff2419 00000001 00afb1f1  
3fff4270:  3fff42d0 3fff2419 00000001 40241f93  
3fff4280:  00000000 00000000 00000000 3fff87cc  
3fff4290:  0000000f 0000000a 3fff879c 0000000f  
3fff42a0:  0000000a 00000030 3fff42d0 4025f058  
3fff42b0:  3fff259e 00000000 00000000 00000005  
3fff42c0:  00000010 00000004 00000000 40242b55  
3fff42d0:  3fff8774 0000001f 00000011 4010020c  
3fff42e0:  3fff23c4 3fff0e58 3fff4310 40100690  
3fff42f0:  00000000 00000000 00000000 00000011  
3fff4300:  3fff1b68 3fff26a8 3fff4610 40244435  
3fff4310:  00000000 00000000 00000000 00000011  
3fff4320:  3fff1b68 3fff0e58 3fff4610 40264db4  
3fff4330:  ae4f0759 3fff0e58 3fff4560 40262844  
3fff4340:  3fff0030 3fff0e58 3fff45a8 00000011  
3fff4350:  3fff1b68 3fff0e58 3fff4610 4024a0c7  
3fff4360:  692c737c 732c3631 0066664f 0083126e  
3fff4370:  3fff5ccc 3fff4670 3fff4478 00000030  
3fff4380:  4021515a 3fff4670 3fff64b4 3fff645c  
3fff4390:  0000072f 0000002f 00000000 00000089  
3fff43a0:  00000000 00000000 0000001f 40104ec9  
3fff43b0:  4000050c 3fff1b00 3fff4470 00000030  
3fff43c0:  40000f83 00000030 00000000 ffffffff  
3fff43d0:  40285942 00000023 3fffbff8 00000e71  
3fff43e0:  000006b0 00000e70 00000001 3fff8a78  
3fff43f0:  000000c2 3fff64dc 00000001 3fff4b5c  
3fff4400:  3fff3640 00000000 00007388 00000030  
3fff4410:  40103e13 3ffef740 50800000 4000050c  
3fff4420:  3fffc278 40101f8c 3fffc200 00000022  
3fff4430:  3fff4440 00000023 3fff84fc 40214afa  
3fff4440:  ffffe660 00000020 00000000 4010053d  
3fff4450:  401004ba 00000023 00000003 3fff4490  
3fff4460:  3fff1b00 3fff4468 3fff4670 40214bbc  
3fff4470:  3fff4470 3fff4470 3fff4478 3fff4478  
3fff4480:  3fff4480 3fff4480 3fff4488 3fff4488  
3fff4490:  3fff4490 3fff4490 3fff4498 3fff4498  
3fff44a0:  3fff44a0 3fff44a0 3fff44a8 3fff44a8  
3fff44b0:  3fff44b0 3fff44b0 3fff44b8 401071ee  
3fff44c0:  3fff44c0 3fff44c0 3fff4610 40210069  
3fff44d0:  3fff44d0 3fff44d0 3fff44d8 40244d60  
3fff44e0:  3fff44e0 3fff44e0 3fff44e8 3fff44e8  
3fff44f0:  3fff44f0 3fff44f0 3fff44f8 3fff44f8  
3fff4500:  00000000 000006c0 000006c0 4010020c  
3fff4510:  3fff3640 000006c0 000006c0 00000046  
3fff4520:  3fff3648 00000000 3fff16d0 40285968  
3fff4530:  00000014 000035d8 000006bb 40207b90  
3fff4540:  00000010 00000010 3fff178c 4028f34f  
3fff4550:  4025d1f4 3fff4610 3fff178c 4025ebb0  
3fff4560:  00000000 00000000 00000000 4020b47c  
3fff4570:  3fff3640 000006bb 000006bb 4010020c  
3fff4580:  00000004 3fff4610 3fff45b0 40100690  
3fff4590:  3fff875c 0000000f 0000000b 3fff8744  
3fff45a0:  0000000f 00000000 3fff8r‚)ŠLþ
ôŒ
JE0¸
†´ˆˆ9!9¯„ù„!ÁÎì)ÿ18Áù526 : INIT : Free RAM:21328
527 : INIT : I2C
527 : INIT : SPI not enabled
541 : NEXTION075 : serial pin config RX:13, TX:15, Plugin Enabled
543 : NEXTION075 : Using software serial
599 : INFO : Plugins: 71 [Normal] [Testing] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
696 : EVENT: System#Wake
745 : WIFI : Set WiFi to STA
```

TD-er, 2018-08-03:
Maybe you could test this PR: https://github.com/letscontrolit/ESPEasy/pull/1627

I just tested it will compile and did not crash my ESP32 node when pressing the test notification.
But since it is already close to 2am here, I see an opportunity for those still in day-light side of the globe ;)

thomastech, 2018-08-04:
I loaded PR 1627 into two nodeMCU boards. Identical configurations, except only one of them will do MQTT publish. 

I'll run them overnight and check their uptime in the morning. If uptime is wrong then we will know that there was at least one reboot.

- Thomas

thomastech, 2018-08-04:
It's important to note that although both boards have identical software configurations, the one that is working well is missing the the MPU6050 sensor I use to detect machine movement.  So later today I will disable the MPU6050 and see if that stops the resets.
- Thomas

TD-er, 2018-08-04:
Is there a possibility the broker is busy or for some other reason not answering?

thomastech, 2018-08-04:
That is a good question. Openhab is running correctly. It's installed on a RaspberryPi Model 3+.

I studied the log messages and they seem to suggest that the panic occurred while in misc.ino's rulesProcessingFile(). That's because the last log entry before the crash is  "ACT :" which is in rulesProcessingFile().  This log output is printed immediately before the PluginCall(). 

Also, whenever the openhab controller _C005.ino does a MQTTpublish() it prints a "MQTT :" log entry. But we don't see that in the logs. So I don't know if the publish ever occurred.

Not sure what these debug crumbs are telling us. Do you have any ideas?

- Thomas

TD-er, 2018-08-04:
How do you send the publish commands?
Are they commands from a rule?

thomastech, 2018-08-04:
Yes the publish actions are in the ESPEasy rule file. To further debug this I've commented them out and restarted my tests. If the reboot appears then the problem is not caused by MQTT publish.

Here's the rule excerpt:
```
on NEXTION#idx do
  if [NEXTION#idx]>=10 and [NEXTION#idx]<=30
      Publish /%sysname%/NEXTION/idx,[NEXTION#idx]
  endif
  if [NEXTION#idx]>=500  // Touch Events
      Publish /%sysname%/NEXTION/idx,[NEXTION#idx]
      Publish /%sysname%/NEXTION/value,[NEXTION#value]
  endif
endon

```
- Thomas

thomastech, 2018-08-05:
It's been running trouble free for four hours after removing the publish statements from ESPEasy's rule file. So I am back to testing with MQTT publish.

I looked at the core and found the panic function. There's no Serial.flush before the exception, so my theory is that we lose the last buffered serial log messages at the panic reboot. Just a theory.

I edited _C005.ino and added a couple more addLog statements with Serial.flush on both sides of the MQTTpublish(). So hopefully that helps push out the debug messages before a panic reboot. 

I've restore the rule file's publish statements and I am now restarting the testing. 

- Thomas

TD-er, 2018-08-05:
This "Controller Message Interval" variable is being used for multiple things and one of them is actually a blocking wait for that period.
It does freeze the system for that amount of time.
It is also used in places where another variable would make more sense instead of re-use.

So that's one thing I would like to change. It looks like it is meant to be used to give the MQTT broker some time to process requests, but I guess using a buffer at the controller side would make more sense.

This yield panic crash occurs when a `yield` or `delay` (which calls yield) is being used in a function that's being called from somewhere else.
Normally all functions are being called from either `setup()` or `loop()`. But callback functions are called from the core, so their scope does not include access to the `yield` function to run background tasks.

Yesterday I split the rules processing function to make it more readable.
See: https://github.com/TD-er/ESPEasy/commit/be7724d8c7c15582438af34877f100d63ad8efbe
Today I start looking at `ExecuteCommand` to see if there is something that can be handled asynchronous via the scheduler to make it being called from the `loop()` again.

thomastech, 2018-08-05:
@TD-er: Thanks for the explanation. I was aware that you were working on the _Controller Message Interval._

My test run last night was interesting. Only one reboot, whereas I would have expected several. Could have been blind luck, or something about my altered test conditions contributed to stability.

But the overnight test's panic reboot provided another clue. Here's the log excerpt:
 
```
13901389 : Before MQTT
13901393 : After MQTT
13901395 : MQTT : /ESPEZ_Laundry/RSSI/signal -79
13905306 : NEXTION075 : code: |s,i26,sOff
13905308 : Read settings: TaskSettings index: 4
13905312 : EVENT: NEXTION#idx=26.00
RuleDebug Processing:rules1.txt
     flags CMI  parse output:
RuleDebug: 001: endon
RuleDebug: 100: washer#ac do
RuleDebug: 101: if [washer#ac]=1
RuleDebug: 101: nextion,page0.va_washerac.val=1
RuleDebug: 101: else
RuleDebug: 101: nextion,page0.va_washerac.val=0
RuleDebug: 101: endif
RuleDebug: 000: endon
RuleDebug: 100: dryermov#detect do
RuleDebug: 101: if [dryermov#detect]=1
RuleDebug: 101: nextion,page0.va_dryerac.val=1
RuleDebug: 101: else
RuleDebug: 101: nextion,page0.va_dryerac.val=0
RuleDebug: 101: endif
RuleDebug: 000: endon
RuleDebug: 110: nextion#idx do
13905359 : Read settings: TaskSettings index: 1
13905368 : Read settings: TaskSettings index: 2
13905370 : Read settings: TaskSettings index: 4
13905372 : DEBUG DEV: Parsed String='  if 26'
13905375 : Read settings: TaskSettings index: 1
13905381 : Read settings: TaskSettings index: 2
13905385 : Read settings: TaskSettings index: 4
13905389 : DEBUG DEV: Parsed String='  if 26>=10 and 26'
RuleDebug: 111: if 26>=10 and 26<=30
13905396 : [if 26>=10 and 26<=30]=true
13905401 : Read settings: TaskSettings index: 1
13905406 : Read settings: TaskSettings index: 2
13905410 : Read settings: TaskSettings index: 4
13905414 : DEBUG DEV: Parsed String='      Publish /%sysname%/NEXTION/idx,26'
RuleDebug: 111: publish /espez_laundry/nextion/idx,26
13905424 : ACT  : Publish /ESPEZ_Laundry/NEXTION/idx,26
13905441 : Command: publish

Panic C:\Users\Tom\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\core_esp8266_main.cpp:99 __yield

ctx: cont 
sp: 3fff3e70 end: 3fff47c0 offset: 01b0

>>>stack>>>

```
The "Before MQTT" and "After MQTT" are addLog message wrappers I placed before and after the MQTTpublish() in _C005.ini.  I only mention this because you are probably wondering what they are.

The interesting part is the _Command: publish_ log message immediately before the panic reboot. It is log message that occurs inside misc.ino's ExecuteCommand().

And it's even more interesting is that you are starting to look at ExecuteCommand. So hopefully this rabbit hole journey is taking us in the right direction.

- Thomas

TD-er, 2018-08-05:
This afternoon I added some commit just to make the commands return a String stating their return status.
That has nothing really to do with this issue, but it gave me time to look at all the commands to see if there is something that cannot be done asynchronous.
So I will now try to add an entry to schedule these command executions as a scheduled job.
This will change the scheduled job a bit to keep the extra text line and the events in sync, but that's just a minor change.

Hopefully I find enough time this evening to implement it so you can test it.
This time zone difference between us is nice to get some tests done :)

thomastech, 2018-08-06:
Let me know when your changes are complete and need me to try them out.
- Thomas

TD-er, 2018-08-06:
I did change some this morning, but there are still some issues with that, which I will fix.... now ;)

TD-er, 2018-08-06:
I just pushed a fix to the existing pull request.
Could you (stress) test it?

In short:

- All commands will be executed from the scheduler, except the commands sent from the Tools webpage.
- Commands run from the web page will reply with a more meaningful reply.
- Rules should be executed faster, since some commands like "publish" will now be started from the scheduler.
- Commands will now be queued and thus not lost. (this queue may take quite some memory per item)

Still to do:

- Maybe introduce some artificial delay between commands when sending a lot at once.

thomastech, 2018-08-06:
I have it running now. Everything is working fine. I'll keep an eye out for reboots or other odd behavior and report back tomorrow (or sooner if something goes wrong tonight).

FWIW, the updated test branch doesn't seem to have the latest src files. For example, it did not have my recent Nextion.ino. I didn't check any other files for "freshness."

Thank you for your efforts to make the code more stable. It is appreciated.

- Thomas

thomastech, 2018-08-07:
After running for about one hour it rebooted. I'll run it again with serial log enabled and capture the exception.

- Thomas

TD-er, 2018-08-07:
@thomastech 
Could you also test without actively using the publish method in the Nextion plugin?
And maybe stress test it also, then I can merge this code and we can have a look at your Nextion code.
I didn't have a proper look at it yet.

thomastech, 2018-08-07:
@TD-er:  My overnight run had 100% success. No reboots, still running (~12 hours). That's a record!  I will continue to let it run and leave it alone. 

I have a second NodeMCU running without publish in the rules. It has not rebooted, 15 hours so far. So the latest PR has not caused any new stability issues.

- Thomas

TD-er, 2018-08-07:
And the 12-hour running version is using Nextion code that was involved before when it crashed?

thomastech, 2018-08-07:
Yes, same Nextion code. 

During this test I shut down my PC and left the device to run without anything messing with it, including me.  Even when I don't touch it, there's plenty of MQTT actions being executed since the device constantly publishes status messages every 20 seconds.

I'm hoping that yesterday's reboot was unrelated to the rule's publish panic problem. But it's too early to know anything. It's best to watch and wait, then check the serial log next time it reboots.

In the meantime, it would be great if you reviewed the Nextion code. Although the panic reboot occurs while in the rules (not in Nextion), maybe something in the Nextion plugin is involved. 

- Thomas

TD-er, 2018-08-07:
It looks like it is the only plugin calling the `ExecuteCommand` function directly.

thomastech, 2018-08-07:
@TD-er: I have to agree that ExecuteCommand is involved with this issue. If the Nextion module is the only one that directly calls it then that certainly makes it look like a bad guy.

The "12-hour" NodeMCU is still running OK. I've been randomly interacting with it to stress test the code and perhaps provoke a reboot.

- Thomas

thomastech, 2018-08-07:
While writing that last comment it did a reboot. This time it was Exception (3) rather than a panic.

Here's a log excerpt leading up to the crash:

```
44981799 : MQTT : /ESPEZ_Laundry/RSSI/signal -78
44982911 : NEXTION075 : code: |s,i26,sOff
44982916 : EVENT: NEXTION#idx=26.00
44982947 : [if 26>=10 and 26<=30]=true
44982958 : ACT  : Publish /ESPEZ_Laundry/NEXTION/idx,26
44982984 : [if 26>=500]=false
44983015 : EVENT: NEXTION#idx=26.00 Processing time:99 milliSeconds
44983017 : EVENT: NEXTION#value=0.00
44983041 : EVENT: NEXTION#value=0.00 Processing time:24 milliSeconds
44983045 : NEXTION075 : Pipe Command Sent: |s,i26,sOff26.00
44983053 : Command: publish
44989810 : NEXTION075 : code: |s,i98,sOff
44989813 : EVENT: NEXTION#idx=98.00
44989849 : [if 98>=10 and 98<=30]=false
44989871 : [if 98>=500]=false
44989900 : ACT  : NEXTION,page7.t_wifi_ssid.txt="SSID: Wizard1_2G4"
44989951 : NEXTION075 : WRITE, Command is page7.t_wifi_ssid.txt="SSID: Wizard1_2G4"
44989956 : ACT  : NEXTION,page7.t_ip.txt="IP: 192.168.1.217"
44989998 : NEXTION075 : WRITE, Command is page7.t_ip.txt="IP: 192.168.1.217"
44990014 : ACT  : NEXTION,page7.t_signal.txt="RSSI: -78dBm"
44990055 : NEXTION075 : WRITE, Command is page7.t_signal.txt="RSSI: -78dBm"
44990061 : ACT  : NEXTION,page7.t_date.txt="Date 08:07:18"
44990102 : NEXTION075 : WRITE, Command is page7.t_date.txt="Date 08:07:18"
44990106 : ACT  : NEXTION,page7.t_time.txt="Time 08:35:16 "
44990147 : NEXTION075 : WRITE, Command is page7.t_time.txt="Time 08:35:16 "
44990162 : ACT  : NEXTION,page7.t_uptime.txt="Uptime 749.0days"
44990207 : NEXTION075 : WRITE, Command is page7.t_uptime.txt="Uptime 749.0days"
44990209 : EVENT: NEXTION#idx=98.00 Processing time:396 milliSeconds
44990211 : EVENT: NEXTION#value=0.00
44990236 : EVENT: NEXTION#value=0.00 Processing time:24 milliSeconds
44990239 : NEXTION075 : Pipe Command Sent: |s,i98,sOff98.00
44991053 : NEXTION075 : code: |s,i98,sOff
44991056 : EVENT: NEXTION#idx=98.00
44991088 : [if 98>=10 and 98<=30]=false
44991110 : [if 98>=500]=false
44991140 : ACT  : NEXTION,page7.t_wifi_ssid.txt="SSID: Wizard1_2G4"
44991190 : NEXTION075 : WRITE, Command is page7.t_wifi_ssid.txt="SSID: Wizard1_2G4"
44991196 : ACT  : NEXTION,page7.t_ip.txt="IP: 192.168.1.217"
44991239 : NEXTION075 : WRITE, Command is page7.t_ip.txt="IP: 192.168.1.217"
44991254 : ACT  : NEXTION,page7.t_signal.txt="RSSI: -78dBm"
44991299 : NEXTION075 : WRITE, Command is page7.t_signal.txt="RSSI: -78dBm"
44991304 : ACT  : NEXTION,page7.t_date.txt="Date 08:07:18"
44991344 : NEXTION075 : WRITE, Command is page7.t_date.txt="Date 08:07:18"
44991349 : ACT  : NEXTION,page7.t_time.txt="Time 08:35:18 "
44991390 : NEXTION075 : WRITE, Command is page7.t_time.txt="Time 08:35:18 "
44991404 : ACT  : NEXTION,page7.t_uptime.txt="Uptime 749.0days"
44991450 : NEXTION075 : WRITE, Command is page7.t_uptime.txt="Uptime 749.0days"
44991452 : EVENT: NEXTION#idx=98.00 Processing time:396 milliSeconds
44991454 : EVENT: NEXTION#value=0.00
44991481 : EVENT: NEXTION#value=0.00 Processing time:27 milliSeconds
44991484 : NEXTION075 : Pipe Command Sent: |s,i98,sOff98.00
44991501 : SW   : State 0.00
44991504 : EVENT: WASHER#ac=0.00
44991515 : [if 0=1]=false
44991518 : else = true
44991520 : ACT  : NEXTION,page0.va_WasherAC.val=0
44991552 : NEXTION075 : WRITE, Command is page0.va_WasherAC.val=0
44991572 : EVENT: WASHER#ac=0.00 Processing time:68 milliSeconds
44992044 : MQTT : /ESPEZ_Laundry/WASHER/ac 0
45001103 : SW   : State 0.00
45001106 : EVENT: WASHER#ac=0.00
45001117 : [if 0=1]=false
45001120 : else = true
45001121 : ACT  : NEXTION,page0.va_WasherAC.val=0
45001154 : NEXTION075 : WRITE, Command is page0.va_WasherAC.val=0
[... 264 lines cut ...]

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
ªU105 : 

INIT : Booting version: (custom) (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
106 : INIT : Warm boot #2 - Restart Reason: Exception
110 : FS   : Mounting...
184 : FS   : Mount successful, used 77057 bytes of 2949250
227 : CRC  : No program memory checksum found. Check output of crc2.py
276 : CRC  : SecuritySettings CRC   ...OK 
553 : INIT : Free RAM:21280
554 : INIT : I2C
554 : INIT : SPI not enabled
568 : NEXTION075 : serial pin config RX:13, TX:15, Plugin Enabled
568 : NEXTION075 : Using software serial
581 : INFO : Plugins: 71 [Normal] [Testing] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
582 : EVENT: System#Wake
607 : WIFI : Set WiFi to STA
```
- Thomas

TD-er, 2018-08-07:
Do you also happen to trace/log the memory usage?

TD-er, 2018-08-07:
That would be great.
I probably can't decode your stack trace, since it is your own build

TD-er, 2018-08-07:
Power supply can indeed cause these issues.
But also a failed memory allocation, that's why I asked for tracking memory usage to see if it is decreasing over time

thomastech, 2018-08-07:
@TD-er:
Does ESPEasy have a built-in feature for tracking memory usage? 

Or do I need to use a special tool like the one I mentioned earlier? Do you have a better recommendation that works with Arduino IDE?

- Thomas

TD-er, 2018-08-07:
You can add a general sysinfo plugin and set it to log memory usage

thomastech, 2018-08-07:
> Just a espeasy plugin ;)

That's the ticket. Installed and I'm now monitoring memory. Only 14K bytes and dips to 12K on occasion; Wiki says >20K is ideal.

- Thomas

TD-er, 2018-08-07:
This recommendation of 20k is from before the new webserver we're now using.
About a year ago, the entire web page was rendered in memory, which needed a String containing the entire page (up-to 11 k)
Now we render the entire page in 400 bytes of memory :) (@s0170071 made this happen)

The number of included plugins also has impact on the amount of free memory.
The left-over free memory also decreased over time with newer core libraries.

With "normal" builds, you get around 15k free.
The events I buffer for the commands are also quite huge, which I should also have a look at.
And we're still holding a lot in memory about the present plugins. A lot of that information can be loaded from flash at runtime. (flash is about as fast as RAM when reading on ESP8266)

thomastech, 2018-08-07:
@TD-er: Thanks for the clarification on memory requirements. My conclusion is that the publish related crash/reboot isn't a dirty power or low memory issue. 

Other than performing tests, I'm not much use to you. If you need me to do anything special then please let me know. 

- Thomas

TD-er, 2018-08-07:
Could you make it less hot over here? That would be great. :)
They were [pouring salt on the roads](https://www.rtvnoord.nl/nieuws/197501/Midden-Groningen-stuurt-strooiwagen-op-pad) (something you normally see in winter), to protect them from melting.

So if you want to help... please do that, will you? ;)

TD-er, 2018-08-07:
I made a [build](https://www.dropbox.com/s/nxats6wfgdasj6g/ESPEasy_mega-20180804-11-g7783cf4.zip?dl=0) of the latest merges, maybe you could test one of those?
It will be the same as tomorrow's, unless I will merge something else too ;)

thomastech, 2018-08-07:
My USA state is on fire (literally). And also suffering from high temperature weather. So hot salty roads seem like a dream paradise to me. 

Regarding your _Generic HTTP Advanced crash #1536_, my home automation system sometimes responds back to ESPEasy's publish actions using HTTP. It also uses HTTP messages to ESPEasy during remotely activated features that use Alexa voice actions. 

I think it is important to mention that during the most recent crash I was testing some functions that involved HTTP command messages to ESPEasy. It was part of my stress testing.

Does any of this sound like it could be related to your #1536. Is it possible there are multiple reasons for the crash/reboots and we're chasing all the symptoms like a cat chasing a laser pointer?

> I made a build of the latest merges, maybe you could test one of those?

I will test you new build. 

- Thomas

TD-er, 2018-08-07:
@thomastech Hmm fires is indeed a bit worse.
We (fortunately) only had a few and today a big one.
@Grovkillen also had a lot of (really big) fires in Sweden. ([they bombed one to fight the fire](https://www.svt.se/nyheter/flera-skogsbrander-i-landet))

And I think the issue reported in #1536 indeed may also need to have a look at.

thomastech, 2018-08-07:
Looks like we're all suffering in similar ways;  Fixing software bugs is a welcome distraction.

I've flashed with your ESP_Easy_mega-20180804-11-g7783cf4_test_ESP8266_4096.bin. It reverted to AP mode and I had to reload config and rules. The "20180804" date in the name seems odd, so hopefully I used the correct file.

I'm doing the usual testing. If there's anything special you want me to do then let me know.

- Thomas

TD-er, 2018-08-07:
I built it on my "Bash on Windows" environment using almost the same script as being used during the nightly builds.
That one bases the name on the last tag and I just fetched the repo. 20180804 is the last tag.

thomastech, 2018-08-08:
@TD-er: The latest test run lasted a little over 4 hours before a crash/reboot. This time it was _Restart Reason: Hardware Watchdog_ See attached log, time 60189 for the start of the crash.

At the beginning of the test the memory was >14K bytes. It slowly decreased and towards the end it was hovering around 10K. The memory values are in the log file as entries marked with _EVENT: FreeRam#ram_.  So besides the reboot issue there appears to be a memory leak. 

- Thomas

[log3.zip](https://github.com/letscontrolit/ESPEasy/files/2268563/log3.zip)

thomastech, 2018-08-08:
@TD-er: After the previous reboot I restarted the test. This time I disabled all logs, switched Nextion from softserial to UART serial, avoided all HTTP access to ESPEasy, and ran on a dedicated power supply. It's been running for >13 hours without any issues. The memory leak is gone too.

My second NodeMCU test system has been running for 40 hours. No memory leak. Log is disabled, Nextion is UART serial, MQTT is disabled, and no HTTP. This system is really not doing much activity, other than breathing. Not an ideal reference since it isn't being stressed. But still useful info.

I have not experienced a "Panic" reboot since installing the scheduler PR firmware. Going down the list of possible culprits to the recent reboots, I recall reading a post about stability issues caused by enabling serial log. Is that still a concern? 

- Thomas

TD-er, 2018-08-08:
Serial log may still act strange, when there is no reader.
I am not sure what's happening then. It may be something stalling and thus causing a software/hardware watchdog reboot, or just filling memory. I don't know.

thomastech, 2018-08-08:
@TD-er: That's good to know.  I'm still suspicious of Serial log, so I'll do further investigation. 

In the meantime I've turned Nextion Soft Serial back on. It's been running for about an hour. No reboots or memory leaks yet. I'll let it run for a few more hours and if all is well I'll move down the suspect list.

Lastly, you mentioned that Nextion is the only plugin that directly calls ExecuteCommand(). I had forgotten that early-on in my Nextion hardware serial update I removed that function because it appeared unnecessary. Was removing ExecuteCommand() a bad idea and perhaps the reason for all this trouble? Or accidental genius? :)

- Thomas

TD-er, 2018-08-08:
> Was removing ExecuteCommand() a bad idea and perhaps the reason for all this trouble? Or accidental genius? :)

I guess the best answer to that question is: "yes" ;)

thomastech, 2018-08-08:
> I guess the best answer to that question is: "yes" ;)

I see what you did there. I will ask my Magic 8-ball for a more decisive opinion.

The test system is still running (~17 hours) and no memory leaks.  Yet.

- Thomas

thomastech, 2018-08-08:
@TD-er: The test system is still running OK. Next I will turn on Serial log to see if it contributes to reboots or memory leaks.

I did an experiment with serial log on my second test device. It found evidence that the ESP8266's serial Tx buffer is being over-filled, which is something that can cause code blocking. My research (google) found that this is a situation that can cause a ESP8266 crash.

The experiment was simple. I added a Serial.availableForWrite() to the addToLog() function (in misc.ino). It checks to see how much room is left in the Tx buffer. If space is insufficient it trims the log string so that it can fit in the buffer (without causing a "full" condition).

This is the test code:
```
void addToLog(byte logLevel, const char *line)
{
  if (loglevelActiveFor(LOG_TO_SERIAL, logLevel)) {
    Serial.print(millis());
    Serial.print(F(" : "));
    if(strlen(line)+5 >  (uint16_t)(Serial.availableForWrite())) {
        uint16_t snip = (uint16_t)(Serial.availableForWrite());
        if(snip <6) snip = 6;
        String linestr = line;
        linestr.substring(0,snip-6);
        Serial.print(linestr);
        Serial.println(F(",snip!"));             // Sorry, but not enough room for your entire log message.
    }
    else {
        Serial.print(F("["));
        Serial.print(String(Serial.availableForWrite()));
        Serial.print(F("], "));
        Serial.println(line);
    }
  }
  if (loglevelActiveFor(LOG_TO_SYSLOG, logLevel)) {
    syslog(logLevel, line);
  }
  if (loglevelActiveFor(LOG_TO_WEBLOG, logLevel)) {
    Logging.add(logLevel, line);
  }

#ifdef FEATURE_SD
  if (loglevelActiveFor(LOG_TO_SDCARD, logLevel)) {
    File logFile = SD.open("log.dat", FILE_WRITE);
    if (logFile)
      logFile.println(line);
    logFile.close();
  }
#endif
}

```
I'm seeing periodic log entries that are being trimmed. So that means the native log message would have filled the buffer and caused a temporary block. 

The log messages from the plugins are not a problem. The offending messages are from the system (PluginStats, Plugin call, etc.). Here's some examples, where trimmed messages end with _snip!_

```
20834 : [120], NEXTION075 : Cmd Statement Line-1 Sent: page0.j1.val=45
20856 : [120], SYS  : -79.00
20860 : [120], EVENT: RSSI#signal=-79.00
20880 : [120], EVENT: RSSI#signal=-79.00 Processing time:19 milliSeconds
20958 : [120], MQTT : /ESPEZ_Laundry/RSSI/signal -79
21094 : [120], SYS  : 15040.00
21096 : [116], EVENT: FreeRam#ram=15040.00
21117 : [120], EVENT: FreeRam#ram=15040.00 Processing time:21 milliSeconds
25449 : Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: -420 minSTD time start: 2018-10-28 02:00:00 offset: -480 min,snip!
25451 : EVENT: Time#Initialized,snip!
25473 : [120], EVENT: Time#Initialized Processing time:23 milliSeconds
25478 : [99], EVENT: Clock#Time=Wed,13:55
25498 : [120], EVENT: Clock#Time=Wed,13:55 Processing time:20 milliSeconds
26040 : [120], SYS  : 15152.00
26043 : [116], EVENT: FreeRam#ram=15152.00
26063 : [120], EVENT: FreeRam#ram=15152.00 Processing time:19 milliSeconds
30668 : [120], SW   : State 0.00
30670 : [114], EVENT: WASHER#ac=0.00
30678 : [120], [if 0=1]=false
[... 32 lines cut ...]
33076 : SensorSendTask()     stats: Count: 11 Avg/min/max 78003.37/24929/131677 usec,snip!
33084 : sendData()           stats: Count: 10 Avg/min/max 63515.10/24490/127186 usec,snip!
33092 : Compute formula      stats: Count: 10 Avg/min/max 139.00/10/1295 usec,snip!
33099 : setNewTimerAt()      stats: Count: 1759 Avg/min/max 187.72/158/283 usec,snip!
33106 : timeDiff()           stats: Count: 1666387 - CPU cycles per call: 8.44,snip!
33114 : Scheduler stats: (called/tasks/max_length/idle%) 527198/1751/12/74.60,snip!
36040 : [120], SYS  : 15152.00
36042 : [116], EVENT: FreeRam#ram=15152.00
36064 : [120], EVENT: FreeRam#ram=15152.00 Processing time:22 milliSeconds
40668 : [120], SW   : State 0.00
40671 : [114], EVENT: WASHER#ac=0.00
40679 : [120], [if 0=1]=false
40681 : [120], else = true
40683 : [109], ACT  : NEXTION1,page0.va_WasherAC.val=0
40717 : [120], NEXTION075 : WRITE, Command is page0.va_WasherAC.val=0
40734 : [120], EVENT: WASHER#ac=0.00 Processing time:63 milliSeconds
40795 : [120], MQTT : /ESPEZ_Laundry/WASHER/ac 0
40834 : [120], NEXTION075 : Cmd Statement Line-1 Sent: page0.j1.val=45
40857 : [120], SYS  : -77.00
40860 : [120], EVENT: RSSI#signal=-77.00
40882 : [120], EVENT: RSSI#signal=-77.00 Processing time:21 milliSeconds

```

- Thomas

TD-er, 2018-08-08:
This is really great, since I tried it a while back and then I could not get a reliable detection whether the output was read or not.

My idea was to temporary disable log to serial when the buffer stays full.

thomastech, 2018-08-08:
> This is really great, since I tried it a while back and then I could not get a reliable detection whether the output was read or not.

@TD-er: I believe the ESP8266's USB serial port still sends the Tx data if there's nothing reading it.  

I made a mistake in the log message trimming code. It was just adding _trim!_ at the end of the offending log messages. I've edited my previous post and it has the code corrections.

Now I can see that the problem is much worse than expected. There are long periods of possible code blocks if the log messages are not trimmed. Here's a revised log excerpt (serial log baud rate set to 115.2K):
```
122828 : [119], EVENT: NEXTION#idx=16.00
122861 : [119], [if 16>=10 and 16<=30]=true
122873 : [119], ACT  : Publish /ESPEZ_Laundry/NEXTION/idx,16
122899 : [119], [if 16>=500]=false
122933 : [119], EVENT: NEXTION#idx=16.00 Processing time:105 milliSeconds
122935 : [60], EVENT: NEXTION#value=0.00
122958 : [119], EVENT: NEXTION#value=0.00 Processing time:23 milliSeconds
123315 : [119], NEXTION075 : Pipe Command Sent: |s,i16,sOff16.00
123335 : LoopStats: shortestLoop: 36 longestLoop: 1924098 avgLoopDuration: 58.37 loopCounterMax: 833333 loo,snip!
123336 : ,snip!
123336 : ,snip!
123337 : ,snip!
123338 : ,snip!
123340 : ,snip!
123341 : ,snip!
123343 : ,snip!
123344 : ,snip!
123346 : ,snip!
123347 : ,snip!
123349 : ,snip!
123350 : ,snip!
123352 : ,snip!
123353 : ,snip!
123355 : ,snip!
123356 : ,snip!
123358 : ,snip!
123359 : ,snip!
123361 : ,snip!
123362 : ,snip!
123364 : ,snip!
123365 : ,snip!
123366 : ,snip!
123368 : ,snip!
123369 : ,snip!
123371 : ,snip!
123372 : ,snip!
123374 : ,snip!
123375 : ,snip!
123377 : ,snip!
123378 : ,snip!
123380 : ,snip!
123381 : ,snip!
123383 : ,snip!
123384 : ,snip!
123394 : [86], Command: publish
131101 : [119], SW   : State 0.00
131104 : [114], EVENT: WASHER#ac=0.00
131115 : [119], [if 0=1]=false
131118 : [119], else = true
131120 : [108], ACT  : NEXTION,page0.va_WasherAC.val=0
131153 : [119], NEXTION075 : WRITE, Command is page0.va_WasherAC.val=0
131171 : [119], EVENT: WASHER#ac=0.00 Processing time:67 milliSeconds
131232 : [119], MQTT : /ESPEZ_Laundry/WASHER/ac 0
132691 : [119], NEXTION075 : code: |s,i26,sOff
132697 : [119], EVENT: NEXTION#idx=26.00
132729 : [119], [if 26>=10 and 26<=30]=true
132741 : [119], ACT  : Publish /ESPEZ_Laundry/NEXTION/idx,26
132769 : [119], [if 26>=500]=false
132802 : [119], EVENT: NEXTION#idx=26.00 Processing time:104 milliSeconds
132803 : [60], EVENT: NEXTION#value=0.00
132827 : [119], EVENT: NEXTION#value=0.00 Processing time:24 milliSeconds
132831 : [86], NEXTION075 : Pipe Command Sent: |s,i26,sOff26.00
141100 : [119], SW   : State 0.00
141103 : [115], EVENT: WASHER#ac=0.00
141115 : [119], [if 0=1]=false
141118 : [119], else = true
141119 : [109], ACT  : NEXTION,page0.va_WasherAC.val=0
141152 : [119], NEXTION075 : WRITE, Command is page0.va_WasherAC.val=0
141172 : [119], EVENT: WASHER#ac=0.00 Processing time:69 milliSeconds
141233 : [119], MQTT : /ESPEZ_Laundry/WASHER/ac 0
141252 : [119], Command: publish
141298 : [119], SYS  : -80.00
141302 : [119], EVENT: RSSI#signal=-80.00
141326 : [119], EVENT: RSSI#signal=-80.00 Processing time:23 milliSeconds
141796 : [119], MQTT : /ESPEZ_Laundry/RSSI/signal -80
142708 : [119], NEXTION075 : code: |s,i16,sOff
142714 : [119], EVENT: NEXTION#idx=16.00
142746 : [119], [if 16>=10 and 16<=30]=true
142757 : [119], ACT  : Publish /ESPEZ_Laundry/NEXTION/idx,16
142786 : [119], [if 16>=500]=false
142818 : [119], EVENT: NEXTION#idx=16.00 Processing time:104 milliSeconds
142819 : [60], EVENT: NEXTION#value=0.00
142843 : [119], EVENT: NEXTION#value=0.00 Processing time:24 milliSeconds
142847 : [87], NEXTION075 : Pipe Command Sent: |s,i16,sOff16.00
151100 : [119], SW   : State 0.00
151103 : [115], EVENT: WASHER#ac=0.00
151114 : [119], [if 0=1]=false
151117 : [119], else = true
151119 : [109], ACT  : NEXTION,page0.va_WasherAC.val=0
151152 : [119], NEXTION075 : WRITE, Command is page0.va_WasherAC.val=0
151171 : [119], EVENT: WASHER#ac=0.00 Processing time:67 milliSeconds
151232 : [119], MQTT : /ESPEZ_Laundry/WASHER/ac 0
151274 : [119], NEXTION075 : Cmd Statement Line-1 Sent: page0.j1.val=45
151283 : [119], Command: publish
152013 : [119], WD   : Uptime 3 ConnectFailures 0 FreeMem 14360
152710 : [119], NEXTION075 : code: |s,i26,sOff
152713 : [108], EVENT: NEXTION#idx=26.00
152745 : [119], [if 26>=10 and 26<=30]=true
152757 : [119], ACT  : Publish /ESPEZ_Laundry/NEXTION/idx,26
152783 : [119], [if 26>=500]=false
152817 : [119], EVENT: NEXTION#idx=26.00 Processing time:103 milliSeconds
152818 : [60], EVENT: NEXTION#value=0.00
152842 : [119], EVENT: NEXTION#value=0.00 Processing time:24 milliSeconds
152846 : [87], NEXTION075 : Pipe Command Sent: |s,i26,sOff26.00
152901 : LoopStats: shortestLoop: 36 longestLoop: 1924098 avgLoopDuration: 55.05 loopCounterMax: 833333 loo,snip!
152902 : ,snip!
152903 : ,snip!
152903 : ,snip!
152905 : ,snip!

```

I also tested 9600 baud with the trim code removed (reverted back to the original misc.ino). That gave repeating crash/reboots ever few seconds. That's the smoking gun we needed!

Re-installing the trim code allowed 9600 baud to run. But the log messages are severely trimmed. Basically, any serial log baud rate under 115K baud is worthless for logs. And even it needs some help, such as a big local software circular buffer dedicated to serial log.

**Latest status:**
I am using your PR with the scheduler patch (Aug 6). From what I have seen it has fixed the Panic crash/reboot issue. All we need now is a practical fix on the serial log's Tx buffer. That is to say, something more elegant than trimming useful log messages. 

- Thomas

s0170071, 2018-08-09:
@TD-er I know that you found in some experiments that there should be a serial reader for your output. 
However, afaik there is no explaination for that behaviour. There is a technique that is called handshaking https://www.codrey.com/embedded-systems/rs232-serial-communication/#What_is_Handshaking but it is not used by the ESP and can merely slow down the transmission. It would not lead to any crashes. 

I think @thomastech is on to something. The TX buffer overflow. Of course it makes no sense to write data faster than it can be sent through the wire. This would cause the TX buffer to overflow. Data is lost then. 
We should have a strategy for handling this. It could be: 
- delay the system and just wait for the buffer to become available (gives uncertain timing)
- enlarge the buffer (see #1588) (uses memory and is still not guaranteed to work)
- trim the messages by half (loss of information)
- smart trim ( trimm  more the fuller the buffer is- e.g. divide free buffer by 3 or so )

Other than that there may also be a bug in the core. There is a new release out: https://github.com/esp8266/Arduino/releases/tag/2.4.2 Maybe try this.

TD-er, 2018-08-09:
We already have a log buffer for the web log viewer.
That piece of code should be changed a little to allow to get new lines since the last log timestamp (given as argument by the requester).
Then this buffer can also be placed in front of the serial output.

And I will try to move my 'chatty' piece of statistics code to a JSON web output. That's easier to request :)

thomastech, 2018-08-09:
> The TX buffer overflow. Of course it makes no sense to write data faster than it can be sent through the wire. This would cause the TX buffer to overflow. Data is lost then.
@s0170071 : Fortunately there's never any Tx data loss with the ESP8266. Unfortunately, while the buffer is full it code blocks until the buffer empties out, then execution continues. It's these code blocking periods that provoke a crash.

> I would opt for option 1 (delay the system) 
I am working on the delay strategy. It would be a good temporary solution. But for fastest ESPEasy performance I think a big serial buffer would be nice. Plus we need to prevent the user from enabling  slow serial log baud rates.

Speaking of serial log, I'm finding some issues with the Tools->Advanced->Baud setting. The highest baud that works for me is 115.2K. Setting a higher baud seems to do nothing, it remains at 115.2K. And changing the baud requires a reboot.  When the dust settles I'll either open a bug ticket or submit a fix.

> We already have a log buffer for the web log viewer. ... Then this buffer can also be placed in front of the serial output.
@TD-er: Very nice.

- Thomas

TD-er, 2018-08-09:
@thomastech Just to be clear (you probably already got the idea), my idea is to use the same buffer, not just another instance of that buffer.

thomastech, 2018-08-09:
@TD-er: That's what I hoped you would do. But I have never been a fan of the web log. For me, it is slow and sometimes unreliable. And if the serial log is synced to the web log output then I fear those issues will affect serial log too.

I implemented the "delay" workaround for the serial log. The only log message that is trimmed is the overly verbose loopstats. All other messages are intact. Works with 9600 baud too, which is a worst case condition for filling the Tx buffer.

But TD-er will make this workaround obsolete, so no need to post it.

For now I'm going to flash the modified log code into both of my NodeMCU's and let them run for a few hours. I'm doubt it's eliminated all sources of the crash/reboots, but I expect an improvement.

- Thomas

TD-er, 2018-08-09:
The web log view isn't perfect and has a few technicalities to deal with, the serial port doesn't.
For example, requesting a JSON file containing the log takes between 50 and a few 100 msec, mainly depending on the load of the ESP and the size to output (and CPU-power, network connectivity and presence of virusscanner on the viewing side, the browser)
This means the log on the web-side (spelling with a 'd' is intended) of things can only be updated every now and then, which will lead to lost log lines.
This output is per N lines.

The log output on the serial port can update per character and continue as soon as the buffer has a character free to write to. Well you may want to wait to do a bit more in bursts to be non blocking, but you get the idea.
So I am convinced the same buffer will perform better on the serial output than it does on the web-viewer.

s0170071, 2018-08-09:
The cursor comes back of you toggle the preview once again.

TD-er, 2018-08-09:
The log buffer is now working using N lines of M characters (fixed length)
N is depending on the build type (normal/dev/testing/ESP32).
So it is wasting a lot of memory on this, due to its fixed size.

My ideas to improve this buffer:
I will make it either using a circular buffer, or something more dynamic.
Messages will be kept up to 2 or 3 seconds, or mainly limited by some max. amount of memory used and the log-reader (web/serial) will have to send the timestamp of the last read log line. This allows lost requests, which happen sometimes on the web log viewer.
And you can have more log readers, like a second web client connected, or even new output methods like MQTT. There is already a syslog, but that one also cannot keep up at some times. So that one may also benefit from such a new log buffer.

The bursty statistics should not be sent via the log buffer, but rather generated on the fly when queried. The data structure containing all timings is much smaller than the lines themselves.
Output via JSON on the web interface should be much more memory efficient.

thomastech, 2018-08-09:
@s0170071: Thanks for the lost cursor tip.

@TD-er: You ideas sound amazing! If you are looking for a show of hands, I like circular buffers. Simple and efficient. But it's your baby, so choose what you feel is best.

- Thomas

thomastech, 2018-08-09:
> The log buffer object also needs an index to keep track of the start of each log entry and store the timestamp for easy access.
> So it is already a bit more complicated than just a read and a write pointer.
I was thinking that you would store the fully annotated log message in the circular buffer. That is, store the entire line, like this:
```
    String msg = String(millis());
    msg += F(" : ");
    msg += line;         // Log Message String
    msg += F("\n");

```
There's no need to keep track of anything special. All the stored log data in the circular buffer will be formatted and ready for print.

`I am currently looking into the MessageDelay variable.`

The MessageDelay has come up in your other discussions. More peeling the onion, more fixes to do.

- Thomas

thomastech, 2018-08-10:
Stability has improved with the temporary serial log fix. So that is a relief.

Over the past 18 hours I've experienced two exception (28) crashes; I don't recall seeing those before. But no more yield panics or watchdog related reboots. Should we close this "Panic" issue ticket and start a new one for the serial log induced crashes? 

Regarding exception 28, the docs say it is LoadProhibitedCause: _A load referenced a page mapped with an attribute that does not permit loads_. So it's a memory related issue, such as invalid pointer.

I installed ESPExceptionDecoder (https://github.com/me-no-dev/EspExceptionDecoder). After a lot of frustration I got it to run. But the decoded stack trace it provides seems to be nonsense. Other users have experienced this too.  So back to square one, looking for suggestions on how to debug it.

- Thomas

s0170071, 2018-08-10:
If you self compiled the source the decoder usually works fine.

TD-er, 2018-08-10:
Such errors may very well be caused by some memory allocation that failed.
For example, when I try to allocate some memory and assign it to a pointer, it may fail to do so if there isn't enough free memory.
When the result of this allocation is not checked (look in the code, it is almost never checked), crashes like these may occur when such an invalid pointer is being used.

That's just one example of what may happen to get exception 28.
It can also be some pointer to an element in a string or vector. But when that container (string/vector) is increased, the pointer may become invalid, since those containers usually allocate a larger block, copy all data over to the new block and delete the old block. Any pointer to that old block has become invalid. It may still point to some of the old data, so it may seem to work OK, but as soon as another allocation is done, that block may be re-used.
If you were just trying to copy a char array, that copy may try to continue until a '0' is found. But with the new data it may be that '0' takes a while and in the meantime you're writing beyond the allocated space of the target memory structure.

These exceptions are the worst, since they are:
- Hard to reproduce
- Hard to get a proper stack trace (the bug may even overwrite code, so you'll end up somewhere else)
- Seem to occur at other (unrelated) locations in the code.

But judging your stack trace, I guess it is a combination of the two I mention above and caused by something in the [parseTemplate](https://github.com/letscontrolit/ESPEasy/issues/1605) function.

thomastech, 2018-08-10:
@TD-er: Thanks for your insight into my stack trace. Seems like with each incremental success, something else comes along that creates a detour from achieving stable operation.
- Thomas

[29 more replies]

closed 2019-10-10 by TD-er