letscontrolit/ESPEasy #1628: [SDS011] No longer works with new scheduler

TD-er, 2018-08-04:
There was a report on the [forum](https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=5645) about this.
Just to check, I loaded the last build on my (very stable) node running SDS011 sensor.
And now it does not update anymore like it should.

TD-er, 2018-08-05:
I've been running it for a while now.

```
Aug  4 22:21:03 SDS011_dust EspEasy: SDS  : act 3.70 4.00
Aug  4 22:31:03 SDS011_dust EspEasy: SDS  : act 4.60 6.20
Aug  4 22:36:03 SDS011_dust EspEasy: SDS  : act 4.50 5.20
Aug  4 22:41:03 SDS011_dust EspEasy: SDS  : act 5.20 6.20
Aug  4 22:46:03 SDS011_dust EspEasy: SDS  : act 5.40 7.40
Aug  4 23:01:03 SDS011_dust EspEasy: SDS  : act 6.20 7.40
Aug  4 23:06:03 SDS011_dust EspEasy: SDS  : act 5.90 6.60
Aug  4 23:16:03 SDS011_dust EspEasy: SDS  : act 6.50 7.50
Aug  4 23:21:03 SDS011_dust EspEasy: SDS  : act 6.30 7.30
Aug  4 23:26:03 SDS011_dust EspEasy: SDS  : act 6.10 8.00
Aug  4 23:31:03 SDS011_dust EspEasy: SDS  : act 6.30 7.40
Aug  4 23:36:03 SDS011_dust EspEasy: SDS  : act 6.60 10.20
Aug  4 23:41:03 SDS011_dust EspEasy: SDS  : act 5.90 8.10
Aug  4 23:46:03 SDS011_dust EspEasy: SDS  : act 6.00 6.60
Aug  4 23:51:03 SDS011_dust EspEasy: SDS  : act 5.50 7.20
Aug  4 23:56:03 SDS011_dust EspEasy: SDS  : act 5.40 7.00
Aug  5 00:11:02 SDS011_dust EspEasy: SDS  : act 3.80 9.10
Aug  5 00:16:02 SDS011_dust EspEasy: SDS  : act 3.40 5.40
Aug  5 00:26:02 SDS011_dust EspEasy: SDS  : act 3.50 6.80
Aug  5 00:31:02 SDS011_dust EspEasy: SDS  : act 3.30 6.30
Aug  5 00:36:02 SDS011_dust EspEasy: SDS  : act 3.00 6.00
Aug  5 00:41:02 SDS011_dust EspEasy: SDS  : act 3.40 8.10
Aug  5 00:51:02 SDS011_dust EspEasy: SDS  : act 3.10 6.10
Aug  5 01:01:02 SDS011_dust EspEasy: SDS  : act 4.10 7.90
Aug  5 01:11:02 SDS011_dust EspEasy: SDS  : act 2.70 6.90
Aug  5 01:16:02 SDS011_dust EspEasy: SDS  : act 2.70 8.50
Aug  5 01:21:02 SDS011_dust EspEasy: SDS  : act 3.40 7.70
Aug  5 01:31:02 SDS011_dust EspEasy: SDS  : act 2.90 8.80
Aug  5 01:36:02 SDS011_dust EspEasy: SDS  : act 2.70 8.00
Aug  5 01:41:02 SDS011_dust EspEasy: SDS  : act 2.70 7.00
Aug  5 01:46:02 SDS011_dust EspEasy: SDS  : act 2.70 8.90
Aug  5 01:51:02 SDS011_dust EspEasy: SDS  : act 3.00 9.30
Aug  5 01:56:02 SDS011_dust EspEasy: SDS  : act 2.40 5.10
Aug  5 02:01:02 SDS011_dust EspEasy: SDS  : act 2.30 5.90
Aug  5 02:11:02 SDS011_dust EspEasy: SDS  : act 2.40 6.30
Aug  5 02:21:02 SDS011_dust EspEasy: SDS  : act 2.30 5.80
Aug  5 02:26:02 SDS011_dust EspEasy: SDS  : act 2.20 8.00
Aug  5 02:36:02 SDS011_dust EspEasy: SDS  : act 2.20 6.50
Aug  5 02:41:02 SDS011_dust EspEasy: SDS  : act 2.00 5.40
Aug  5 02:46:02 SDS011_dust EspEasy: SDS  : act 2.40 9.90
Aug  5 02:51:01 SDS011_dust EspEasy: SDS  : act 2.40 6.90
Aug  5 02:56:01 SDS011_dust EspEasy: SDS  : act 2.10 5.80
```

It skips a 'beat' every now and then and the start to get the first sample took about 30 minutes.
It could be the samples are a bit mangled every now and then due to missing the first (few) bits in the burst, since the ESP8266 doesn't have proper interrupts on the software serial pins.

Still it takes a long time to get synchronized.

joggelxx, 2018-08-26:
I'm having the same issue with my SDS011 on mega-20180818. It is set to 5 min sleep, but I'm frequently getting only measurements every 10 min and sometimes even 15 min. BME280 is running fine.

Can you tell me which release you used before that was stable? Thanks.

Grovkillen, 2018-08-26:
You should try the latest release first.

TD-er, 2018-08-26:
@Grovkillen, the new controller handling is not yet merged. ;)

You could try to set the plugin interval to 30 or 60 seconds.
It will only update the values when a new value is received.

Grovkillen, 2018-08-26:
@TD-er Ah! Sorry :relieved:

joggelxx, 2018-08-26:
Running now with mega-20180826 since a couple of hours, but as expected still the same behavior.
The interval is already set to 60s, I will give 30s a try. 
Anyway, good to know something is coming to address this :smiley:.

st3fan87, 2018-08-27:
Im facing the same problem. Random measurements. data receivend in 5min,5min,10min,5min,15min,5min,5min etc :)

closed 2023-09-04 by TD-er