letscontrolit/ESPEasy #1629: PCF8574 MCP23017 LCD no Text

jannnfe, 2018-08-05:
### Summarize of the problem/feature request

No text will be displayed on the LCD

### Expected behavior

Text should displayed

### Actual behavior

No text will be displayed on the LCD

### System configuration

Hardware: Wemos D1 Mini & PCF8574 MCP23017 LCD
[image]
[image]

ESP Easy version: mega-20180804

ESP Easy settings/screenshots: 
[image]
[image]
[image]

ghtester, 2018-08-05:
First make sure you are using the correct I2C address. I believe it should be 0x3F in your case. Did you try Tools - I2C Scan menu option?

TD-er, 2018-08-05:
Next step is checking the contrast of the display module.
In the picture, I do not see a faint texture of the pixels.
Even when set to "white" the pixels should be slightly visible. 
If not, then the contrast is probably incorrect.

ghtester, 2018-08-05:
Don't believe to I2C Scan too much :-)  and please try to set 0x3F address, then reboot your device.

jannnfe, 2018-08-05:
Thanks @TD-er and @ghtester 
The Contrast was bad. Now it works.
Sorry für my "issue" i was thinking I was doing everything right.
Thanks and have a nice day :)

ghtester, 2018-08-05:
No problem, thanks for the feedback, good to know it works. :-)
Have a nice day too and please close the issue.

closed 2018-08-05 by jannnfe