letscontrolit/ESPEasy #1632: Nightly build of ESP32: analog input plugin is missing

tobiasfaust, 2018-08-06:
Hi,
in current nightly build bin file of ESP32 i miss the analog input plugin. 
At ESP32 Espeasy page this plugin is listed as base plugin (https://www.letscontrolit.com/wiki/index.php/ESPEasy32)
Somthing went wrong?
Its possible to integrate the testing bin file in nightly build of ESP32 too?

TD-er, 2018-08-07:
I just added a few plugins. The wiki is maybe reflecting the first build made by @mvdbro 
The current build is based on my attempt to make the ESP32 be included in the nightly builds to see compiler issues sooner and start testing it a bit.
So I grabbed just a few plugins that would compile.

I will check if the analog plugin is working and whether it needs something to change VCC mode.
If it is working, it will be included in the builds.

tobiasfaust, 2018-08-08:
Thanks, i´m waiting for you.  I'm curious about it.
So i would be able to get VOC (room air quality) of my TGS2600 Sensor :)

TD-er, 2018-08-08:
Looks like that sensor may output up to 5V.
So you may want to look into using resistors to get that voltage into range of the A/D converter of the ESP.
For ESP8266 this is 0...1 V. You have to look into the datasheet of the ESP32 to verify its range.
The ADC of the ESP8266 is not really accurate and some boards already have resistors added. These resistors may have impact on your own set of resistors to get the voltage down. (behavior may be non linear due to these extra resistors)

I think it is worth to try, but you'll probably end up with using an external A/D converter. :)

tobiasfaust, 2018-08-08:
the ESP32 takes up to 3.3V.
For my TGS Sensor i have a resistor network of 68K/100k to reduce the level to 3.3V. The tests with my own sketch shows a good and a stable level for the same VOC
It would be nice if the output of analog plugin shows at least the measured voltage 0 .. 3.3V :)

TD-er, 2018-08-08:
On the ADC input side?
It is possible, since the ADC output ranges from 0... 4096.

tobiasfaust, 2018-08-08:
Hi,
quote from esp32 Manual:

> 4.1.2 Analog-to-Digital Converter (ADC)
> ESP32 integrates 12-bit SAR ADCs and supports measurements on 18 channels (analog-enabled pins). The ULP coprocessor in ESP32 is also designed to measure voltage, while operating in the sleep mode, which enables low-power consumption. The CPU can be woken up by a threshold setting and/or via other triggers. With appropriate settings, the ADCs can be configured to measure voltage on 18 pins maximum.

> When atten=3 and the measurement result is above 3,000 (voltage at approx. 2,450 mV), the ADC accuracy will be worse than described in the table above.

> • To get better DNL results, users can take multiple sampling tests with a filter, or calculate the average value. By default, there are ±6% differences in measured results between chips. ESP-IDF provides couple of calibration methods for ADC1. Results after calibration using eFuse Vref value are shown in Table7. For higher accuracy, users may apply other calibration methods provided in ESP-IDF, or implement their own.

Source: https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf

TD-er, 2018-08-08:
Hmm, apparently the ADC implementation on ESP32 is completely different from ESP8266.
Then it is probably not just a matter of enabling the plugin for that build.
You also should select the pin to be used and set some state to that pin definition.

tobiasfaust, 2018-08-08:
below, this is my single sketch-tab to get VOC value.
For espeasy its sufficient to get plain VCC. Mabe its could help?

#include <driver/adc.h>

#define TGS2600_PIN 35
#define VOC_ADC_CHANNEL ADC1_CHANNEL_7

#define TGS2600_DEFAULTPPM         10             //default ppm of CO2 for calibration
long    TGS2600_DEFAULTRO        = 45000;         //default Ro for TGS2600_DEFAULTPPM ppm of CO2
#define TGS2600_SCALINGFACTOR      0.3555567714   //CO2 gas value
#define TGS2600_EXPONENT           -3.337882361   //CO2 gas value
#define TGS2600_MAXRSRO            2.428          //for CO2
#define TGS2600_MINRSRO            0.358          //for CO2
float TGS2600_RL = 5000;                         // Load Resistance

boolean tgs2600_setup() {
  pinMode(TGS2600_PIN, INPUT_PULLUP);
  adc1_config_width(ADC_WIDTH_12Bit);
  adc1_config_channel_atten(VOC_ADC_CHANNEL, ADC_ATTEN_11db); //11db for 3.3V  
  
  int analogVal = analogRead(TGS2600_PIN);
  dbSerialPrint("VOC raw value: "); dbSerialPrintln(analogVal); 
  
  if (analogVal < 4095 && analogVal > 0) {
    TGS2600IsPresent = true;
    dbSerialPrintln("...gefunden: VOC Sensor TGS2600");
  } else {
    TGS2600IsPresent = false;
    dbSerialPrintln("...NICHT gefunden: VOC Sensor TGS2600");
  }
  return TGS2600IsPresent;
}

float calcSensor_VOC(int valr)
{
  float val;
  float TGS2600_ro;
  float val_voc;

  //Serial.println(valr);
  if (valr > 0) {
    val =  ((float)TGS2600_RL * (1024 - (valr/4)) / (valr/4)); //10bit = 1024, 12bit= 4096
    TGS2600_ro = TGS2600_getro(val, TGS2600_DEFAULTPPM);
    //convert to ppm (using default ro)
    val_voc = TGS2600_getppm(val, TGS2600_DEFAULTRO);
  }
  
    Serial.print ( F("Vrl / Rs / ratio:"));
    Serial.print ( val);
    Serial.print ( F(" / "));
    Serial.print ( TGS2600_ro);
    Serial.print ( F(" / "));
    Serial.println (val_voc);
  return val_voc;
}

/*
   get the calibrated ro based upon read resistance, and a know ppm
*/
long TGS2600_getro(float resvalue, float ppm) {
  return (long)(resvalue * exp( log(TGS2600_SCALINGFACTOR / ppm) / TGS2600_EXPONENT ));
}

/*
   get the ppm concentration
*/
float TGS2600_getppm(long resvalue, long ro) {
  float ret = 0;
  ret = (float)TGS2600_SCALINGFACTOR * pow(((float)resvalue / ro), TGS2600_EXPONENT);
  return ret;
}

void loopVocRead() {
  if (TGS2600IsPresent) {
    char msg[50];
    char topic[50];
    
    int analogVal = analogRead(TGS2600_PIN);
    Serial.print(F("VOC analogVal: "));
    Serial.println(analogVal);

    Serial.print(F("VOC ppm: "));
    Serial.println(calcSensor_VOC(analogVal));

    snprintf (msg, sizeof(msg), "%.2f", calcSensor_VOC(analogVal));
    snprintf (topic, sizeof(topic), "%s%s", MQTT_ROOT.c_str(), MQTT_GetVOC);
    MQTTClient.publish(topic, msg);
  }
}

redskinhu, 2019-01-16:
Hi

I'm using the ESP32 ADC with self compiled fw (measuring VCC and battery voltage). Just need to change 2 values in the _P002_ADC.ino before compile.

Those 1023 needs to be change to 4095 (ESP8266 10bit vs ESP32 12bit ADC), after you can calibrate on the ESPEasy webpage the whole range 0-4095 not just 0-1023.  And it works like a charm.

```
        addFormNumericBox(F("Point 1"), F("p002_adc1"), PCONFIG_LONG(0), 0, 1023);
        html_add_estimate_symbol();
        addTextBox(F("p002_out1"), String(PCONFIG_FLOAT(0), 3), 10);

        addFormNumericBox(F("Point 2"), F("p002_adc2"), PCONFIG_LONG(1), 0, 1023);
```
T

TD-er, 2019-01-17:
@redskinhu Thanks, I will add it, so it will not be forgotten :)