letscontrolit/ESPEasy #1633: [PMS003ST] Add support for new version of Plantower dust sensor

TD-er, 2018-08-08:
See PR at playground: https://github.com/letscontrolit/ESPEasyPluginPlayground/pull/104

> existing P053 is PMSx003, have PM2.5 PM1.0 PM10 three parameter, all only for dust.
> 
> this new sensor is PMSx003ST, Upgrade of PMSx003. it have PM2.5 TEMP HUMI HCHO four parameter. Can detect dust, temperature, humidity and formaldehyde.

(set project to "Broken sensors" to keep track of it)

remyderuysscher, 2019-06-29:
What is the status of this merge? I would like to test it but it's still not merged in playground.

TD-er, 2019-06-29:
I have not looked at it (sadly), but the PR has been merged for the playground.

Stalkeros000, 2021-07-16:
Hello. I received a PMS5003ST sensor. Can't I connect it still? To show temperature, humidity and formaldehyde parameters? 
I installed the latest version (03.05) but I do not see this sensor in the selection list.
Thank you.

tonhuisman, 2021-07-17:
There is a plugin PMSx003 available in all Normal, Test and Max builds, but I'm not sure it supports all parameters you mention. Can you give that a try?

tonhuisman, 2021-07-17:
Did some more investigation, but this plugin won't accept readings from the `T` or `ST` models, as they send out a larger 'frame' of data, and the size of that frame is checked during data read.

To avoid code duplication, (and the, nearly identical, playground plugins _P216_PMS5003ST/_P216_PMSx003ST need a big overhaul to work as intended with current ESPEasy anyway), the current P053 plugin should be extended to add support for the `T` and `ST` models, by adding extra options in the settings, including a selection of the desired `Values`, as ESPEasy only allows for 4 values, and these sensors provide up to 6. This will take some work before it's available.

tonhuisman, 2021-07-17:
@Stalkeros000 I created a PR to add support for the PMS5003ST to plugin P053 (PMSx003).

Can you please test with your sensor, and report your findings here, I can provide a .bin file if needed.

Stalkeros000, 2021-07-19:
>  I can provide a .bin file if needed.

Yes that woud be great, coz Im newbie and I trying to compile in VScode and Atom and got error. Thanks.

tonhuisman, 2021-07-19:
What .bin file do you use on your ESP? There are too many choices to 'just pick one', and the entire build is ~540 MB 😉

Stalkeros000, 2021-07-19:
Last time I using normal_esp8266_4M1M.

tonhuisman, 2021-07-19:
This is the NORMAL build, as downloaded from the GitHub actions page for the referenced pull request:
[ESPEasy_normal_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6843056/ESPEasy_normal_ESP8266_4M1M.zip)

You can use the Update Firmware button on the Tools page to update (unzip the downloaded file first). After the upload is successful it will take ~20 seconds before the ESP will reboot, as it's working on an internal move of the data, so don't reset it during that process! The browser page will refresh in due time.

Stalkeros000, 2021-07-21:
All work great, thank you so much!
[image]
 Sorry my question, I cant see all data together in that window (temp, humi, all PM and formaldehyde)?

tonhuisman, 2021-07-21:
ESPEasy only supports max. 4 measurements to be displayed, that's why I added an option to select 2 subsets of these values, with 1 of the sets representing the values that where available before.
You can store the not displayed values by enabling the Events option, and add 2 eventhandlers in the rules to store these values in a Dummy device.

Stalkeros000, 2021-07-22:
Rename values. thanks. In the next release that functionality will be default (HCHO)?

tonhuisman, 2021-07-22:
I'm more in favor of keeping backward compatibility as much as possible, so I would like to leave that to the user to update according to the device and settings used. I'll make a clear note in the documentation (see below) about that.
The HCHO term comes from the sensor documentation and the plugin sources found in the PluginPlayground, you are free to use the Formaldehyde value name. 👍 

I'll create the documentation (based on the ESPEasy Wiki documentation, with updated screenshots, including the new options, of course), after which the PR is ready to be merged.

PR #3707 (2021-07-31): [P053] Add support for PMS5003ST, PMS2003 and PMS3003
- Add support for the extra measurement values available in the PMS5003ST
- Add support for the slightly different data format of PMS2003 and PMS3003
- Add option to select the output values (only applicable for PMS5003ST)
- Add option to send event for values not simultaneously available for PMS5003ST

Resolves #1633
Resolves #2002

NB: No sensors available here, so testing should be done by others, please.
NB2: These extensions are disabled on 1M builds to avoid exceeding the max. build size for these modules.

TODO:
- [x] Testing (extern [... cut]
closed 2021-07-31 by TD-er