letscontrolit/ESPEasy #1642: ESP always starts again [mega-20180809]

Spohky, 2018-08-11:
Hello everybody,

Today I wanted to test my Nextion Display and tested the "Dev" and "Test" versions.
The ESP8266 will work for a few minutes and then it will restart forever.
I also tested it with the normal version, so that's the same problem.
I also tested 4 more ESP chips.
With the version go my ESP chips: mega-20180324

Serial I get the following return:
` ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
   ▒U107 :

INIT : Booting version: mega-20180809 (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
108 : INIT : Warm boot #75 - Restart Reason: Power on
112 : FS   : Mounting...
119 : FS   : Mount successful, used 75802 bytes of 113201
546 : CRC  : program checksum       ...OK
556 : CRC  : SecuritySettings CRC   ...OK
576 : INIT : Free RAM:21152
577 : INIT : I2C
577 : INIT : SPI not enabled
588 : NEXTION075 : serial pin config RX:14, TX:12, Plugin Enabled
588 : NEXTION075 : Using software serial
600 : INFO : Plugins: 71 [Normal] [Testing] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
601 : EVENT: System#Wake
606 : WIFI : Set WiFi to STA
639 : WIFI : Connecting NicWiFi attempt #0
653 : EVENT: System#Boot`

I hope you can help me.
Greetings Spohky

TD-er, 2018-08-11:
It looks like the issues reported in this issue:
https://github.com/letscontrolit/ESPEasy/issues/1625

Only thing is that it resets due to watchdog timer reset, but the restart reason is: "Power on".
Are you sure the power supply is sufficient?

What plugins and controllers do you have active?
And a test, could you disable serial log?

Spohky, 2018-08-11:
Well, I have tested this with my ESP8266, if I use "mega-20180324" then my ESP chips will work.
If I use one of the more recent versions (August), then it is only enough if I restart the ESP after successful WiFi connection.
Then he always starts again.

Serial Log is already stored as "none".
I did not set up the ESP, I just programmed it, connected to WiFi and then restarted.

TD-er, 2018-08-11:
Hmm, is the Nextion connected to the ESP?

Spohky, 2018-08-11:
I do not build it myself, I use it from:
https://github.com/letscontrolit/ESPEasy/releases

I have tried the "Dev", "Test" and "Normal" version of:
8.9. // 04.08. // 02.08. tries.

The Nextion is currently not connected, I use a wemos d1.

Spohky, 2018-08-12:
So, I have tested it once individually.
Always changed something and then restarted the ESP.

I noticed now, as soon as I put under "Hardware"> "Wifi Status LED" on "GPIO-2", then the ESP does not start anymore.

I did not do it and test if it works.

Spohky, 2018-08-12:
Since I left the "Wifi Status LED" on "none" it works without problems.
In addition, I also use "OpenHAB MQTT" without problems.

closed 2019-10-27 by TD-er