letscontrolit/ESPEasy #1644: MQTT Import Error

anthonyhou04, 2018-08-13:
Hi,
My MQTT Broker disconnects a few minutes when using MQTT Import

MQTT Broker log:
Socket error on client ESP_Easy-Import     , disconnecting.

No special log in ESPEasy

TD-er, 2018-08-13:
Can you please give some more information, like:
- version of ESPeasy used
- Build yourself, or nightly build
- Some log of ESPeasy showing MQTT status and WiFi connection status
- Some configuration of the MQTT import.
- Do you actually use the values of that import?
- What "interval time" is set at the MQTT import plugin?
- Some timing information on when it connects and when a disconnect occurs (and when reconnect)

With MQTT it is "normal behavior" of the broker to disconnect the client if there has been no message from the client in the last N seconds. N is often 10 seconds + 50% (= 15 seconds) when using a default config of Mosquito.

The MQTT controllers have some checks running every now and then to call the broker, to avoid these issues. But the MQTT import plugin has not.
So you have to set the interval timer of that plugin rather low, like 10 seconds or less.

anthonyhou04, 2018-08-13:
Hi,
I am using the latest version: Release mega-20180809
I am not sure about the config of MQTT import
My MQTT controller message Interval is 100ms, I am not sure about how it would effect MQTT import.

TD-er, 2018-08-13:
I've just looked into the MQTT import page and indeed, there is no "Interval" setting there, like there is on (almost all?) other plugin pages.
I have to look into the source to see what may be the problem here.

If you could answer the other points I mentioned as much as possible, then that would be helpful.

Grovkillen, 2018-08-13:
@TD-er I think he's refereeing to the "message interval" found in advanced settings.

Grovkillen, 2018-10-23:
@TD-er  we can close this one right? If not then reopen it.

closed 2018-10-23 by Grovkillen