letscontrolit/ESPEasy #1646: HTTP commands gives errors

berbergh, 2018-08-13:
### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I use HTTP calls from Domoticz into ESPEasy to send data to OLED or NEXTION.
Examples:
http://xxx.xxx.xxx.xxx/control?cmd=NEXTION,page0.vLight.val=1
http://xxx.xxx.xxx.xxx/control?cmd=OLED,clear

### Expected behavior

I expect the command to be carried out, and it does, but I get an error message in domoticz.
However, when I send http://xxx.xxx.xxx.xxx/control?cmd=GPIO,13,0 I do not get an error message.
When I issue the latter command in the brower, I get { "log": "GPIO 12 Set to 0", "plugin": 1, "pin": 12, "mode": "output", "state": 0 }  in response.
With the OLED and NEXTION commands, in the browser, I get no response.

### Actual behavior

2018-08-13 19:21:11.724 Error: Error opening url: http://192.168.0.76/control?cmd=NEXTION,page7.t1.txt="0"

### Steps to reproduce

1. Make an ESPEasy with build ESPEasy_mega-20180809 (I also tried ESPEasy-2.0-20180116, same result)
2. Connect an OLED display and in the browser enter http://xxx.xxx.xxx.xxx/control?cmd=OLED,clear. See the result. Try the same from a switch in domoticz.
3. Also enter in the browser http://xxx.xxx.xxx.xxx/control?cmd=GPIO,13,0 and see the result. Try the same from a switch in domoticz

### System configuration

Hardware: WeMos or NodeMCU

ESP Easy version: 
ESPEasy_mega-20180809 (I also tried ESPEasy-2.0-20180116

ESP Easy settings/screenshots: 
[image]
[image]
[image]
[image]
[image]

### Rules or log data

TD-er, 2018-08-13:
Note that not all commands are made accessible via http, for security reasons.
Can you try the command via the command prompt on the tools page? Or via rules?
Does that command work like expected, then the command is not exposed via http.

berbergh, 2018-08-13:
So, I enter http://xxx.xxx.xxx.xxx/control?cmd=NEXTION,page0.vFont.val=1
In the Command box in Tools. Command Output is Unknown command.

But, when I issue http://xxx.xxx.xxx.xxx/control?cmd=GPIO,12,0 it also says that.

In the mean time, issued from the browser, both work fine, although the NEXTION one gives no reponse.

Verzonden vanuit Mail voor Windows 10

Van: Gijs Noorlander
Verzonden: maandag 13 augustus 2018 19:52
Aan: letscontrolit/ESPEasy
CC: berbergh; Author
Onderwerp: Re: [letscontrolit/ESPEasy] HTTP commands gives errors (#1646)

Note that not all commands are made accessible via http, for security reasons.
Can you try the command via the command prompt on the tools page? Or via rules?
Does that command work like expected, then the command is not exposed via http.

berbergh, 2018-08-13:
The problem was caused by the plugins not sending the reply.
I think I can call it SOLVED

closed 2018-08-13 by berbergh