letscontrolit/ESPEasy #1647: %rssi% variable doesnt work any more in rules

jimmys01, 2018-08-13:


TD-er, 2018-08-13:
The parsing of the string does work. I just tested it on my test node with an OLED Framed plugin running.
It may still be it is no longer working in rules, which can be because of 2 reasons:

- negative values are no longer working in rules
- string replacement is not done at the right moment anymore.

If the second reason applies, then there should be other similar ones that no longer work in rules.
For example `%sysheap%` which provides a positive value.

jimmys01, 2018-08-13:
Will post sample code tomorrow

thomastech, 2018-08-13:
Have you tried the solution discussed here:
https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=5133&start=10

There is another solution. You can use Generic System Info device, Wifi RSSI. Name: RSSI, value: signal. 
In rules you can add your actions, such as this MQTT:
```
on RSSI#signal do
  Publish /%sysname%/RSSI/signal,[RSSI#signal]
endon
```
- Thomas

jimmys01, 2018-08-13:
with ESP_Easy_mega-20180625_normal_ESP8266_4096 this code works.

TaskValueSet 12,1,%rssi%
  if [Dummy#RSSI]<=-100
   TaskValueSet 12,1,0
  else
  if [Dummy#RSSI]>=-50
   TaskValueSet 12,1,100
  else
   TaskValueSet 12,1,[Dummy#RSSI]+100
TaskValueSet 12,1,[Dummy#RSSI]*2
  endif
endif

in the latest release it does not.

TD-er, 2018-08-13:
Could you try it with this build I made here to test?
https://github.com/letscontrolit/ESPEasy/pull/1641#issuecomment-412307113

jimmys01, 2018-08-14:
@TD-er the result is the same. 

[RSSI#signal] has also the same effect.
`TaskValueSet 12,1,[RSSI#signal]
  if [Dummy#RSSI]<=-100
   TaskValueSet 12,1,0
  else
  if [Dummy#RSSI]>=-50
   TaskValueSet 12,1,100
  else
   TaskValueSet 12,1,[Dummy#RSSI]+100
TaskValueSet 12,1,[Dummy#RSSI]*2
  endif
endif
Publish ESP_Easy/%sysname%/tele/IP,%ip%
Publish ESP_Easy/%sysname%/tele/STATE,{"Time":"%lcltime%","Uptime":"%uptime%","Vcc":%vcc%,"Wifi":{"AP":1,"SSId":"%ssid%","RSSI":[Dummy#RSSI],"RSSI2":%rssi%,"APMac":"%bssid%"}}
Publish ESP_Easy/%sysname%/stat/ALARM,[Dummy#ALARM]`

The mqtt message that this returns is {"Time":"1970-00-00 00:00:00","Uptime":"0","Vcc":%vcc%,"Wifi":{"AP":1,"SSId":"Automations",**"RSSI":0,"RSSI2":-68**,"APMac":"66:D1:54:A9:05:AC"}}

it used to be 
{"Time":"1970-00-00 00:00:00","Uptime":"0","Vcc":%vcc%,"Wifi":{"AP":1,"SSId":"Automations",**"RSSI":64,"RSSI2":-68,"**APMac":"66:D1:54:A9:05:AC"}}
%rssi% returns a correct value, but the code i described does not work. It acts as it is 0

jimmys01, 2018-08-16:
This aproach works thank you. But also my code was also working without need to do this juggling.

Grovkillen, 2018-10-23:
This is not a problem anymore:

This work:
```
on rules#timer=1 do
taskValueSet,1,1,(-%rssi%)
publish,TEST,[Dummy#Test]
timerSet,1,1
endon
```

And also using Generic - System Information with the formula `-%value%` work just fine.

closed 2018-10-23 by Grovkillen

jimmys01, 2018-10-23:
will this work also? 
taskValueSet,1,1,((-%rssi%)+100)*2

Grovkillen, 2018-10-23:
Yep it runs just fine