letscontrolit/ESPEasy #1650: Browser support

fmuntean, 2018-08-14:
I get reports now that the browser is not supported from latest build.
Safari on the IOS11 and IE11 both report this.
Also the site is not working correctly with these browsers.

The system should fall back on features and/or fully support active browsers that are not EOL.

TD-er, 2018-08-14:
On what pages do you get these reports?
Is it JavaScript related?
Only pages where JavaScript is used are the Devices tab and on the web log viewer.

fmuntean, 2018-08-14:
Build | 20102 - Mega
-- | --
Libraries | ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3

log viewer and the menu to switch between pages is not working

fmuntean, 2018-08-14:
I think is Java script. but I use my phone to monitor and configure the devices.

TD-er, 2018-08-14:
The menu to switch pages is using stuff that's been around for ages.
Only the JavaScript does use some things that may leave out some old browsers, although iOS11 doesn't sound too old I guess.
Maybe something in JavaScript that's now considered less secure?

fmuntean, 2018-08-15:
[image]
I attached a screenshot from my phone. you can see the three lines on the top and can't do anything to move from this page to another page. I have to type the url manualyl for every page.

I have IOS 11.4.1

fmuntean, 2018-08-15:
and here is the image going to the /log url manually 

![image-1 1](https://user-images.githubusercontent.com/1547916/44126866-988a5f5e-a008-11e8-
9b84-738e3df7746b.jpg)

fmuntean, 2018-08-16:
still doesn't work when rotating the device.

Grovkillen, 2018-08-16:
I'm really sorry for not spending time on the GUI. I spend quite some time doing a overhaul of how we handle the HTML code so this will be fixed once that get in place. Is this critical so I should stop that for a while or can it wait until the release of the new GUI.

TD-er, 2018-08-16:
Last build was mainly an upgrade of the core libraries and only that, so any new issues can be related to changes in the core library.
I will change the menu, to be visible always, but try to find a way to show only small buttons when used on a small screen.
So I will have to take a look at CSS options on how to do that without having to add a lot of small icons into the code.

Grovkillen, 2018-08-16:
@TD-er thanks for that intermediate fix (if you find it :wink: ) :+1:

Grovkillen, 2018-08-16:
You can remove the hover on that "hamburger" button if you want to make it static.

TD-er, 2018-08-16:
For each tab, we should find an Unicode symbol to represent the tab.
See for example: https://www.w3schools.com/charsets/ref_utf_symbols.asp
If someone has any suggestion for a tab, please add it here.

Grovkillen, 2018-08-16:
Main: &#9751; (BLACK SHOGI PIECE)
Config: &#9881; (GEAR)
Controllers: &#9732; (COMET)
Hardware: &#9875; (ANCHOR)
Devices: &#9878; (SCALES)
Rules: &#9749; (HOT BEVERAGE)
Notifications: &#9873; (BLACK FLAG)
Tools: &#9874; (HAMMER AND PICK)

giddyhup, 2018-08-16:
The hamburger button is not working for me with Safari under iOS 11.4 (iPhone). Under iOS 12 (public beta 6) on an iPad I get the desktop version of the interface which is working.

TD-er, 2018-08-16:
A lot of the symbols suggested by @Grovkillen showed as a square in Notepad++, so I guess they will also show up as such in older browsers.

Even some of the other types I used below here, are not shown in Notepad++, so there is an alternative given.

If someone with the browsers failing to open the hamburger icon, could report what is shown below, that would be nice. (screenshot preferable) 

Main: ⌂  (HOUSE)
https://www.charbase.com/2302-unicode-house

Config:  ⚙  (Gear)
https://www.charbase.com/2699-unicode-gear
Alternative:  🔑 (Key)
https://www.charbase.com/1f511-unicode-key

Controller:  💬  (Speech Balloon)
https://www.charbase.com/1f4ac-unicode-speech-balloon

Hardware:  📌(Push Pin)  (Hardware is about pin pull-up config and other pin related stuff)
https://www.charbase.com/1f4cc-unicode-pushpin

Devices:  🔌 (Electrical Plug, page is about "plugins")
https://www.charbase.com/1f50c-unicode-electric-plug

Rules:  ⧴  (Rule-Delayed, looks a bit like programming)
https://www.charbase.com/29f4-unicode-rule-delayed
Alternative: 📐 (Triangular Ruler)
https://www.charbase.com/1f4d0-unicode-triangular-ruler

Notifications:  📧  (E-MAIL)
https://www.charbase.com/1f4e7-unicode-e-mail-symbol
Alternative: ✉ (envelope)
https://www.charbase.com/2709-unicode-envelope

Tools:  🔧  (Wrench)
https://www.charbase.com/1f527-unicode-wrench

TD-er, 2018-08-16:
[image]

[image]

Layout with the icons is a it wobbly, but I will fix that later.

Grovkillen, 2018-08-17:
Nice fix @TD-er ! :)

TD-er, 2018-08-17:
A build was not made last night, for whatever reason, I am running the build script now on my own laptop, so people can test.

TD-er, 2018-08-17:
Build is [here](https://www.dropbox.com/s/eewg6pilcc7jyon/ESPEasy_mega-20180815-2-gb63c375.zip?dl=0) , made on my laptop.

Grovkillen, 2018-08-17:
I feel this is a solid fix. @fmuntean Please report back here and I can close it.

TD-er, 2018-08-17:
Also if better B/W utf icons are available, please suggest them here.

fmuntean, 2018-08-19:
what is the fix ? can you show me the source code ?
Can you point me to the GIT location? I have to merge those into my code as I have changed some of the settings to match my design.

TD-er, 2018-08-19:
These are the changes:
https://github.com/letscontrolit/ESPEasy/pull/1657/files

And it is included in the last released nightly build also.
Could you please also test the official build, just to make sure _that_ one is working like expected.
Apparently you made some changes and I want to make sure the 'official' version does not have these issues.
If you can confirm this is working, I can complete the layout to make sure all items are aligned (they are now a bit jumping around) and maybe try to find more matching icon styles.

giddyhup, 2018-08-19:
New icons work with Safari under iOS 11.4 (iPhone).

Grovkillen, 2018-08-19:
Okay, thanks for that feedback @giddyhup

closed 2018-08-19 by Grovkillen

fmuntean, 2018-08-20:
The changes I made are around the number of devices supported and number of controllers. Thus trying to switch between 'official' version and the 'custom' one messes up the config. I will have to dig up another ESP8266 board where I can test the 'official' version

fmuntean, 2018-08-20:
Merged the code and is working now in the phone browser in both orientations. Thank you.