letscontrolit/ESPEasy #1651: [NTP] Can't sync time with NTP

maxx-ukoo, 2018-08-15:
ESPEasy can't sync time with "NTP : No reply" error.

I changed in TimeESPeasy.ino
`udp.begin(123);`

to 

`udp.begin(22123);`
Issue solved. 
 Can we use any port in range 1024-... as source or use random port for NTP source?

TD-er, 2018-08-15:
NTP is a standard protocol, why would 22123 work?

maxx-ukoo, 2018-08-15:
I updated source port for UDP packet, destination (123) unchanged.

TD-er, 2018-08-15:
Ah, now I get it.
So anything above some range should work?
Any idea why it would not work anymore?

maxx-ukoo, 2018-08-15:
Internet provider in his network can block services on know IANA management port in range 0-1023
From [https://en.wikipedia.org/wiki/Ephemeral_port](https://en.wikipedia.org/wiki/Ephemeral_port)

> The Internet Assigned Numbers Authority (IANA) suggests the range 49152 to 65535 (215+214 to 216−1) for dynamic or private ports.[1] 
I suggest do the same - use random port number for source.

TD-er, 2018-08-15:
That's a good suggestion.
I can imagine it is also an issue when accessing other services like http, when it is initiated with port 80 as source.

TD-er, 2018-08-31:
This is fixed by the way (about a week ago)

closed 2018-08-31 by TD-er