letscontrolit/ESPEasy #1652: P135 - MQ135 - not integrated into custom build

stoffelll, 2018-08-15:
### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

Hi,
I compiled ESPEasy with extra plugins, like BME680 (P119) and MQ135 (P135). The BME680 is working quite good, but the plugin for the MQ135 seems not be integrated into the build.
At least the sensor is not listed in the sensor dropdown list.

I'm using ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 with Arduino 1.8.5 and 
ESPEasy_mega-20180809 as base.

Has someone any idea what I have to do differently.

### Expected behavior

I copied the _P135_MQ135.ino file into the same folder as the _P119_BME680.ino and therefore I thought it should work.

### Actual behavior

_P135_MQ135.ino seems not to be used

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

clumsy-stefan, 2018-08-15:
did you define `USES_P135` when building?
The P135 plugin seems already be updated to use the new plugin_sets definitions... so you need to have a `#define USES_P135 ` somewhere in your custom.h or compiler-flags...

stoffelll, 2018-08-15:
Hi,
Thanks for your reply.

I modified "define_plugin_sets.h" and added "#define USES_P135" similar like the BME280 plugin, means:
```
#ifndef USES_P135
    #define USES_P135   // MQ135
#endif
```

After compiling it again, it is still missing.
I also tried to use the custom.h, but I failed most likely because I didn't figured out how to do that:

> 3) define USE_CUSTOM_H as a build flags. ie : export PLATFORMIO_BUILD_FLAGS="'-DUSE_CUSTOM_H'" 	

```
//#undef	DEFAULT_NAME
//#define DEFAULT_NAME		"MyEspEasyDevice"			// Enter your device friendly name

//#undef	DEFAULT_SSID
//#define DEFAULT_SSID		"MyHomeSSID"			   // Enter your network SSID

//#undef	DEFAULT_KEY
//#define DEFAULT_KEY			"MySuperSecretPassword"		// Enter your network WPA key

//#undef	DEFAULT_AP_KEY
//#define DEFAULT_AP_KEY		"MyOwnConfigPassword"		// Enter network WPA key for AP (config) mode

#define USES_P135

#ifdef BUILD_GIT
#undef BUILD_GIT
#endif

#define BUILD_GIT "MyBuildNum"
```

Thanks for your patience and your support!

TD-er, 2018-08-16:
Where did you add it in the `define_plugin_sets.h` ?
I suggest adding it here:
https://github.com/letscontrolit/ESPEasy/blob/306cbf3107d8b970ffb8a786ebf4dbdff34ccd26/src/define_plugin_sets.h#L335-L391

clumsy-stefan, 2018-08-16:
@stoffelll: you added it in the PLUGIN_SET_ONLY_TEMP_HUM section, which I geuss is not included by default!

depending if you do a stable, test or devel build you need to add it in different seciotns.. try to add it in the PLUGIN_SET_STABLE section (lines 335-390) as @TD-er suggested, otherwise it won't get included if you don't change the build type...

or add a -DCUSTOM_H to the compiler flags and define your own custom.h set ;)

stoffelll, 2018-08-16:
Hi,
I tried to add only what TD-er recommended. But it is still not working.
Currently I'm using Arduino framework. Tried now plattform IO as well, but I have some issues and cannot compile at all.

Can someone please test it with Arduino, if he can include the P135 in the compilation?

Thanks!

clumsy-stefan, 2018-08-17:
@stoffelll: try to remove the first and last line in the P135 plugin code (#ifdef USES_P135 / #endif). If this is not working you seem to have a different issue or not including the file.

In Arduino framework, did you a "sketch -> add file -> select the P135" so it's included in the build at all?

stoffelll, 2018-08-17:
@clumsy-stefan: Thanks for your hint, which I tried already. But it doesn't worked as well.
Also adding the P135 via sketch->add file failed.

In the meanwhile I was able to get a build from plattform IO and now it worked immediately.
Strange ...but finally it worked at least with PIO.

@All: Thanks for your support!

closed 2018-08-17 by stoffelll

clumsy-stefan, 2018-08-17:
@stoffelll: good to hear it works... as I also use Arduino framework this is really strange... obviously you need to do both, add it via "sketch->add file"  AND do the correct #defines...