letscontrolit/ESPEasy #1653: PWM flicker is back again with new core 2.4.2

DittelHome, 2018-08-15:
With latest build pwm flicker bug is back.
ESP_Easy_mega-20180815_normal_ESP8266_4096.bin

It seems that with new core (2.4.2) pwm is not working correctly. With lower pwm values you get a flicker.
Same as with core version 2.3. With Core 2.4.1 no flickering.
Info says 0000000 instead of 2.4.2:
Libraries:  ESP82xx Core 00000000, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3

DittelHome, 2018-10-13:
Only for clarification:
PWM is working with core 2.4.1
PWM is NOT working with core 2.4.2

uzi18, 2018-10-13:
is it possible for you to measure pwm frequency on 2.4.1 and 2.4.2?

Domosapiens, 2018-10-14:
Will do on Monday

Domosapiens, 2018-10-15:
@uzi18 
Mega-20181011=core 2.4.1: mean freq 998Hz, min freq 994Hz
Mega-20181001=core 2.4.2: mean freq 980Hz, min freq 768Hz

uzi18, 2018-10-15:
is it possible for You to check also duty cycle on both firmwares?
did you tried to find arduino/esp core issue related to. this?

uzi18, 2018-10-15:
It could be related to https://github.com/esp8266/Arduino/pull/4640 and https://github.com/esp8266/Arduino/pull/4872

Domosapiens, 2018-10-15:
@uzi18 
Duty cycle depends on PWM command, I used PWM,13, 500
Means 500/1023 or around 50%.
Core 2.4.1 measured Std Dev is less than 1%% (meaning the jitter is less then 3%% (3/10th procent!) in 99% of the cases)
Core 2.4.2 measured Std Dev of 5% (meaning the jitter is less then 15% in 99% of the cases

> did you tried to find arduino/esp core issue related to. this?

Nope, my programing skills are too rusty.

s0170071, 2018-10-16:
The PWM is done in software with a timer interrupt. Thats where the jitter comes from. It may improve if you run the ESP at 160MHz. You do so by selecting the speed in the Arduino IDE dropdown menu prior to flashing.

Domosapiens, 2018-10-16:
@s0170071 
PWM has no need for 160Mhz.
Core 2.4.1 is perfect and has almost no jitter
Core 2.4.2 creates the problem.

uzi18, 2018-10-16:
pwm/tone is generated in new way on 2.4.2, please try to switch to 160 MHz maybe it is solution or still exist bug in arduino 2.4.2, so someone will. need to fill bug report there.

closed 2023-09-04 by TD-er