letscontrolit/ESPEasy #1658: HTTP connection fails only when waking up from deep sleep.

Wandmalfarbe, 2018-08-17:
### Summarize of the problem/feature request

ESPEasy works reliably when deep sleep is disabled. I have a BME280 sensor that sends data to the controller ThingSpeak. Because I'm using ThingSpeak, I set the value *Message Interval* in the advanced settings to `15000`. The controller and task settings are as follows:

**Controller Settings**

[image]

**Task Settings**

[image]

Up until this point everything works and I can see the values in ThingSpeak. Now I change the Sleep Mode values like in the following picture and save the page. Afterwards I connect `D0` with `RST` on my ESP and wait for it to wake up and send values to ThingSpeak.

[image]

Unfortunately no values arrive, the web interface is not accessible and the serial monitor displays the following:

```
INIT : Booting version: mega-20180815 (ESP82xx Core 00000000, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
248 : INIT : Rebooted from deepsleep #10 - Restart Reason: Deep-Sleep Wake
254 : FS   : Mounting...
260 : FS   : Mount successful, used 75802 bytes of 113201
575 : CRC  : program checksum       ...OK
585 : CRC  : SecuritySettings CRC   ...OK 
603 : INIT : Free RAM:27160
604 : INIT : I2C
604 : INIT : SPI not enabled
620 : INFO : Plugins: 48 [Normal] (ESP82xx Core 00000000, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
621 : WIFI : Set WiFi to STA
653 : WIFI : Connecting SomeNetwork attempt #0
1433 : BMx280 : Detected BME280
1988 : WD   : Uptime 0 ConnectFailures 0 FreeMem 25800
4398 : BME280: dew point 15.63C
4411 : BME280 : Address: 0x76
4411 : BME280 : Temperature: 26.33
4411 : BME280 : Humidity: 51.83
4411 : BME280 : Barometric Pressure: 1012.22
15018 : HTTP : connection failed
15032 : EVENT: System#Sleep
15038 : SLEEP: Powering down to deepsleep...
sl$ܟ|⸮d⸮|⸮

INIT : Booting version: mega-20180815 (ESP82xx Core 00000000, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
254 : INIT : Rebooted from deepsleep #11 - Restart Reason: Deep-Sleep Wake
260 : FS   : Mounting...
266 : FS   : Mount successful, used 75802 bytes of 113201
581 : CRC  : program checksum       ...OK
591 : CRC  : SecuritySettings CRC   ...OK 
608 : INIT : Free RAM:27160
609 : INIT : I2C
610 : INIT : SPI not enabled
626 : INFO : Plugins: 48 [Normal] (ESP82xx Core 00000000, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
627 : WIFI : Set WiFi to STA
659 : WIFI : Connecting SomeNetwork attempt #0
1439 : BMx280 : Detected BME280
1994 : WD   : Uptime 0 ConnectFailures 0 FreeMem 25800
4404 : BME280: dew point 15.08C
4418 : BME280 : Address: 0x76
4419 : BME280 : Temperature: 26.19
4419 : BME280 : Humidity: 50.44
4419 : BME280 : Barometric Pressure: 1012.17
15017 : HTTP : connection failed
15031 : EVENT: System#Sleep
15036 : SLEEP: Powering down to deepsleep...
```

Please note that I never checked the box *Sleep on connection failure* but the esp immediately goes back to sleep when the message `HTTP : connection failed` appears.

### System configuration

Name | Value
---|---
Build | 20102  - Mega
Libraries | ESP82xx Core 00000000, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
GIT version | mega-20180815
Plugins | 48 [Normal]
Build Md5 | b0538558c147ca339d29617cc1f2fd5
Md5 check | passed.
Build time | Aug 15 2018 02:14:42
Binary filename | ESP_Easy_mega-20180815_normal_ESP8266_1024.bin
ESP board
ESP Chip ID | 4827644 (0x49A9FC)
ESP Chip Freq: | 80 MHz

Name | Value
---|---
Flash Chip ID | Vendor: 0xEF Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE speed: | 40 MHz
Flash IDE mode: | DIO
Flash Writes | 31 daily / 8 boot
Sketch Size | 719 kB (156 kB free)

TD-er, 2018-08-17:
Setting MessageInterval to such large values will cause very strange results.
I am working on moving the interval timer for controllers to the controller themselves and make a queue which will be scheduled.
The current implementation is not right and causes all kinds of other issues.
For example setting the MessageInterval to 15 seconds will at this moment completely stop operations on the ESP for 15 seconds, which is not what should be happening.

Wandmalfarbe, 2018-08-22:
Just to note it here: The option *Message Interval* is called *Message Delay* in the wiki. A time of 15 seconds [is recommended there](https://www.letscontrolit.com/wiki/index.php/EasyProtocols#ThingSpeak) when working with ThingSpeak.

TD-er, 2018-08-22:
@Wandmalfarbe I am aware of what ThingSpeak needs and that's one of the reasons I started this: https://github.com/letscontrolit/ESPEasy/pull/1669
The general idea is to have the controllers decide when to allow to send a new message and to schedule these tasks.
So in terms of responsibilities of the new approach I am writing:

- Plugin is responsible for getting a sample from the sensor
- Controller is responsible for sending it to the receiving end at the configured pace. (and thus put a request in the schedulers list)
- Scheduler is responsible for calling the right functions at the set times.

The current implementation is:

- Plugin has sample and hands it over to controller
- Controller decides to wait, so the whole system has to wait.

And waiting for so long means all kind of things timeout, or even crash.

tonhuisman, 2022-04-18:
This looks to have been fixed already, so the issue can be closed.

closed 2022-04-18 by TD-er