letscontrolit/ESPEasy #1659: Watchdog Reset with newer buils

DittelHome, 2018-08-17:
Hardware: ESP8286 pro
With ESP_Easy_mega-20180809_normal_ESP8266_4096.bin (latest 2.4.1 core) i get 
Reset Reason:  Hardware Watchdog
 I have catched it with WEBLOG:

> Aug 17 18:24:53 Wohnzimmer_LCD EspEasy: ACT  : lcd,2,1,In  T:23.2 H:56    x
> Aug 17 18:24:53 Wohnzimmer_LCD EspEasy: ACT  : timerSet,2,2
> Aug 17 18:24:53 Wohnzimmer_LCD EspEasy: EVENT: Rules#Timer=3 Processing time:112 milliSeconds
> Aug 17 18:24:53 Wohnzimmer_LCD EspEasy: EVENT: Rules#Timer=6
> Aug 17 18:24:53 Wohnzimmer_LCD EspEasy: ACT  : lcd,1,1,Rain 0.00 l/qm      x
> Aug 17 18:24:53 Wohnzimmer_LCD EspEasy: ACT  : timerSet,4,2
> Aug 17 18:24:53 Wohnzimmer_LCD EspEasy: EVENT: Rules#Timer=6 Processing time:102 milliSeconds
> Aug 17 18:24:53 Wohnzimmer_LCD EspEasy: Command: timerset
> Aug 17 18:24:53 Wohnzimmer_LCD EspEasy: Command: timerset
> Aug 17 18:24:56 Wohnzimmer_LCD EspEasy: EVENT: Rules#Timer=2
> Aug 17 18:24:56 Wohnzimmer_LCD EspEasy: ACT  : lcd,2,1,Fri 17-08-2018     x
> Aug 17 18:24:56 Wohnzimmer_LCD EspEasy: ACT  : timerSet,3,2
> Aug 17 18:24:56 Wohnzimmer_LCD EspEasy: EVENT: Rules#Timer=2 Processing time:99 milliSeconds
> Aug 17 18:24:56 Wohnzimmer_LCD EspEasy: EVENT: Rules#Timer=4
> Aug 17 18:24:56 Wohnzimmer_LCD EspEasy: ACT  : lcd,1,1,Wind 8 km/h     x
> Aug 17 18:24:56 Wohnzimmer_LCD EspEasy: ACT  : timerSet,5,2
> Aug 17 18:25:10 Wohnzimmer_LCD EspEasy: Webserver: start
> Aug 17 18:25:10 Wohnzimmer_LCD EspEasy: Dummy: value 1: 0.00
> Aug 17 18:25:10 Wohnzimmer_LCD EspEasy: Dummy: value 2: 0.00
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: Dummy: value 3: 0.00
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: Dummy: value 4: 0.00
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: EVENT: Dummy#PWM=0.00
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: EVENT: Dummy#unused=0.00
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: EVENT: Dummy#unused=0.00
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: EVENT: Dummy#unused=0.00
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 120 minSTD time start: 2018-10-28 03:00:00 offset: 60 min
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: EVENT: Time#Initialized
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: ACT  : timerSet,2,3
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: EVENT: Time#Initialized Processing time:65 milliSeconds
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: EVENT: Clock#Time=Fri,18:25
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: EVENT: Clock#Time=Fri,18:25 Processing time:47 milliSeconds
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: IMPT : MQTT 037 Intentional reconnect
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: IMPT : Connected to MQTT broker with Client ID=Wohnzimmer_LCD-Import
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: EVENT: MQTTimport#Connected
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: ACT  : timerSet,4,3
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: EVENT: MQTTimport#Connected Processing time:75 milliSeconds
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: IMPT : [MQTT_IMPORT#out_temp] subscribed to /Wetterstation/temperature
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: IMPT : [MQTT_IMPORT#out_hum] subscribed to /Wetterstation/humidity
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: IMPT : [MQTT_IMPORT#ws] subscribed to /Gartenhaus/Windspeed
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: IMPT : [MQTT_IMPORT#Raintoday] subscribed to /Gartenhaus/Raintoday
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: WD   : Uptime 0 ConnectFailures 0 FreeMem 12736
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: LoopStats: shortestLoop: 2171 longestLoop: 4229782 avgLoopDuration: 476341.53 loopCounterMax: 13818 loopCounterLast: 10 countFindPluginId: 1
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: PluginStats P_2_Analog input - internal READ                 Count: 1 Avg/min/max 691.00/691/691 usec
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: MQTT : Intentional reconnect
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: MQTT : Connected to broker with client ID: Wohnzimmer_LCD_5
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: Subscribed to: /Wohnzimmer_LCD/#
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: EVENT: MQTT#Connected
> Aug 17 18:25:11 Wohnzimmer_LCD EspEasy: EVENT: MQTT#Connected Processing time:49 milliSeconds

With version ESP_Easy_mega-20180712_normal_ESP8266_1024.bin i get uptime for many weeks..
The only hint in the log is the line: (i think)
**Webserver: start**

Maybe this lines are also useful:

> Aug 17 18:24:42 Wohnzimmer_LCD EspEasy: WD   : Uptime 12 ConnectFailures 0 FreeMem 15416
Aug 17 18:24:43 Wohnzimmer_LCD EspEasy: LoopStats: shortestLoop: 33 longestLoop: 4233452 avgLoopDuration: 68.61 loopCounterMax: 909090 loopCounterLast: 425271 countFindPluginId: 0
Aug 17 18:24:43 Wohnzimmer_LCD EspEasy: PluginStats P_2_Analog input - internal READ                 Count: 1 Avg/min/max 1831.00/1831/1831 usec
Aug 17 18:24:43 Wohnzimmer_LCD EspEasy: PluginStats P_2_Analog input - internal ONCE_A_SECOND        Count: 27 Avg/min/max 8.11/4/15 usec
Aug 17 18:24:43 Wohnzimmer_LCD EspEasy: PluginStats P_2_Analog input - internal TEN_PER_SECOND       Count: 250 Avg/min/max 6.03/3/35 usec
Aug 17 18:24:43 Wohnzimmer_LCD EspEasy: PluginStats P_2_Analog input - internal WRITE                Count: 20 Avg/min/max 22.30/15/23 usec
Aug 17 18:24:43 Wohnzimmer_LCD EspEasy: PluginStats P_2_Analog input - internal EVENT_OUT            Count: 5 Avg/min/max 22.80/22/23 usec
Aug 17 18:24:43 Wohnzimmer_LCD EspEasy: PluginStats P_2_Analog input - internal FIFTY_PER_SECOND     Count: 1200 Avg/min/max 4.88/2/23 usec
Aug 17 18:24:43 Wohnzimmer_LCD EspEasy: PluginStats P_5_Environment - DHT11/12/22  SONOFF2301/7021 READ                 Count: 3 Avg/min/max 24693.33/24553/24772 usec

regards
Klaus

DittelHome, 2018-08-17:
It seems not to be a generally problem, because on other ESP's the same version runs without a watchdog reset.

TD-er, 2018-08-17:
Can you give a list of used plugins and controllers and used rules?

DittelHome, 2018-08-17:
No Problem: (the
[image]
 listing is from the old firmware)

> {"System":{
"Build":20102,
"Git Build":"mega-20180712",
"System libraries":"ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3",
"Plugins":48,
"Plugin description":" [Normal]",
"Local time":"2018-08-17 20:41:46",
"Unit":5,
"Name":"Wohnzimmer_LCD",
"Uptime":99,
"Last boot cause":"Cold boot",
"Reset Reason":"Software/System restart",
"Load":54,
"Load LC":9058,
"Free RAM":14376
},
"WiFi":{
"Hostname":"Wohnzimmer-LCD-5",
"IP config":"Static",
"IP":"192.168.1.24",
"Subnet Mask":"255.255.255.0",
"Gateway IP":"192.168.1.1",
"MAC address":"5C:CF:7F:34:44:92",
"DNS 1":"192.168.1.1",
"DNS 2":"0.0.0.0",
"SSID":"xxxxx",
"BSSID":"30:B5:C2:EB:DB:7D",
"Channel":9,
"Connected msec":5934158,
"Last Disconnect Reason":1,
"Last Disconnect Reason str":"(1) Unspecified",
"Number reconnects":0,
"RSSI":-62
},
{
"TTL":60000,
"DataAcquisition": [
{"Controller":1,
"IDX":0,
"Enabled":"false"
},
{"Controller":2,
"IDX":0,
"Enabled":"false"
},
{"Controller":3,
"IDX":0,
"Enabled":"false"
}],
"TaskInterval":20,
"Type":"Display - LCD2004",
"TaskName":"LCD",
"TaskEnabled":"true",
"TaskNumber":1
},
{
"TaskValues": [
{"ValueNumber":1,
"Name":"out_temp",
"NrDecimals":1,
"Value":27.6
},
{"ValueNumber":2,
"Name":"out_hum",
"NrDecimals":0,
"Value":45
},
{"ValueNumber":3,
"Name":"ws",
"NrDecimals":0,
"Value":4
},
{"ValueNumber":4,
"Name":"Raintoday",
"NrDecimals":2,
"Value":0.00
}],
"TTL":60000,
"DataAcquisition": [
{"Controller":1,
"IDX":0,
"Enabled":"false"
},
{"Controller":2,
"IDX":0,
"Enabled":"false"
},
{"Controller":3,
"IDX":0,
"Enabled":"false"
}],
"TaskInterval":60,
"Type":"Generic - MQTT Import",
"TaskName":"MQTT_IMPORT",
"TaskEnabled":"true",
"TaskNumber":2
},
{
"TaskValues": [
{"ValueNumber":1,
"Name":"in_temp",
"NrDecimals":1,
"Value":23.2
},
{"ValueNumber":2,
"Name":"in_hum",
"NrDecimals":0,
"Value":56
}],
"TTL":10000,
"DataAcquisition": [
{"Controller":1,
"IDX":0,
"Enabled":"false"
},
{"Controller":2,
"IDX":0,
"Enabled":"false"
},
{"Controller":3,
"IDX":0,
"Enabled":"false"
}],
"TaskInterval":10,
"Type":"Environment - DHT11/12/22  SONOFF2301/7021",
"TaskName":"T&H",
"TaskEnabled":"true",
"TaskNumber":3
},
{
"TaskValues": [
{"ValueNumber":1,
"Name":"Light",
"NrDecimals":0,
"Value":7
}],
"TTL":10000,
"DataAcquisition": [
{"Controller":1,
"IDX":0,
"Enabled":"false"
},
{"Controller":2,
"IDX":0,
"Enabled":"false"
},
{"Controller":3,
"IDX":0,
"Enabled":"false"
}],
"TaskInterval":60,
"Type":"Analog input - internal",
"TaskName":"LDR",
"TaskEnabled":"true",
"TaskNumber":4
},
{
"TaskValues": [
{"ValueNumber":1,
"Name":"PWM",
"NrDecimals":0,
"Value":1
},
{"ValueNumber":2,
"Name":"unused",
"NrDecimals":0,
"Value":0
},
{"ValueNumber":3,
"Name":"unused",
"NrDecimals":0,
"Value":0
},
{"ValueNumber":4,
"Name":"unused",
"NrDecimals":0,
"Value":0
}],
"TTL":10000,
"DataAcquisition": [
{"Controller":1,
"IDX":0,
"Enabled":"false"
},
{"Controller":2,
"IDX":0,
"Enabled":"false"
},
{"Controller":3,
"IDX":0,
"Enabled":"false"
}],
"TaskInterval":60,
"Type":"Generic - Dummy Device",
"TaskName":"Dummy",
"TaskEnabled":"true",
"TaskNumber":5
}

Rules:

```

> on System#Boot do
>  timerSet,1,1
>  lcdcmd,off 
>  lcdcmd,clear
> endon
> 
> on Rules#Timer=1 do 
>  lcdcmd,clear
>  lcd,1,1,Waiting... 
>  lcd,2,1,Waiting...
>  pwm,5,200
>  TaskValueSet 5,1,200
>  lcdcmd,on
>  endon
> 
> on MQTTimport#Connected do
>  timerSet,4,3
> endon
> 
> on Time#Initialized do
>  timerSet,2,3
> endon
> 
> on LDR#Light do
>   If [LDR#Light] >550 and  [LDR#Light] <650 and [Dummy#PWM] != 20
>    pwm,5,20,3000
>    TaskValueSet 5,1,20
>   endif
>   If [LDR#Light] >650 and  [LDR#Light] <750 and [Dummy#PWM] != 40
>    pwm,5,40,3000
>    TaskValueSet 5,1,40
>   endif
>   If [LDR#Light] >750 and  [LDR#Light] <900 and [Dummy#PWM] != 50
>    pwm,5,50,3000
>    TaskValueSet 5,1,50
>   endif
>   If [LDR#Light] >900 and  [LDR#Light] <1010 and [Dummy#PWM] != 100
>    pwm,5,100,3000
>    TaskValueSet 5,1,100
>   endif
>   If [LDR#Light] >=1010  and  [Dummy#PWM] != 800
>    pwm,5,800,3000
>    TaskValueSet 5,1,800
>   endif
>   If [LDR#Light] <=550 and  [Dummy#PWM] != 1
>    pwm,5,1,3000
>    TaskValueSet 5,1,1
>   endif
> endon
> 
> on Rules#Timer=2  do
>   lcd,2,1,%sysweekday_s% %sysday%-%sysmonth%-%sysyear%     x
>   timerSet,3,2
> endon
> 
> on Rules#Timer=3  do
>   lcd,2,1,In  T:[T&H#in_temp] H:[T&H#in_hum]    x
>   timerSet,2,2
> endon
> 
> on Rules#Timer=4  do
>   lcd,1,1,Wind [MQTT_IMPORT#ws] km/h     x
>   timerSet,5,2
> endon
> 
> on Rules#Timer=5  do
>   lcd,1,1,Out T:[MQTT_IMPORT#out_temp] H:[MQTT_IMPORT#out_hum]   ende
>   timerSet,6,2
> endon
> 
> on Rules#Timer=6  do
>   lcd,1,1,Rain [MQTT_IMPORT#Raintoday] l/qm      x  
>    timerSet,4,2
>  endon

```

DittelHome, 2018-08-18:
> So I have some system with known stability (very, very stable) and configuration that could help me.
ESP_Easy_mega-20180712_normal_ESP8266_1024.bin is very very stable for example....
Configuration see above

TD-er, 2018-08-18:
Good to know some versions that were stable, that makes it a bit easier to pinpoint changes.

2bontb, 2018-08-25:
I have the same problem on one of my SonOff Pow (R1) using self compiled version of mega-20180720 (4M). Problem happens every so many days.
I am using  2 other SonOFF Pow R1 with same SW and the problem never appeared.
Also my Wemos and Sonoff Basic with same SW release never showed this problem.

TD-er, 2018-08-28:
... Or the number of retries for a connection?

What services do you have running that require the wifi?

- NTP
- DNS
- Controllers?

DittelHome, 2018-08-28:
I use ser2net and mqtt. The connection to the Router is always up (I saw it in the Router log). Sometimes a have very long pings, greater than 2 seconds, what causes the watchdog...

DittelHome, 2018-08-28:
I also See in the Router log, that the ESP request a disconnect !!

TD-er, 2018-08-28:
I will change ping features to become async, meaning they will send out a request and call a callback function when result is in.
I also changed (recently) the way how connection attempts are checked based on their return value.
It was just a check for `!= 0`, but the connect function may also return negative values, which may also lead to strange issues when trying to send data after a negative return value.

I think we're on to something. In the linked thread (MDNS.begin...) it is also suggested the wifi events may also have some correlation to these issues.
I've seen it happen sometimes that the internal state of the wifi differs from the real state. So there may be some issue here too.
I have some ideas to change the wifi handling, but I wanted to open that can of worms after the release.

DittelHome, 2018-09-02:
I have changed some router settings (Bandwith reduced vom 40 Mhz to 20 Mhz). Since then I have no watchdogs anymore. The pareticular ESP is conneted over a repeater.

2bontb, 2018-09-12:
So far I had only one Sonoff that had this problem; But this afternoon on a channel switch of my Wifi AP, I lost 4 units. 2 Sonoffs POW and 2 Wemos. So it definitly has to do with Wifi.

TD-er, 2018-09-30:
See PR #1834 
I just merged a change which moves the address space of the Arduino stack to be on top of the System stack.
The latest core library appears to have shifted the Arduino stack to overlap a bit with the System stack to save about 4k of memory.
But since we're allocating quite a lot on the System stack, this may have led to an increase in reports of HW Watchdog resets.

So please test with the October 1st build, as soon as that one is ready.

TD-er, 2018-10-23:
The log in this topic mentions MQTTimport.
Yesterday I added a delay(1) to the readByte part of MQTT client PubSubClient.
Can you please test if this is now still an issue?

closed 2018-12-02 by DittelHome