letscontrolit/ESPEasy #1661: PCF8574 problem...

giig1967g, 2018-08-18:
Hi,
I am having the following problem: switching off one PCF output, switches off also all other output.

How to reproduce:
Initial status: 
all rele are off and pcf is set = 1 (PCFGPIO,65-72,1)

http://192.168.88.202/control?cmd=PCFGPIO,66,0
-> result: rele #2 turns on (correct)
http://192.168.88.202/control?cmd=PCFGPIO,69,0
-> result: rele #5 turns on (correct)
http://192.168.88.202/control?cmd=PCFGPIO,69,1
-> result :rele #5 turns off (correct)
-> AND rele #2 turns off (WRONG)

Wemos D1 mini
PCF8574 as input (8 switches)
PCF8574 as output (8 relay)
version: mega-20180818
rules disabled

[image]

any idea?

uzi18, 2018-08-18:
realys are low or hight trigger?

uzi18, 2018-08-18:
why ports on screenshot have numner from 1 to 8 and not from 65-72?

giig1967g, 2018-08-18:
hi,
relay are low trigger.
I have two PCF8574.

One with ports 1 to 8 are the switches (input).
One with ports 65-72 are the relays (output).

uzi18, 2018-08-18:
ok i will try to simulate this behavior

papperone, 2018-08-18:
I may be OT but I'd like to share my experience with port expanders.
After so many issues with PCF8574 I switched to MCP23017 and since then I have the stability I've always aimed for... oldest module running 21 days flawlessy (I manually rebooted it due to power outage). To control 8 relays I connected ULN2803 to 8 GPIOs of MCP23017 to drive the relays and use the others 8 GPIOs as input (push buttons)...

uzi18, 2018-08-18:
pcf8574 is realy old generation chip, some problems are related with this.
I have no mcp23017 on desk to test.

ShardanX, 2018-08-18:
I'm using both, PFC and the MCP.
Mostly for a kind of remote control, says 8 buttons and 8 LEDs. Older Version uses one PCF for Buttons and one for LEDs, newer version uses one MCP for the same. Both running without any problems.
It was said they have to run on 5V to run stable... my ones are running on 3.3V flawlessly.

A first try might be to vary the I²C address of one of the PCF's to make sure it's not an encoding/decoding error in the plugin.

Many faults occur from power sources. Be sure to put a 100nF ceramic capacitor over +3.3v/gnd directly under the chip to suppress spikes on the 3.3V line. Another problem can be caused by power spikes from the relays.

Last but not least make sure you have pullup resistors on both I²C-lines. See the I²C-Basics howto in the wiki for a description.

giig1967g, 2018-08-20:
Hi,
I resolved changing the power supply connected to the relays.

closed 2018-08-20 by giig1967g

papperone, 2018-08-20:
good to hear it was indeed power supply not being able to delivery enough current!

uzi18, 2018-08-20:
Give us some more data for future, what is your new and old supply.

giig1967g, 2018-08-21:
Actually I haven't changed the power supply of the device, but the power supply controlled by the relays:
ESP8266 -> Relay board -> controlling a 220V-12V transformer -> led lights.
What I changed was the 220V-12V transformer. Weird, but worked.