letscontrolit/ESPEasy #1662: Starting from mega-20180722 action is triggered in rules with no reason at startup

giig1967g, 2018-08-18:
### Summarize of the problem/feature request
Starting from mega-20180722 if I put in the rules the following code, it gets triggered at startup:

```
on p1#Switch=0 do
  event,R1toggle
endon

on R1toggle do
  if [Relay1#r1]=0
    PCFGPIO,65,1 
    taskvalueset 1,1,1
  else
    PCFGPIO,65,0
    taskvalueset 1,1,0
  endif 
  timerSet,4,1
endon

```

this piece of code is executed just after startup, and as a result it switches on the relay.

### Expected behavior
The above piece of code should not be executed.

Attached my config:
[image]

giig1967g, 2018-08-18:
I am not sure, but the issue persist even if I set the initial value of the switch in the System#Boot event.

I recall of another similar issue in the rotary encoder plugin that was triggered at startup and that I fixed, but it issue was in the plugin itself. (Bug #1142, fixed by me with #1343)

TD-er, 2018-08-18:
I am just now looking at the calls to `pinMode` and `setPinState` and there is something strange going on here.

First `hardwareInit()` is called from `Setup()` and then the init of all plugins, controllers and notifications.
In the `PLUGIN_INIT` of `P001_Switch` the pin mode is set to be an input no matter the setting of the `Settings.PinBootStates` of that pin.

I guess maybe the rules are now performed faster than before?

giig1967g, 2018-08-21:
news from this issue?
Any clue looking in the changed between the two releases?
I confirm that 20180720 didn't have this problem, while starting from 20180722 the problem is there.

TD-er, 2018-08-21:
Nope not yet, I have been working on the controllers last few days.

giig1967g, 2018-09-03:
I suspect that the reason is this snippet (from __plugin.ino):

```
               if (Function == PLUGIN_INIT) {
                  // Schedule the plugin to be read.
                  schedule_task_device_timer_at_init(TempEvent.TaskIndex);
                }
```

Probably with the new timer logic, the function "schedule_task_device_timer_at_init" fires the rules on INIT.

Could it be?

closed 2018-11-07 by giig1967g