letscontrolit/ESPEasy #1665: Feature Request: Uart - which for debug and which for comm

s0170071, 2018-08-19:
actually, this is not a question, but a feature request. 

If I need serial asynchronous communication (UART), what I do is: 
- hook up my device to Rx2/Tx2 so it is not interfering with bootloading.
- swap() Uart0 to Rx2/Tx2 pins
- connect Tx1 to Tx0 so I can send debug commands via Tx1 and read them on the monitor.

In a plugin all I can do is check if serial is disabled (and show a warning ) so I can assume control over serial. (Both)

I see the potential to improve this by e.g. adding the possibility to debug via Serial1. There could be like a drop down menu offering "serial disabled", "debug to serial1", "debug to serial0". 

What are your thoughts on this ?

TD-er, 2018-08-19:
I think it is a good idea to change it from how these things work at this moment.
But I am not sure yet what would be the best approach.
I guess making a selection to what serial port the log should be sent would already suffice your request?
Or am I missing something?

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er