letscontrolit/ESPEasy #1675: Problem rfid em4100 tag at 125khz with a RDM6300 reader

rr10, 2018-08-23:
hello i'm trying to read rfid em4100 tag at 125khz with a RDM6300 reader and espeasy firmware. The problem is that it does not read the tag.
What can I try?

Release mega-20180822

https://www.aliexpress.com/store/produc ... 28741.html

https://www.itead.cc/wiki/RDM6300 

[image]
[image]
[image]
[image]
[image]

TD-er, 2018-08-23:
I think this is more something for the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1)
It is not about problems in the code, or a feature request, so please open a topic over there.

If it later appears to be a bug in the code, you can re-open this issue.
For now, I will close this one.

closed 2018-08-23 by TD-er

TD-er, 2018-08-23:
Just one tip... 
Those breadboard power boards are meant to be placed with their power line pins in the long lines on the side of the breadboard.
Exactly where you connect the power to with the jumper wires.