letscontrolit/ESPEasy #1676: Feature request TaskValueSet vs [TaskName#ValueName]

Grovkillen, 2018-08-24:
Given this example rule:

```
On Clock#Time=Sat,16:00 do 
Taskvalueset,8,1,1
Endon

On Clock#Time=Sat,23:00 do 
Taskvalueset,8,1,0
Endon

on Pir_Sensor#Pir=1 do
if [LDR#Analog]<500 AND [Sat#dummy]=0
Publish RGB/0017F169/color/set,200,200,200
Publish RGB/0017F169/brightness/set,200 
else
Publish RGB/0017F169/brightness/set,0 
endif
endon
```
It would be much more intuitive for newcomers to just use `[Sat#dummy]=0` instead of `Taskvalueset,8,1,0`. If possible we should let users use the name of the value as well as the task value set command. Essentially we could use an array with all the tasks and their value names and just replace the TaskName with its corresponding TaskNumber, and the ValueName with its corresponding ValueNumber.

Grovkillen, 2018-08-24:
The syntax could be:

`Set,[TaskName#ValueName]=<float_value>`

example:

`Set,[Switch#Status]=1`

Grovkillen, 2019-10-10:
@td-er is this something we could/should do?

TD-er, 2019-10-10:
Sure it is more intuitive, but it also raises the question if this taskvalueset should be limited to dummy plugins only?
Maybe this is easier to add:
```
TaskValueSet,[foo#bar],123
```
Where `[foo#bar]` is then interpreted as task 8, value 1.

Grovkillen, 2019-10-11:
I like that suggestion.

Misiu, 2019-11-05:
I think that `TaskValueSet,[foo#bar],123` is much readable than `TaskValueSet,8,1,123`

Misiu, 2020-07-03:
@TD-er any updates on this?
I plan to return to using ESP Easy for another project :)

TD-er, 2020-07-03:
There is a pending PR for this feature request: https://github.com/letscontrolit/ESPEasy/pull/3098

PR #3098 (2021-04-10): [TaskValueSet(AndRun)] Use TASKname and/or VARname instead of TASKnr/VARnr
Implementation of allowing TASKname and/or VARname instead of TASKnr/VARnr in the TaskValueSet an TaskValueSetAndRun rules statements.
A substitution of the TASKname by TASKnr and VARname by VARnr is performed by looking up their values when the nr's are invalid.
Advantage of this method is, that it is a drop-in replacement of numbers by names, making rules 'backward compatible', but much more readable, and dummy variables no longer fixed-position in the devices list without changes in rules, for example (partial quote  [... cut]
closed 2021-04-10 by TD-er