letscontrolit/ESPEasy #1677: Feature request: Could _P025_ADS1115 support ADS1015 as well

craptec, 2018-08-24:
### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

Hello everyone,
I recently ran into an issue with an ADS1015 ADC I tried to use with ESPEasy. So far I found out that its not supported - only the ADS1**1**15 is. I looked trough the code of the plugin and the datasheets of both ADC and could not really find a difference (except for 12/16Bit resolution and different sample rates).
I think it would be really great if the plugin would support both ADC by choosing the correct one in a drop down menu within the plugin.
Both ADC are available on different platforms looking very similar and I am afraid that some users (like me) might order the ADS1015 instead of the ADS1115. 
Unfortunately my coding skills are not good enough to contribute the necessary changes to support both ADC.

### Expected behavior

Support both ADS1*15 devices by offering a drop down menu or radio button to choose the correct ADC which is connected.

### Actual behavior

When connecting an ADS1015 it delivers some value. While there is something connected which should provide ~0,5VDC to IN0 the measured value was around 4300 (single ended measurement)

### Steps to reproduce

1. Connect ADS1**0**15 (Vcc 5V) and connect something to one of its inputs
2. Configure the corresponding Input to be single ended

### System configuration

Hardware:
Adaruit HUZZAH
ADS1015

ESP Easy version: 2.0.0.xx (I'll have to check once again and will edit this)

ESP Easy settings/screenshots: 

### Rules or log data

```

```

craptec, 2018-08-24:
Hello everybody,
I was playing with the ADS a little more and found it might be working already. 
I put the output of a potentiometer to one of the inputs and when changing the resistance/voltage I can see the value of the input changing. It seems to be in bits instead of voltage. But after dividing by 8 it looks reasonable. 
Unfortunately I do not have a ADS1115 on hand to check what the output of it would be.

ShardanX, 2018-08-24:
It's likely that the ADS1015 basically works with the existing plugin.
The big difference is resolution: 
ADS1015: 12bit
ADS1115: 16bit
Other features are mostly same as far as i can see from a quick view through datasheets.
Please be aware that it needs a very detailed check of datasheet and references to see if there are some differences that might cause troubles.

craptec, 2018-08-25:
That makes sense to me.
Value 4016 equals 502mV if you divide it by 125µV/bit (PGA set to 1X). Though the ADS1015 should deliver 251 at this setting which would equal 502mV at 2mV/bit. Might this be as easy as just changing something in the way the bits are read? 
Just a quick draft here: 251 "becomes" 4016 if you shift it by four bits which makes sense in a 12 to 16Bit comparison.
I think it should be possible to use the existing plugin and upgrade it by a few minor changes:
1. Add a selector for ADS1015 or ADS1115 -> This will change the way the bits in the received registers are interpretet. In Case of ADS1015 divide everything by 16 (shift of four bits).
2. Change the descriptions in the GAIN-Dropdown to suit the ADS1015 (all voltages x16)

Of course we should check if there are any other differences between the two devices. I compared a few pages of the datasheets and could not find differences so far (except of 12/16Bit of course).

As a nice upgrade for the plugin it would be awesome if the output would be in mV instead of the decimal register value which has to be multiplied manually. We do have the neccessary information from the Gain already. This should help unexperienced users a bit.

ShardanX, 2018-08-25:
Please remember both ADS can give negative vaules if you use the differential inputs (AIN0-AIN1 for example). A 16bit value could be 0 to 65535 - here we have a signed value so we get -32767 to 32768!
For some more info about that and more info about using ADC in general see https://letscontrolit.com/wiki/index.php/Basics:_Analog_to_Digital_Converters

A value in mV isn't a good idea in my opinion. Usually you won't read a voltage but a sensor value instead so mV gives no better reading then the ADC "ticks". If I use a dust sensor GP2Y10 for example I get 234 ticks now - what is better with 228,52 mV instead? Both values must be calculated to the real dust value.  It does not make calculation of the value I want to get easier in any way, it just needs more code and cpu load for two calculations instead of one.

craptec, 2018-08-28:
I think the signed/unsigned values will not get corrupted if a regular division is used instead of bitshifting. Both ADS send a 16Bit register where the most left bit is marking positive or negative values. The 12bit value of the ADS1015 is left justified which leaves the last four bits always zero. (see datasheets on pages 26 for ADS1115 and 22 for ADS1015).
As soon as I have some time on my hands I will dig deeper into this topic and test a few things. I am pretty sure there should be a way to extend the plugin with minor changes to support the ADS1015 quiet good. A present I am using a simple division by 8 which seems to work just fine on single ended inputs. As soon as there is some spare time I will test what happens with differential inputs and if I can find a nice way to implement the ADS1015 as smooth as possible,

Regarding my suggestion about mV as a unit: I do understand your concern and I think you are right. Although I would suggest to give a hint or something similar to point out that the output value of the ADS is ticks and not volts. Just to point it out for unexperienced users. 
Regarding ticks vs. mV: The dust sensor's output is given in 500mV/0.1mg/m3 in case you would use 4,096V scale (125µV/Tick) and you would like to see the output in mg/m³ you would have to multiply the ticks by 125µV and then multiply by 0,2 to have the mg/m³. Obviously it saves some cpu load if you combine both multiplications to times 0,000025 although I am not really sure if this does a lot. But it might be a gamechanger when the ESP is really beeing pushed by a lot of tasks.

ShardanX, 2018-08-28:
> Although I would suggest to give a hint or something similar to point out that the output value of the ADS is ticks and not volts. Just to point it out for unexperienced users.
Maybe a link to the wiki text about ADC?

craptec, 2018-08-28:
Good idea! Maybe even both. A text would state that the value represents ticks. And one of those questionmark thinks leading to the wiki about adc (which was very interesting and helpful btw).

PR #4786 (2023-09-06): [ADS1x15] Support ADS1015 & add sample rate selector
Make timeout depending on set sample rate + add sample rate selector

ToDo:
- [x] Document
- [x] Show detected version (ADS1015/ADS1115)
- [x] Add option for multiple taskvalues per task

Fixes: #4768 
Fixes: #1677
Fixes: #1564 
Fixes: #926
closed 2023-09-06 by TD-er