letscontrolit/ESPEasy #1683: Problem with saving settings

melwinek, 2018-08-26:
The problem occurs after updating from mega-20180611 to mega-20180808 or mega-20180826.
In devices i have two relays, like in screenshoot. After reboot, devices page show state=0 but unfortunately they are turned on. The rules do not work then. I need to enter the settings of both switches in Devices Page and press Submit. Then everything works. Until the next reboot.

The second problem. Factory reset button from tools only reboot my node, instead of erasing the settings. "reset" command entered from the keyboard erases devices,controlers, but not config, hardware, advanced. 

Resetting and reconfiguring does not solve the above problem.

It looks like that in some version the structure of the configuration file has been changed.

Tomorrow I will clean ESP by uploading blank_1MB. Because this is probably the only way.

But I think that the factory reset should always clean all the settings.

giig1967g, 2018-08-26:
maybe your problem is related to #1662 ?

TD-er, 2018-08-26:
Can you test to run "reset" followed by "save" command?

melwinek, 2018-08-27:
I run save, then reset and it is still wrong, as above.

TD-er, 2018-08-27:
No, reset first, than save and then reboot

melwinek, 2018-08-27:
It probably will not solve the problem.
After "reset" I have not cleared tabs: config, hardware, advanced
He does not want to clear completely. Even static IP address remains.

melwinek, 2018-08-29:
reset, then save not solve the problem.
uploading blank_1MB not solve the problem.

TD-er, 2018-08-29:
What brand or manufacturer of flash chip do you have?

TD-er, 2018-08-30:
That's not PUYA, at least not by name.
So that is probably also not the cause.

TD-er, 2018-08-30:
Can you copy the sysinfo page here?
What size do you use?
Can you also test with the _1024 version, although the chip type suggests 32 Mbit (4 MB)

melwinek, 2018-08-30:
Unit | 3
-- | --
Uptime | 1 days 0 hours 0 minutes
Load | 19.30% (LC=15719)
Free Mem | 19664 (18128 - sendContentBlocking)
Boot | Cold boot (0)
Reset Reason | External System
Network ❔
Wifi | 802.11N (RSSI -78 dB)
IP config | Static
IP / subnet | 192.168.3.13 / 255.255.255.0
GW | 192.168.3.49
Client IP | 192.168.3.187
DNS | 8.8.8.8 / 0.0.0.0
Allowed IP Range | All Allowed
Serial Port available: | 1 (128 , 0)
STA MAC | 60:01:94:3B:52:2C
AP MAC | 62:01:94:3B:52:2C
SSID | lipowo (78:11:DC:42:B1:F1)
Channel | 6
Connected | 23h59m
Last Disconnect Reason | (1) Unspecified
Number reconnects | 0
Firmware
Build | 20102 - Mega
Libraries | ESP82xx Core 00000000, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
GIT version | mega-20180826
Plugins | 46 [Normal]
Build Md5 | 68b5a4e37d60a2651976a1c4178e370
Md5 check | passed.
Build time | Aug 26 2018 02:14:44
Binary filename | ESP_Easy_mega-20180826_normal_ESP8266_4096.bin
ESP board
ESP Chip ID | 3887660 (0x3B522C)
ESP Chip Freq: | 80 MHz
Storage
Flash Chip ID | Vendor: 0xC8 Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE speed: | 40 MHz
Flash IDE mode: | DIO
Flash Writes | 4 daily / 4 boot
Sketch Size | 709 kB (2360 kB free)

melwinek, 2018-09-10:
In 20180910 rules working ok.
After the reboot, the outputs do not turn on themselves.

closed 2019-10-10 by TD-er