letscontrolit/ESPEasy #1684: P078 Eastron does not work

vmfs1968, 2018-08-26:
I tested this plugin and it does not work
does not make any kind of milk
I currently have my Eastron running with the soft tasmota that works fine
but wanted to run it with esp_Easy
Can someone help?

vmfs1968, 2018-08-26:
I with easy esp never managed to work
I have to run is with the spft tasmota this link here
https://github.com/arendst/Sonoff-Tasmota
now I can not see what was wrong with the plugin
I know that in SDM.h it looks like it uses #ifdef USE_HARDWARESERIAL
rx tx

TD-er, 2018-08-27:
What pins do you want to use (RX/TX and the 3rd pin) and what baudrate do you use on Tasmota?

vmfs1968, 2018-08-27:
I am using gpio 1 and 3
but also in other gpios
baudrate 9600
addresses 001

E-NINA, 2018-08-28:
Hello,
I've managed to compile and overcome the compilling issue not on a clean way but changing entries on SDM library of files:

SDM.h
line 13 change ->           SoftwareSerial.h  to ESPeasySoftwareSerial.h
lines 153 and 168 change ->    SoftwareSerial  to  ESPeasySoftwareSerial

and 
SDM.cpp from:
lines 17 change ->    SoftwareSerial  to  ESPeasySoftwareSerial

Its necessary also define prior the compile the required softwareserial inputs as settings on easyeasp webpage don't works.

The issue I've right now is posting data to EmonCMS, server cann't see any input 

[image]

vmfs1968, 2018-08-28:
Very well friend
I'll try
And now send mqtt data to domoticz, see if it works
I'll give you news later

TD-er, 2018-08-28:
@E-NINA 
I don't get it why you had such compiler issues.
I did change exactly that in the .cpp and .h files as you mentioned.
So all I can conclude is that you didn't get all the files from the repository.

The settings from the plugin page are indeed not yet working, which is a known issue at this moment.

What baudrate do you use? (for what node)
And do you use 2 or 3 pins?

E-NINA, 2018-08-29:
I'm compiling with arduino-ide, using pins 12 and 13 and as I'm using a MAX485 with DE/RE lines (pin 4) see bellow:

ESPeasySoftwareSerial swSerSDM(12, 13);  //config SoftwareSerial (rx->pin6 / tx->pin7)
SDM Plugin_078_SDM(swSerSDM, 2400, 4);      //config SDM

Regarding my compilation issues where exactly the same that I found on some forums:

http://www.letscontrolit.com/forum/viewtopic.php?f=4&t=5665&start=10
https://github.com/letscontrolit/ESPEasy/issues/205#issuecomment-415838710

With equivalente method I could also compile on previous version of easyesp with eastron on playground status and old SDM template branch (on that, reference needs be changed only on sdm.h as file SDM.cpp don't exists)

Regarding emoncms, I'm using my own instance of emonemcs I'dont know if somethins is broken but if I set on propose wrong API key I could see that easyesp is trying connect. As soon I set the right API key, no errros but also no any new input.

[image]

vmfs1968, 2018-08-29:
already works
and sends the data to domoticz mqtt correctly

LeeNX, 2018-08-29:
Now I am going to have to hook up my hardware and test! ;-)  
  
@vmfs1968 - Was this a self compile/build or nightly test build?  
  
Expected feedback from me too soon

vmfs1968, 2018-08-29:
was this
https://github.com/letscontrolit/ESPEasy/archive/mega-20180826.zip

LeeNX, 2018-08-29:
I think @vmfs1968 is saying that he self-compiled the URL he posted in response to my question on   what he used.  
  
@TD-er - Thanks  
  
I was hoping it was the nightly test build of the firmware ... See what I can do in the next day or two. Free time is so limited ... ;-)

vmfs1968, 2018-08-29:
I compiled the normal version, I just added the 078 testing plugin

E-NINA, 2018-08-29:
Hello again,

One more thing I've found the issue with EMONCMS posting. Again I'm not a expert programmer and probably isn't the best solution but as I've my own istance of EMONCMS running on a virtual server and I had already some inputs running without problems from an iotawatt where isn't necessray made any change to the code it works on the fly. On easyESP I had changed the following line on _C007.ino

from
String postDataStr = F("GET /emoncms/input/post.json?node=");

to -> (I removed the "/emoncms")
String postDataStr = F("GET /input/post.json?node=");

And started to work. I don't know if this is a issue of local installations of emoncms or if is also hapenning on "emoncms.org"
See picture below:

[image]

E-NINA, 2018-08-29:
one more thing I'm using port 8080 on emoncms server

Maikel-K, 2018-09-09:
I want to start using this plugin in the coming week, i got the SDM120C.

Is anybody able to point me in the right direction to get it working? Or is it no working straight out of the box when selecting the plugin?

TD-er, 2018-09-09:
At this moment the settings made in the web interface are not being used.
So for now, you have to build a version yourself.
It will get fixed later of-course.
I added it to the main branch (as testing) since a lot of people were having problems building it from the playground. But I haven't had time to give it the attention it really needs.

Maikel-K, 2018-09-10:
Mm, i then have to find out how to do it properly... damn...

TD-er, 2018-09-10:
This plugin simply has to wait, I don't have time enough

Maikel-K, 2018-09-10:
I understand, i Will try to fix it myself. The only thing i need to find out is how to edit and compile it correctst myself.

LeeNX, 2018-09-11:
@TD-er - How much time do you need? I can send you some! ;-) haha ...  
  
@Maikel-K - Have you tried one of the nightly test builds? This is my plan, as soon as I find of the time we joking about .  
  
Also just received another MODBus/RS485 Meter ```DDS-238-2 ZN/S```, but more related to #1017

TD-er, 2018-09-11:
@LeeNX Well, see the issue-list :)
And that list doesn't even include issues for all my own ideas.
But if you can send me some boxed time, that would be great.

LeeNX, 2018-09-11:
@TD-er suffers from ```DI2``` (Development Induced Insomnia)!  
  
Thanks for all the awesome work, don't know how you do it!

Maikel-K, 2018-09-11:
I will try it in the next week or so, thursday they will come over for installing the solarpanels, i Will prepar the SDM120 and wiring tomorrowevening so that it can start logging the first produced watts!

vmfs1968, 2018-09-12:
> I will try it in the next week or so, thursday they will come over for installing the solarpanels, i Will prepar the SDM120 and wiring tomorrowevening so that it can start logging the first produced watts!

I have my sdm working fine
already several days

Maikel-K, 2018-09-12:
Can you tell me how exactly. Im new into this whole compiling thing. So any help would be highly appreciated!

LeeNX, 2018-09-13:
@Maikel-K - I think that @TD-er moved a hardcoded version into the test build, would just need to check the hardcoded pins and config for the RS485 interface pins.

Maikel-K, 2018-09-14:
So, solar panels installed, and today from around 11:15 4,25 kWh was produced and measured with the SDM120C.

I also recieved the D1 mini, loaded ESP_Easy_mega-20180914_dev_ESP8266_4096.
Created 3 devices: Power, Import, Export.

Is it now only needed to connect the following pins?
You need connect 2 wires from SDM120 to TTL adapter and 4 wires to Wemos D1 mini.
TTL VCC to Wemos 3.3
TTL TXD to Wemos D6
TTL RXD to Wemos D7
TTL GND to Wemos GND

Thats according this post: https://community.openhab.org/t/how-to-get-a-cheap-smart-meter-with-real-time-power-consumption/7205/11?u=maikelk

Maikel-K, 2018-09-27:
Is anybody able to help me set this up in a proper way? I cant get it to work unfortunatly...

Please tell me step for step what to set, i can use the D6 and D7 or the real TX/RX, i dont have preferences.

Findus76, 2018-10-08:
I also have a great interest to connect my Eastron to ESPeasy.
Unfortunately, I'm not a software engineer. Have looked at the code, but finds do not think it works.
First of all, I do not find any bus / transmit enable signal to the RS-232 <-> RS485 converter. Modbus is based on RS485, and it is not full duplex on. Therefore, the bus needs to reverse direction, which is done by bus / transmit enable.
I can not see that ESPeasySoftwareSerial has that feature.
I want to use ESPeasy as software.

Maikel-K, 2018-10-11:
Can anyone send me a working firmware for a Wemos D1 mini?
I would really love to have this working...

giig1967g, 2018-10-24:
I have just received my EASTRON.
Will start looking into the plugin

giig1967g, 2018-10-24:
Wrong model arrived: MBUS instead of MODBUS.
Have ordered a new one. Will have to wait 30 days.

TD-er, 2018-10-24:
> Wrong model arrived: MBUS instead of MODBUS.
> Have ordered a new one. Will have to wait 30 days.

... or add MBUS support ;)

giig1967g, 2018-10-24:
Is there a library of MBus for Arduino?

TD-er, 2018-10-24:
Probably, but I never looked into it.
All I know is there is a difference between the two.
- MBus is sending all values upon request.
- Modbus must be queried per register (or set of registers) to return values.

Both are uniquely addressable per device on a bus.
Not sure if signal levels are compatible.

giig1967g, 2018-10-24:
If someone can point me to a working library I will try to build the MBUS support.

giig1967g, 2018-10-25:
@LeeNX I have the MBUS version, the one with 2 wires only: BUS1 & BUS2 (see image below).
@TD-er I checked MBUS but it seems that it requires an external power supply of 24-34V to work. So n my opinion would be too complicated to implement.

[image]

[image]

LeeNX, 2018-10-25:
@giig1967g - Thanks! I have SDM120, SDM220 and SDM320 - https://www.aliexpress.com/store/group/Modbus-Energy-Meter/529519_510229023.html  
  
I also have a HiKing DDS238-2 ZN/S, but not been hooked up or tested yet.  
  
Still wondering if my Eastron SDM320 is an M-Bus? ;-\

TD-er, 2018-10-25:
Eastron has the "M" in the numbers to indicate Modbus. At least that's the correlation I thought to see when I looked for it a while ago.

TD-er, 2018-10-25:
I have also a SDM120 here and on its side is printed "SDM120-Modbus"
It has "G B- A+"  labeled near position 8, 9 and 10

TD-er, 2018-10-27:
I added a PR for this issue #1967

Would be nice if someone can test it, before I merge it.

TD-er, 2018-10-29:
Anyone already tested the latest build for this plugin?

giig1967g, 2018-10-29:
Hi
I don't have tthe right model yet.
Will arrive in 1 month

LeeNX, 2018-11-01:
R&R and now gifts from @TD-er ! Can't wait to get home and play with them toys!!  
  
Thanks @TD-er

LeeNX, 2018-11-04:
I am sure the following bits are know, just listing here because once this is working for me, not sure if I am going to come back to this  
  
 * [ ] Support baud rate change (hard coded - 2400)
 * [ ] Support meter address ( hard coded - 1)
 * [ ] Support to select the Variable ( hard coded)
 * [ ] Support Multiple Eastron Meters per ModBus
 * [ ] Support to select own ModBus address
  
 Currently polling in 5sec, looks to basically be working. Really need to be able to pull ```Total Active Energy```, other bits can work around.

uzi18, 2018-11-04:
@LeeNX  Total Active Energy is available in actual release by @TD-er

LeeNX, 2018-11-04:
@uzi18 Thanks, but the variable pull down works, but when you submit, it does not keep your selection. Revisit the device config and it's at the default settings.  
  
Another thing I just noticed, was that when I moved my test rig into a long term test, the init start up, does not seem to collect data. I can see the comes for RS485, but nothing in MQTT nor in the device display. Have to go an edit the device and submit, then data collection starts.  
  
Possible that it was an odd error, but I am sure this is what I saw on my testing, but did not take notice.

LeeNX, 2018-11-04:
@uzi18 again, thanks. I am using the nightly builds for my test. As I stated, when you revisit the device config, the above list re-sets back to their defaults.  
  
I have attached screenshots. If I am doing something wrong, please let me know.

[image]
[image]
[image]

uzi18, 2018-11-04:
@LeeNX ok, found it and fix is ready to merge

LeeNX, 2018-11-05:
@uzi18 - Thanks, that is awesome. Downloaded nightly build and uploaded the 4096 Test firmware and can confirm that selections are been kept.  
  
This is so awesome!!  
  
Still seeing that after book, Device is not polling data to the Dashboard
[image]

Only after edit and sumbit, does the dashboard start showing data
[image]

LeeNX, 2018-11-05:
Screenshot of dashboard showing Total Active Power
[image]

TD-er, 2018-11-05:
So you mean the settings of the plugin should be saved after adding to make it work?

Findus76, 2018-11-05:
Converter between UART to RS-485 is a signal to activate / deactivate the transmission line.
Where do I set GPIO for that signal?

LeeNX, 2018-11-05:
If I reboot my NodeMCU, the dashboard nor MQTT topics. are not been updated, until I edit the device and re-submit.  
  
Reboot  
https://user-images.githubusercontent.com/792948/47988573-41c27700-e0eb-11e8-857a-fb09e3b12af2.png
  
After device edit and submit  
https://user-images.githubusercontent.com/792948/47988608-5f8fdc00-e0eb-11e8-823b-ddf669e8a333.png  
  
If you see right after booting device up, one can see ```NaN``` in values for Eastron device. Edit device, then submit, even without changing any setting and then I can see values in the dashboard.

LeeNX, 2018-11-05:
@Findus76 - my RS485 to TTL does not have another single to transmit/receive.  
  
I can post a photo if that helps?

LeeNX, 2018-11-05:
@Findus76 I don't the P078 current supports the flow control signal. I think @TD-er was asking about this, but I don't have any modules that need this.

Findus76, 2018-11-05:
I started writing "own" code for this.
Uses the same codes to the instrument as espeasy. Those codes are already prepared for flow control.
Implementing it in espeasy should be quite simple.
I guess, have no knowledge of how espeasy works under the hood.
I gladly drop my development, if flow control will be implemented.

Which converter do you use witout flow control?

LeeNX, 2018-11-05:
I RS485 to TTL looks like  
[image]
  
I found it on eBay by searching for the following ```Stable RS485 to TTL Module UART Serial Port to RS485 Converter Function Module```  
  
As you can seeing, only RX/TX/5V/GND for TTL and A/B/Earth for RS485

LeeNX, 2018-11-05:
Just a small sample of data from mqtt
```
/data/ESP_Easy_SDM120/SDM120C/V 225.10
/data/ESP_Easy_SDM120/SDM120C/A 0.65
/data/ESP_Easy_SDM120/SDM120C/W 127.60
/data/ESP_Easy_SDM120/SDM120C/Wh 5.46
/data/ESP_Easy_SDM120/SysInfo/Uptime 267.00
/data/ESP_Easy_SDM120/RSSI/RSSI -77.00
```

TD-er, 2018-11-05:
> @Findus76 I don't the P078 current supports the flow control signal. I think @TD-er was asking about this, but I don't have any modules that need this.

Yep, the DE signal, is it needed?
If so, I can add it in the settings. (default to "no pin")

I also will have a look at the report of incorrect initializing the plugin.

[39 more replies]

closed 2018-11-12 by TD-er