letscontrolit/ESPEasy #1685: WeMos D1 mini I2C Dual Motor Driver Shield Driver Module TB6612FNG

stafil, 2018-08-27:
Pls support this module WeMos D1 mini I2C Dual Motor Driver Shield Driver Module TB6612FNG

https://wiki.wemos.cc/products:d1_mini_shields:motor_shield
https://github.com/wemos/WEMOS_Motor_Shield_Arduino_Library

very thanks

TD-er, 2018-08-27:
Is that a different shield than the already existing one?
See: https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P079_Wemos_Motorshield.ino

stafil, 2018-08-28:
not working for me. 
i send this command over event
WemosMotorShieldCMD,0,Forward,100

TD-er, 2018-08-28:
Have you also tried to reboot the node after setup?
In some issue tracker of the library used, I saw some comment about the I2C implementation of this shield.
It is not really conform the I2C standards. So if you're sending something to that shield's address over I2C which isn't 4 bytes, it will get in some undefined state and blocking all other I2C communications also.
This means running an I2C scan will ruin the possibility to use it until a reset of the shield (thus reboot)

stafil, 2018-08-28:
Yes I reboot after setup.

log:
ACT : WemosMotorShieldCMD,1,Forward,100
29485: WemosMotor: Motor = 1 Direction = forward Speed = 100

but motor not run.

stafil, 2018-08-28:
I mistake, motor run ....  thaaankas

TD-er, 2018-08-28:
I don't have this shield, so I cannot test it.
Maybe @TungstenE2 can help, since he's the one who made the pull request: https://github.com/letscontrolit/ESPEasy/pull/1492

Make sure you also have the motor power supply connected to the shield.

TungstenE2, 2018-08-28:
see also: https://hackaday.io/project/18439-motor-shield-reprogramming

waiting for a wiki page in order to give more details for setup

Grovkillen, 2018-08-28:
@TungstenE2 Sorry, here's the link https://www.letscontrolit.com/wiki/index.php?title=Wemos_Motorshield&action=edit

I have also created the redirect for plugin79: https://www.letscontrolit.com/wiki/index.php/Plugin79

TungstenE2, 2018-08-28:
thx, initial draft updated

tobiasfaust, 2020-04-30:
Hi,
i have 2 motorDriver Boards. Both of them i have been reprogrammed the firmware and both of them response in an i2c scanner.
At one board, i have been soldered the bridge of ic2 standby, at the other one not.
I test the board separately. But at both of them no motor will drive :(
I tested also the wemos example (https://github.com/wemos/WEMOS_Motor_Shield_Arduino_Library/blob/master/examples/motor_base/motor_base.ino) anso also no motor run.

Do you have any idea?

TungstenE2, 2020-04-30:
what exact motorshield do you have?

did you follow all the steps?
https://www.letscontrolit.com/wiki/index.php/WemosMotorshield

tobiasfaust, 2020-04-30:
Hi,
it works! The solder to close STBY for i2c was cold.
Unfortunally, in our wiki article nobody says something about this. :(
Can anybody add the info, that at back, the STBY bridge to ic2 has to solder?
Thanks ;)

TungstenE2, 2020-04-30:
see comments on this page. It is already linked in the wiki.

https://hackaday.io/project/18439-motor-shield-reprogramming