letscontrolit/ESPEasy #1688: Feature request: add dew point conversion

Grovkillen, 2018-08-30:
https://www.letscontrolit.com/wiki/index.php/EasyFormula

Similar to the conversions already in place, we should have a %RH/tempC or tempF to dew point conversion.

Possible syntax:

```
%dewpoint%([TaskName#RH],[TaskName#Celsius])
%dewpoint_F%([TaskName#RH],[TaskName#Fahrenheit])
```

Grovkillen, 2018-08-30:
Discussion going on here: https://www.letscontrolit.com/forum/viewtopic.php?f=2&t=3013&p=30841#p30841

blb4github, 2018-08-30:
I'm deploying several BME280 sensors, I see Dew Point in Log but not as sensor output. It would be great if Dew Point can be added to the output of this sensor.

TD-er, 2018-08-30:
Adding dew point as an extra parameter in the output would break support for Domoticz.

blb4github, 2018-08-30:
ok, can the dew point value as calculated in the sensor log be published  ia Rules with something like [BME280#DewPoint]?

TD-er, 2018-08-30:
I will look into the suggestion made by @Grovkillen 
When adding such value as you suggest, you make it impossible to use "DewPoint" as variable.
Also lookups like that are breaking other parts of the code.

TD-er, 2018-09-02:
Fixed by PR #1697

closed 2018-09-02 by TD-er