letscontrolit/ESPEasy #1690: OLED Display - HTTP Command error

mchacher, 2018-08-30:
I would like to display floating point numbers on an OLED screen but the "." character is causing an issue.

To give you a simple example : 
http://192.168.0.200/control?cmd=OLED,4,1,Temperature 20 is working perfectly
... but, 
http://192.168.0.200/control?cmd=OLED,4,1,Temperature 20.3 is not showing any data on the screen and _I got Unknown or restricted command!_

TD-er, 2018-08-30:
Have you tried using quotes?
http://192.168.0.200/control?cmd=OLED,4,1,"Temperature 20 .1"

Or maybe use `%22` instead of the double quote.
Not sure if that will work, but maybe worth a try.

mchacher, 2018-08-30:
yes I did it, and I just did with %22 as per your suggestion and it doesn't work.
Worth a try I agree ... but not successful.

Grovkillen, 2018-08-30:
Could you try:

`http://192.168.0.200/control?cmd=OLED,4,1,Temperature+20.3`

mchacher, 2018-08-30:
Yes - same issue.

mchacher, 2018-08-30:
One more clue, I have tested several releases of mega.
mega-20180522 is the last one working. Starting mega-20180524 the bug is effective.
This should help to identify the root cause. I was not able to do it.

v-a-d-e-r, 2018-08-30:
No issue here..... :-)
/control?cmd=OLED,1,1,Temperature 20.1
/control?cmd=OLED,2,1,Temperature 20.1
/control?cmd=OLED,3,1,Temperature 20.1
/control?cmd=OLED,4,1,Temperature 20.1

[image]

mchacher, 2018-08-30:
Strange - only difference is that I use a 128x64 screen

v-a-d-e-r, 2018-08-30:
I have also that one on stock.... ;-)

[image]

mchacher, 2018-08-30:
I don't understand. 
Are you using the latest ESP_Easy_mega-20180826_normal_ESP8266_4096.bin build?
I'm lost.

v-a-d-e-r, 2018-08-30:
I have meanwhile my own ESPeasy version running. But for this test I used the orig. source (P023) from GIT.

mchacher, 2018-08-30:
I think the issue is not on the source P023 but on some parsers located in other files. Not in the plugin

mchacher, 2018-08-30:
As said above, I have tested several releases of mega.
mega-20180522 is the last one working. Since mega-20180524 the bug is effective.
This should help to identify the root cause. I was not able to do it.

v-a-d-e-r, 2018-08-31:
Ooops, I was looking at a older source code *g*. The current code of the P023 has a change in the parser! And that change was published at 23. May:
....
        String arguments = String(string);
        int dotPos = arguments.indexOf('.');  <-- Here it looks for a dot
        if(dotPos > -1)
...

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er