letscontrolit/ESPEasy #1692: Add 2nd Receiver for Email notification

Sandominion, 2018-09-01:
### Summarize of the problem/feature request

Would be nice to be able to send a Notification Email to more than one Receiver.

### Expected behavior

When you put a comma-separated list of Receivers (e.g. “John.Doe@gmail.com, Jane.Doe@gmail.com”) in the Notify web-configuration the Notification should be send out to all the Receivers in the list. 

### Actual behavior

The Email is send out always to first Receiver in the list.

### Steps to reproduce

Follow the example I have given inside my Wiki-article 
https://www.letscontrolit.com/wiki/index.php?title=EasyNotifications&oldid=4415#Email_Notification

According to the SMTP-Syntax in https://tools.ietf.org/html/rfc5321#section-4.1.1 the command RECIPIENT (RCPT) “is used to identify an individual recipient of the mail data; multiple recipients are specified by multiple uses of this command.” 

I took a look inside the SMTP handling inside sources of ESPEasy and found the piece of code responsible for setting together the string for RCPT. There is foreseen the command only once for the one-and-only recipient:
if (!NPlugin_001_MTA(client, String(F("RCPT TO:<")) + notificationsettings.Receiver + ">", F("250 "))) break; 
Seems it would be necessary to seek the string for (different) separators and distinguish the receivers. For each optional receiver there needs to be an extra RCPT command. 

Forum discussion which ended up with this issue:
http://www.letscontrolit.com/forum/viewtopic.php?f=18&t=2954&start=18

### System configuration

Hardware: Sonoff POW Rev2
ESP Easy version:  ESP_Easy_mega-20180801_dev_ESP8266_4096.bin

dwarf2k, 2018-09-01:
I just came across this issue after having the same 'problem' a few months ago.  I modified  _N001_Email.ino to split the recipient address(es) by commas and adding a new RCP TO:<addr> line for each one.  I haven't done much C in years so not about to check in any code, but here's how I have it working:

In NPlugin_001_Send:
    // Wait for Client to Start Sending
    // The MTA Exchange
    while (true) {

      if (!NPlugin_001_MTA(client, "",                                    F("220 "))) break;
      if (!NPlugin_001_MTA(client, String(F("EHLO ")) + notificationsettings.Domain,          F("250 "))) break;
      if (!NPlugin_001_Auth(client, notificationsettings.User, notificationsettings.Pass)) break;
      if (!NPlugin_001_MTA(client, String(F("MAIL FROM:<")) + notificationsettings.Sender + ">", F("250 "))) break;

//**** Added to allow multiple email addresses
      String emailTo = "";
      int i=0;
      emailTo = getValue(notificationsettings.Receiver,',',i);

      String log = F("1) EMAIL: To ");
      log += emailTo;
      addLog(LOG_LEVEL_INFO, log);

      if (!NPlugin_001_MTA(client, String(F("RCPT TO:<")) + emailTo + ">", F("250 "))) break;
      emailTo = getValue(notificationsettings.Receiver,',',++i);
      while (emailTo[0]!='\0')
      {
        log = F("x) EMAIL: To ");
        addLog(LOG_LEVEL_INFO, emailTo);
        NPlugin_001_MTA(client, String(F("RCPT TO:<")) + emailTo + ">", F("250 "));
        emailTo = getValue(notificationsettings.Receiver,',',++i);
      }
      addLog(LOG_LEVEL_INFO, F("EMAIL FINISHED"));
// **** END of additions************
      if (!NPlugin_001_MTA(client, F("DATA"),                             F("354 "))) break;
      if (!NPlugin_001_MTA(client, mailheader + aMesg + String(F("\r\n.\r\n")), F("250 "))) break;

      myStatus = true;
      break;
    }

And using this added method:

String getValue(String data, char separator, int index)
{
    int found = 0;
    int strIndex[] = { 0, -1 };
    int maxIndex = data.length() - 1;

    for (int i = 0; i <= maxIndex && found <= index; i++)
    {
        if (data.charAt(i) == separator || i == maxIndex)
        {
            found++;
            strIndex[0] = strIndex[1] + 1;
            strIndex[1] = (i == maxIndex) ? i+1 : i;
        }
    }
    return found > index ? data.substring(strIndex[0], strIndex[1]) : "";
}  //getValue end

Hope this helps someone.  Tested by having it work for me for both 1 and 2 addresses.

TD-er, 2018-09-01:
@dwarf2k I added your suggestion.
See:  https://github.com/letscontrolit/ESPEasy/pull/1694

TD-er, 2018-09-02:
Implemented and merged.

closed 2018-09-02 by TD-er