letscontrolit/ESPEasy #1699: Task vars are not replaced with values

v-a-d-e-r, 2018-09-02:
Example: taskvalueset 1,2,[dummy#1] does nothing, but should be '123'.

[image]

TD-er, 2018-09-02:
Can you give a bit more information?
For example:
Version (self compiled or download)
Is the command given via rules, serial, http?

Have you tried to use names with letters? For example `[Dummy#one]` ?
Have you also tried to test on a 2nd dummy device?

uzi18, 2018-09-02:
what about: taskvalueset 1,2,[Dummy#1]

v-a-d-e-r, 2018-09-02:
Normally I'm a self-compiler, as you know. But to be sure, I used the last official build for testing and it was the same issue. Doesn't matter if rule or web interface command. I looked in the source code and found that the parser is looking for the '[', but that was it! Should there not be a call of 'parseTemplate' anywhere?
@uzi18: Internally all is converted to lower case.
E.g. -> line.toLowerCase(); // convert all to lower case to make checks easier

v-a-d-e-r, 2018-09-03:
Bug fixed... ;-)

closed 2018-09-03 by v-a-d-e-r