letscontrolit/ESPEasy #1701: feather request support homekit protocol

fanlessfan, 2018-09-03:
### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

feather request support homekit protocol

### Expected behavior

ESPEASY can connect to apple homekit directly without using bridge
### Actual behavior

for example, the temperature from the sensor will show in apple home app even the smart home controller, like openhab is down.

```

TD-er, 2018-09-03:
Do you know of any programming examples showing how to connect to Apple Homekit?
I don't have anything from Apple.

papperone, 2018-09-04:
As far as I remember when I read about it this protocol is very memory hungry! I doubt it can be easily matching the ESPasy framework!

fanlessfan, 2018-09-04:
It definitely works on ESP8266, but no idea of espeasy framework

papperone, 2018-09-04:
not saying it does not work, but I've read it requires up to 400kb of memory and it has been designed to be running under RTOS

fanlessfan, 2018-09-04:
you are right. it's RTOS and might not fit current espeasy. but it's great feature