letscontrolit/ESPEasy #1703: Add confirm dialog for Factory Reset

bkpsu, 2018-09-03:
### Summarize of the problem/feature request
Selecting Factory Reset performs an immediate reset. If selected by mistake, without saving the config, that results in complete data loss. 

### Expected behavior
Selecting "Factory Reset" should present user with another confirmation dialog.

### Actual behavior
Immediate reset performed

TD-er, 2018-09-03:
We already made the button red... ;)

I will look into it.

TD-er, 2019-10-27:
It has been implemented, so this one can be closed.

closed 2019-10-27 by TD-er

uzi18, 2019-10-27:
for me reset with wifi configuration presere never works

TD-er, 2019-10-27:
Can you make a new issue for it?
I'm cleaning up old issues, since a lot of them also have lots of sub-requests which are very hard to find later on.