letscontrolit/ESPEasy #1708: P029 - MQTT Helper: device status not updated when GPIO is

wdonker, 2018-09-04:
### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

Using P029 Domoticz MQTT helper with a PWM output the device status is not always updated to the status of the GPIO. When switching PWM to 0 the actual GPIO output is set ok but the device status still shows the PWM value that was before.

512598182: EVENT: Temperatuur#Humidity=67.30
512598214: Domoticz: Sensortype: 2 idx: 102 values: 23.10;67.30;3
512614483: SW : GPIO 14 Set PWM to 450
512614487: EVENT: Testdimmer#Output=450.00
512621791: WD : Uptime 8544 ConnectFailures 0 FreeMem 20712
512624483: SW : GPIO 14 Set PWM to 0
512624487: EVENT: Testdimmer#Output=450.00

### Expected behavior

Device status and GPIO should be the same at all times.

### Actual behavior

See above.

### Steps to reproduce

1. Add a P029 device, connected to a dimmer in Domoticz.
2. Switch on the dimmer to some PWM value in Domoticz.
    EspEasy device status and GPIO are now set to the same PWM value, e.g. 450. 
3. Switch off the dimmer in Domoticz (set PWM to 0.
    ESPEasy GPIO is now set to PWM 0, taskvalue of the device is still 450.

### System configuration

Hardware: NodeMCU v2.0

ESP Easy version: mega-20180826 (and probably lots before)

Grovkillen, 2018-09-14:
https://www.letscontrolit.com/forum/viewtopic.php?t=5825

Is the thing I'm talking about the "bug" here or is this a real bug?

wdonker, 2018-12-07:
Fixed in release mega-20181207

closed 2018-12-07 by wdonker