letscontrolit/ESPEasy #1709: BUG: taskvalueset not working in System#Boot

giig1967g, 2018-09-04:
Hi,
just found a strange bug:
If you create a simple node with just one device as in picture:
[image]

and with the following simple rule:
```
On System#Boot do  
 taskvalueset 1,1,10
 taskvalueset 1,2,10
endon
```

During boot the value of the Dummy device is set to 0, an EVENT is generated with value=0 and then it changes to 10 after few seconds.

See log:
at 622 the command taskvalueset sets the value.
but at 655 there is an EVENT that sends the value = 0
finally at 4530 there is an EVENT that sends the covìrrect value = 10

Same thing happen if task 1 is a switch or other device.

```
514 : INIT : Free RAM:27640
514 : INIT : I2C
515 : INIT : SPI not enabled
531 : INFO : Plugins: 46 [Normal] (ESP82xx Core 00000000, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
532 : EVENT: System#Wake
570 : WIFI : Set WiFi to STA
602 : WIFI : Connecting KeepOut attempt #0
602 : IP   : Static IP : 192.168.1.223 GW: 192.168.1.1 SN: 255.255.255.0 DNS: 8.8.8.8
615 : EVENT: System#Boot
622 : ACT  : taskvalueset 1,1,10
633 : ACT  : taskvalueset 1,2,10
653 : Dummy: value 1: 0.00
653 : Dummy: value 2: 0.00
653 : Dummy: value 3: 0.00
653 : Dummy: value 4: 0.00
655 : EVENT: Relay1#r1=0.00
666 : EVENT: Relay1#r2=0.00
676 : EVENT: Relay1#=0.00
685 : EVENT: Relay1#=0.00
695 : Command: taskvalueset
696 : Command: taskvalueset
1982 : WD   : Uptime 0 ConnectFailures 0 FreeMem 26232
4386 : WIFI : Connected! AP: KeepOut (BC:EE:7B:EF:A3:38) Ch: 9 Duration: 3782 ms
4388 : EVENT: WiFi#ChangedAccesspoint
4400 : IP   : Static IP : 192.168.1.223 GW: 192.168.1.1 SN: 255.255.255.0 DNS: 8.8.8.8
4402 : WIFI : Static IP: 192.168.1.223 (ESP23-23) GW: 192.168.1.1 SN: 255.255.255.0   duration: 16 ms
4413 : EVENT: WiFi#Connected
4424 : Webserver: start
4482 : Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 60 minSTD time start: 2018-10-28 03:00:00 offset: 0 min
4484 : EVENT: Time#Initialized
4500 : EVENT: Clock#Time=Tue,18:53
4527 : Dummy: value 1: 10.00
4527 : Dummy: value 2: 10.00
4527 : Dummy: value 3: 0.00
4527 : Dummy: value 4: 0.00
4530 : EVENT: Relay1#r1=10.00
4542 : EVENT: Relay1#r2=10.00
4553 : EVENT: Relay1#=0.00
4564 : EVENT: Relay1#=0.00
```

giig1967g, 2018-09-04:
Actually what seems to happen is that after System#Boot the system creates one EVENT for each TASK and set the value = 0.
This happens even without any rule set, without controllers, etc.

I think this is the reason for #1693 and for #1662.

I will check when it has been introduced, but I suspect that it was with the new scheduler.

giig1967g, 2018-09-04:
I forgot: FIRMWARE: Mega-20180904
Wemos D1 4M

This is the log of a unit with two device: one dummy and one switch.
No rules, no controllers.

As you can see, the EVENTS for the switch value = 0 are created twice.
Without any user intervention.

514 : INIT : Free RAM:27640
515 : INIT : I2C
515 : INIT : SPI not enabled
533 : INFO : Plugins: 46 [Normal] (ESP82xx Core 00000000, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
534 : EVENT: System#Wake
569 : WIFI : Set WiFi to STA
601 : WIFI : Connecting KeepOut attempt #0
602 : IP   : Static IP : 192.168.1.223 GW: 192.168.1.1 SN: 255.255.255.0 DNS: 8.8.8.8
615 : EVENT: System#Boot
629 : Dummy: value 1: 0.00
629 : Dummy: value 2: 0.00
629 : Dummy: value 3: 0.00
629 : Dummy: value 4: 0.00
632 : EVENT: Relay1#r1=0.00
640 : EVENT: Relay1#=0.00
644 : EVENT: Relay1#=0.00
649 : EVENT: Relay1#=0.00
664 : SW   : State 0.00
667 : EVENT: sw#Switch=0.00
1957 : WD   : Uptime 0 ConnectFailures 0 FreeMem 26024
4392 : WIFI : Connected! AP: KeepOut (BC:EE:7B:EF:A3:38) Ch: 9 Duration: 3789 ms
4393 : EVENT: WiFi#ChangedAccesspoint
4402 : IP   : Static IP : 192.168.1.223 GW: 192.168.1.1 SN: 255.255.255.0 DNS: 8.8.8.8
4403 : WIFI : Static IP: 192.168.1.223 (ESP23-23) GW: 192.168.1.1 SN: 255.255.255.0   duration: 12 ms
4415 : EVENT: WiFi#Connected
4423 : Webserver: start
4587 : Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 60 minSTD time start: 2018-10-28 03:00:00 offset: 0 min
4589 : EVENT: Time#Initialized
4601 : EVENT: Clock#Time=Tue,19:09
4620 : Dummy: value 1: 0.00
4620 : Dummy: value 2: 0.00
4620 : Dummy: value 3: 0.00
4621 : Dummy: value 4: 0.00
4623 : EVENT: Relay1#r1=0.00
4631 : EVENT: Relay1#=0.00
4637 : EVENT: Relay1#=0.00
4645 : EVENT: Relay1#=0.00
4655 : SW   : State 0.00
4657 : EVENT: sw#Switch=0.00

TD-er, 2018-09-04:
One thing that has changed is that I made a new function `schedule_all_task_device_timers()`, which is now being called right after the initial wifi connection is made.
This will schedule the task device timers to run soon after the wifi connection has been made.
With my PR regarding the controller buffers this may no longer be needed.

giig1967g, 2018-09-04:
The issue is happening before the EVENT Wifi#Connected.

giig1967g, 2018-09-04:
See the different behaviour between two versions:
Same scenario as first post but with firmware 20180501:
Same devices and same rule.

In the log at 578 there is taskvalueset and just after that the EVENT reflects the correct value set (=1).
In the first post, with recent versions, the EVENT was reporting 0 (see first log at 655).

```

INIT : Booting version: mega-20180501 (ESP82xx Core 2_4_1)
112 : INIT : Warm boot #37
113 : FS   : Mounting...
138 : FS   : Mount successful, used 76806 bytes of 957314
439 : CRC  : program checksum       ...OK
470 : CRC  : SecuritySettings CRC   ...OK
474 : INIT : Free RAM:23512
475 : INIT : I2C
475 : INIT : SPI not enabled
489 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_1)
489 : EVENT: System#Wake
526 : WIFI : Set WiFi to STA
557 : WIFI : Connecting KeepOut attempt #0
558 : IP   : Static IP : 192.168.1.222 GW: 192.168.1.1 SN: 255.255.255.0 DNS: 8.8.8.8
571 : EVENT: System#Boot
578 : ACT  : taskvalueset 1,1,1
589 : ACT  : taskvalueset 1,2,1
606 : WD   : Uptime 0 ConnectFailures 0 FreeMem 22744
609 : Dummy: value 1: 1.00
609 : Dummy: value 2: 1.00
610 : Dummy: value 3: 0.00
610 : Dummy: value 4: 0.00
611 : EVENT: Relay1#r1=1.00
619 : EVENT: Relay1#r2=1.00
625 : EVENT: Relay1#=0.00
632 : EVENT: Relay1#=0.00
4377 : WIFI : Connected! AP: KeepOut (BC:EE:7B:EF:A3:38) Ch: 9 Duration: 3817 ms
4378 : EVENT: WiFi#ChangedAccesspoint
4389 : IP   : Static IP : 192.168.1.222 GW: 192.168.1.1 SN: 255.255.255.0 DNS: 8.8.8.8
4391 : WIFI : Static IP: 192.168.1.222 (ESP22-22) GW: 192.168.1.1 SN: 255.255.255.0   duration: 14 ms
4405 : EVENT: WiFi#Connected
4415 : Webserver: start
5347 : Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 120 minSTD time start: 2018-10-28 03:00:00 offset: 60 min
5348 : EVENT: Time#Initialized
5363 : EVENT: Clock#Time=Wed,00:11
13314 : EVENT: Clock#Time=Wed,00:12

```

giig1967g, 2018-09-05:
I found the cause of the two problems and would like to propose a solution:
Change functions "schedule_task_device_timer_at_init" and loop() 

The solution is:
1) CHANGE the following function to:
```
void schedule_task_device_timer_at_init(unsigned long task_index) {

  //if interval in the device's page is = 0 then do not schedule the timer
  unsigned long intervaltimer = Settings.TaskDeviceTimer[task_index];
  if (intervaltimer != 0) {
    unsigned long runAt = millis();

    if (!isDeepSleepEnabled()) {
      // Deepsleep is not enabled, add some offset based on the task index
      // to make sure not all are run at the same time.
      // This scheduled time may be overriden by the plugin's own init.
      runAt += (task_index * 37) + Settings.MessageDelay + 3000;
    } else {
      runAt += (task_index * 11) + 10;
    }
    schedule_task_device_timer(task_index, runAt);
  }
}
```

the reasons are:
a) the original function was creating timers also for devices that had interval = 0 (like switches)
b) the first timer was scheduled before the System#boot event had finished, so I added 3000ms for the first scheduling.
This solved both problems reported for this bug.

2) REMOVE the call to the function "schedule_all_task_device_timers()" from the main loop() function.

The reason is because this function is creating nested timers. Infact they will be scheduled after boot at interval set by the device and then after first WiFi connection they are created one more time without disabling the previous ones.

giig1967g, 2018-09-06:
Hi @TD-er:
I will create a PR for my changes as soon as you merge PR #1669 to avoid overlapping.

In the meantime, I resume my 3 proposed changes:

1) set a fixed delay of 3 seconds before starting the scheduler handler to avoid the timer to run before the System#Wake and System#Boot events end:
This value should be verified. Maybe 5 or 10 seconds is more conservative.
Or maybe we should schedule this timer AFTER the System#Boot finishes.
Is there a way to determine it?

2) skip timer creation when interval = 0: 
I used the same logic present in function "process_task_device_timer", where you do NOT SET the next timer, if interval = 0. So with the current code, a plugin with interval=0 would run only once, at boot.
That seems to me incorrect. Also plugins with interval=0 are only those where the interval is optional, therefore are the ones that should never be scheduled if interval=0, not even at boot time.

3) remove call to the function "schedule_all_task_device_timers()" from the main loop() function:
I didn't notice the uniqueness of the ID. Good to know.
But anyway I believe that this call should be done after the controllers are connected and not after a  generic WIFI connection, because the fact that the WIFI is connected does not mean that the controllers are connected (MQTT could not be connected yet, for example).
Maybe the plugins should have an option called "SendData when controller (re)connects" and then when the controller re/connects the scheduler will create a timer only for those tasks that require this function.
I manually have this functionality in my rules to make sure that the broker is always synchronized with the device:
```
on MQTT#Connected do
  timerSet,1,1
endon

on rules#timer=1 do
  taskrun,1
endon

```

giig1967g, 2018-09-06:
Finally there is another thing to be considered:
when millis() reach its maximum value it restarts from 0 (after 49 days of operation). 
Looking into the code there is no check for this event.

giig1967g, 2018-09-06:
Hi,
please consider my commens above and below as constructive help and not as a criticism :)
You are doing an incredible job.

Actually I was referring not to time differences, but to time additions.
In this case:
```
unsigned long runAt = millis();
runAt += (task_index * 37) + Settings.MessageDelay;
```
if millis() is close to 2^32 (=4,294,967,295) and you add something which result goes over 2^32, what happens? Does it overflow with an exception? Or it restarts from 0?

Regarding your second comment, my suggestion was related to the fact that the function schedule_task_device_timer_at_init() is called at least twice:
- Everytime the PLUGIN receives the PLUGIN_INIT 
- Once when WIFI connects for the first time.

The first call is fine (with a delay to avoid the timer to run too early)
The second call as you said is now useless due to the new queue management.
But my suggestion was to force a timer schedule whenever the controller re/connects.
Consider the following scenario:
Unit is connected to wifi and to broker.
Broker starts a software update and reboots several times (W10 for example); the operation takes 30 minutes to complete.
The queue could not be long enough to keep up with this time frame (especially if the logic is ignore new messages), so whenever the broker comes up, the unit sends the current status to the broker to update it.

Regarding the "skip when 0", your current code is already "disabling the plugin for fetching data", but it still send the data once. The bug is that it's sending the data once on INIT.

Finally, consider that if you move the call of the function schedule_task_device_timer_at_init() into the setup() you will miss the plugin config updates, that are triggered with PLUGIN_INIT.

TD-er, 2018-09-06:
Adding something to a unsigned int will cause an overflow.
Meaning you actually count modulo. Or in other words, the most significant bits that won't fit in the uint will get lost.

So adding something to `millis()` can always be done. The problem is in the comparing to see if the timer has passed.
You cannot use just a > or <.
So that's why I created the timeDiff function and the 2 convenience functions calling that function to check if a timer has passed.

I guess those functions should be included in the main Arduino libraries, because a lot of other libraries do incorrect compare of timers.

My point was that there are several reasons why the interval may be set to 0, so we have to check more than just that single value. And I think you really have found the problem causing these issue. The only question that remains is how to properly fix it.

TD-er, 2018-09-06:
I am now looking at `schedule_task_device_timer` and was thinking about this:

```C++
void schedule_task_device_timer(unsigned long task_index, unsigned long runAt) {
  if (task_index >= TASKS_MAX) return;
  byte DeviceIndex = getDeviceIndex(Settings.TaskDeviceNumber[task_index]);
  if (!Device[DeviceIndex].TimerOption) return;
  if (Device[DeviceIndex].TimerOptional && Settings.TaskDeviceTimer[task_index] == 0) {
    return;    
  }
  if (Settings.TaskDeviceEnabled[task_index]) {
    setNewTimerAt(getMixedId(TASK_DEVICE_TIMER, task_index), runAt);
  }
}
```

What do you think, are all options covered, when not to schedule?

giig1967g, 2018-09-06:
Hi Gijs,
I think that all the options are covered.

TD-er, 2018-09-06:
Could you test the latest commit?
I made a [test build for it](https://www.dropbox.com/s/wg3zyb6iih0zhbk/ESPEasy_pr1669-5-g4e95bbe.zip?dl=0)

giig1967g, 2018-09-06:
if I am not wrong, looking at your code changes, you haven't added a delay in the first timer schedule (at INIT) in schedule_task_device_timer_at_init().

TD-er, 2018-09-06:
That is correct.
And if possible, I would not want to either.
It will add to the time needed to run which is not desirable when running from deep sleep.
I want it to work with as little delays as possible and setting some "N seconds should be enough" is not the way I would like to fix stuff.
If some delay is needed, I want to know what delay and why, so then we can make it run at that moment.

giig1967g, 2018-09-06:
your latest build works as expected 👍 

I agree that is better to find the exact moment to insert it.
The 3 or 5 seconds was a workaround until a better solution was identified.

TD-er, 2018-09-06:
Great, we're getting to something stable again :)

closed 2018-10-21 by TD-er