letscontrolit/ESPEasy #1710: strange GPIO behavior

TungstenE2, 2018-09-04:
### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

Created local build from Mega version 20180621 for my new Wemos motorshield plugin P079.
Worked perfect with rules and sunset/sunrise timer for several weeks.

Updated last week to Release mega-20180826, deleted device and created new to make used of the included plugin. Also updated  yesterday to Release mega-20180903.

I can not say what exactly is wrong, but I discovered that GPIO status of reed switches is not correct sometimes and commands are not triggered correct.

Also tried to make use of variable %sunset+15m% instead of timer, but this did not work at all. Timer was working fine.

This resulted in the wrong or not triggered opening/closing of the chicken house door.

Tonight I went back to my local build based on version 20180621 and everything seems to be fine again.

What is the problem?

TungstenE2, 2018-09-04:
update: I also discovered that status of Environment - DS18b20 sensor was 'nan' for long time, before value was shown/updated.

TD-er, 2018-09-04:
Could you test to set the platform in your platformio.ini file to `espressif8266@1.7.1` ?
That was the used platform around 0618.

Just to test if it is a bug introduced in ESPeasy or in the core libraries.

TungstenE2, 2018-09-05:
sure, which line exactly?
https://github.com/letscontrolit/ESPEasy/blob/mega/platformio.ini

TungstenE2, 2018-09-05:
first feedback. after reboot reed switch GPIO = 0 although should be =1, as magnet is next to reed switch. but not sure if this was the case with old version.

TD-er, 2018-09-05:
There are other reports (a lot) about incorrect initialization of GPIO values at boot, so that part is already very high on my urgent list.

TungstenE2, 2018-09-06:
tested now for 24hrs with proposed change (espressif8266@1.7.1) and it seems like that rules are now working as with version from 062018, except the initial GPIO values.

TungstenE2, 2018-09-08:
This morning strange behavior again. 
Rule to open door did not work properly after manually opening the GPIO status of the reed switch changed for 10min several times, while the magnet was in the same position and did not move.

TungstenE2, 2018-09-08:
[image]

TD-er, 2018-09-08:
I guess you are using the P001 Switch input to detect the state of the reed switches?
As far as I know, that one doesn't filter the signal.
Maybe a Hall switch does give a more clean signal?

I did have a short look at P001 and P003.
The last one has some filtering which is useful for mechanical switches.
I will make a filter option for P001 also.

I suspect those switches will give more than one pulse, which will result in unexpected behavior.

TungstenE2, 2018-09-08:
Using this:

Device:  Switch input - Switch

Do not know if P001.

Was working with version from 06/208 very well for several weeks.

[image]

TungstenE2, 2018-09-08:
[image]

TungstenE2, 2018-09-08:
[image]

TD-er, 2018-09-08:
That's indeed P001 in the source.
P003 is the pulse counter plugin.
I plan to split the code both of these plugins could use in some separate class and this allows me to add support for filtering out very short pulses which mechanical switches produce. (so called de-bounce filter)

I don't know what has changed in the core libraries since june, but I do know I eliminated a lot of waiting cycles, which may have made this particular problem worse.

TungstenE2, 2018-09-08:
ok, but the reed switch in my case does not give very short pulses. the motor is very slow turning (do not want the chicken to be headless) and it takes 26 sec for opening the door. when the reed switch is triggered it will stay in this status until new event is triggered the next sunset/sunrise.

TD-er, 2018-09-08:
If you look at the signal from the reed sensor, using a scope, you will see it is not giving a single pulse.
Any mechanical connection will bounce a few times and thus give multiple pulses.

Often this is filtered in software, or a capacitor is used to filter these spikes.

TungstenE2, 2018-09-09:
today I flashed back to own build from 06/2018. door automation is not reliable with build from last week.

TungstenE2, 2018-09-21:
@TD-er seems like you are right. I also get the strange GPIO behaviour with my reed contacts with the old version, but only once or twice per week. So any news on this?

TD-er, 2018-09-21:
You could simply add a small capacitor between the GPIO pin and the GND.
That will dampen the signal a bit.
You could experiment with the value.
Start with 1 nF and work your way up a bit I guess.
I haven't started on the GPIO enhancements yet.
I came back Wednesday, but then didn't touch my laptop and yesterday it was all a bit hectic for me personally.

giig1967g, 2018-09-21:
I have the opposite behaviour:
once a week one of the switches (either P001 or P019) of my Wemos change thier status to 0 (=pressed) even if there is no one touching them.
The wires connected to the switches are quite long (10-15 meters) and go inside the walls together with 230V mains.
Could it be an interference?
How can I filter it?

TD-er, 2018-09-21:
You can also try to use stronger pull-up resistors. (lower value)
Twisting the wires can also improve this.

giig1967g, 2018-09-21:
and a capacitor?
You were suggesting 1nF. Ceramic or electrolytic?

TD-er, 2018-09-21:
Those small ones aren't available as electrolytic.
And the non-elco ones don't have a + and - pole, which is also nice to have in such situations.

TD-er, 2018-09-21:
Just start with one of both and then the capacitor is maybe best for mechanical contacts.
A stronger pull-up is always a good idea for longer leads. Just make sure the other end can handle the extra current needed to pull the pin to GND.

TungstenE2, 2018-09-28:
last night I found a bad wire connection of one of the reed contacts. Updated to build from 27.09. and will test now.

TungstenE2, 2018-10-04:
from my point of view this issue can be closed. seems like my problem was caused by a bad wire, were old version from 06/2018 was more tolerant to the signal. For the last couple of days I test with version 27/09/2018 and fixed the wire.

giig1967g, 2018-10-04:
I haven't yet added the capacitor on the wires and I still have random status change

TungstenE2, 2018-10-04:
did you check the wires? I bought wires that did not fit 100% the connector pins and were some kind of loose. touching them caused GPIO state changes, or even very light natural environment movements.

giig1967g, 2018-10-04:
I use cable crimp. But will recheck all of them...

giig1967g, 2018-10-13:
> Just start with one of both and then the capacitor is maybe best for mechanical contacts.
> A stronger pull-up is always a good idea for longer leads. Just make sure the other end can handle the extra current needed to pull the pin to GND.

@TD-er: Sorry for asking this question again, but would like to be sure:
My config:
WEMOS ----------(15m)-------- SWITCH

Where should I put the capacitors?
a) Close to the Wemos or close to the switch or it doesn't matter?
b) The capacitors should be put between common and switch pin?
c) capacity=1nF ceramic?

Grovkillen, 2018-10-13:
@giig1967g and make sure the logical signal is high enough after traveling 30 meters.

giig1967g, 2018-10-13:
the switch works fine. But I noticed that sometimes the lights are on without any apparent reason.
It happens if I start some tool (drill, soldering, etc.)
So I wanted to protect the switch from interference.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er