letscontrolit/ESPEasy #1713: Unlike version 1.0, the name (hostname) is not just the unit name, it also adds the unit number

Bejobe, 2018-09-05:
### Summarize of the problem/feature request

ESPEasy_mega-20180904: Unlike version 1.0, the name (hostname) is not just the unit name, it also adds the unit number. Please make this optional. (Example: not: esp03-0, but: esp03)

### Expected behavior

Like version 1.0, the name (hostname) should be the unit name. (Without unit number (May be optional.)) 

### Actual behavior

Unlike version 1.0, the name (hostname) is not just the unit name, In my case, I need to control then hostnames completely.

### Steps to reproduce

See the connected WLAN clients at the Wifi-AP

### System configuration

Hardware: NodeMCU 

ESP Easy version:  GIT version: mega-20180904

TD-er, 2018-09-05:
This is used at a lot of places:

- As AP SSID when in AP mode (during setup)
- As MQTT client id alternative to the default "ESPeasy_"+MAC-address
- Hostname for WiFi client.

I guess it should be uniform and apply to all use cases.
As AP during configuration, is probably not an issue, since then the settings are just default.

Bejobe, 2018-09-06:
Thank you for your answer.  Please make Hostname for WiFi client changeable.

- Important (for me in this environment here) is only the "Hostname for WiFi client.". 
- AP name is only during setup. (not interesting.... ;-))
- MQTT client name is changeable! (Controller Settings -> Controller Publish)

First there is no reason to add (without needs) a unit number to the hostname.  Second this a big change to version 1.x of esp easy.

Grovkillen, 2018-09-06:
I get your request but saying that there's no reason to  `hostname = unit name + unit number` is only from your point of view. We (the devs) wanted to remove a common mistake where users named their nodes by giving them the unit name of their placement (kitchen, garage, bedroom etc) and only differenced them with a unit number. So we don't do changes just to do changes, we have a discussion behind the scenes.

But that being said, we understand your request.

Bejobe, 2018-09-07:
Comparation Wifi.ino (ESPEasy_R120) and ESPEasyWifi.ino (ESPEasy_mega-20180904). Please see below: 

- Why do you want not be compatible to the well proven old version? It is proven.

- Why take away the controll of Settings away from the user? Now user can not control hostname by himself.

- Why do you not believe in the users. If they want have hostnames like "kitchen-1" they can put it in the field "Settings name" themself. 

- Please change line 430 of file ESPEasyWifi.ino from
     String hostname(WifiGetAPssid()); 
to
     String hostname(Settings.Name); 
  
- Please give the user the opportunity to control it by himself instead force.

Thank you very much.

Comparation Wifi.ino (ESPEasy_R120) and ESPEasyWifi.ino (ESPEasy_mega-20180904) here:
 
Lines 36ff of file wifi.ino (file date: 31.03.2016 15:30:54) (ESPEasy_R120):
---------------------------------------------------------------------------:
//********************************************************************************
// Connect to Wifi AP
//********************************************************************************
boolean WifiConnect(byte connectAttempts)
{
  String log = "";
    
  char hostName[sizeof(Settings.Name)];
  strcpy(hostName,Settings.Name);
  for(byte x=0; x< sizeof(hostName); x++)
    if (hostName[x] == ' ')
      hostName[x] = '-';
 

Lines 414ff of file ESPEasyWifi.ino (file date: 04.09.2018 02:27:52) (ESPEasy_mega-20180904):
--------------------------------------------------------------------------------------------:
//********************************************************************************
// Determine Wifi AP name to set. (also used for mDNS)
//********************************************************************************
String WifiGetAPssid()
{
  String ssid(Settings.Name);
  ssid+=F("_");
  ssid+=Settings.Unit;
  return (ssid);
}

//********************************************************************************
// Determine hostname: basically WifiGetAPssid with spaces changed to -
//********************************************************************************
String WifiGetHostname()
{
  String hostname(WifiGetAPssid());  // <----- line 430 of file ESPEasyWifi.ino (file date: 04.09.2018 02:27:52) (ESPEasy_mega-20180904)
  hostname.replace(F(" "), F("-"));
  hostname.replace(F("_"), F("-")); // See RFC952
  return (hostname);
}

TD-er, 2018-09-07:
Why do you think we want to force things?
I asked you what would be the best options, with intent to change it to be the most flexible solution.
It hasn't been changed yet, because there are only 24h in a day of which 10h is taken for work, there is a family to take care of, I do reply to a lot of other issues too and also work on a lot of coding.
Oh and also try to have some sleep every now and then.

Bejobe, 2018-09-07:
Thank you for your answers. This new version 2.0 of ESPEasy is a very, very nice and good piece of software. You have add many very good and usefull new things to it, like notifications, rules, ntp, and and and.... so it does not hurry up....

TD-er, 2018-09-14:
Should be fixed, when I merge the linked PR above.

closed 2018-09-14 by TD-er