letscontrolit/ESPEasy #1714: Request for emails with another user name, not only the exact unit name

Bejobe, 2018-09-05:
### Summarize of the problem/feature request

Nice very good feature 'notification'. Please add an option to send the email with another user name, not only the exact unit name (Example not: 'esp03 <myaddress@side.tld>' but: 'name <myaddress@side.tld>' 

### Expected behavior

Please add an option to send the email with a own user name, (Example 'name <myaddress@side.tld'>) 

### Actual behavior

Only the exact unit name (Example 'esp03 <myaddress@side.tld>' ) is possible.

### Steps to reproduce

Sending an email.... 

### System configuration

Hardware: NodeMCU

ESP Easy version:

TD-er, 2018-09-05:
Another sender you mean?

I have to look into how things are stored in the settings, because the reserved spaces for notifications is already filled.

TD-er, 2018-09-05:
The current implementation is using this template and then change parts of it.

```C++
		String mailheader = F(
			"From: $nodename <$emailfrom>\r\n"
			"To: $ato\r\n"
			"Subject: $subject\r\n"
			"Reply-To: $nodename <$emailfrom>\r\n"
			"MIME-VERSION: 1.0\r\n"
			"Content-type: text/html; charset=UTF-8\r\n"
			"X-Mailer: EspEasy v$espeasyversion\r\n\r\n"
			);
```

So I will try to check if the notification sender has some markup and use that.

Bejobe, 2018-10-04:
At first, many thanks for your implementation! 

There is still an issue with this request,  because it does not work.
 
Because I do not know, that I can reopen this request, I put a new request with the number 1807 an the titel "To the (closed) issue #1714 (Request for emails with another user name.....)". But now I put it here:

### Summarize of the problem/feature request

Thank you very much that you have implemented the requested feature so very soon ( 09092018 ! ) . 

### Expected behavior

Unfortunately the feature does not work. If I put ["MyESP" \<pushmail@foo.bar\>] (or  [\<pushmail@foo.bar\>] ! ) in the  Field "Sender:" in the "Notification Settings" no email is send. (If I put [pushmail@foo.bar] in the  Field "Sender:" in the "Notification Settings" all works well -> email is send. )
### Steps to reproduce

I have checked your new code in _N001_Email.ino beginning from line 92 . Exept the two lines 110 and 111 (which are not necessary but not really wrong), the code works in my tests well. (I have put the code (line 76 - 114 of _N001_Email.ino) in a test.ino and it works well). So I not really unterstand why no Email is send. Please test and check it. Thank you very much.  I Think it has something to do with line 123 in _N001_Email.ino:
"if (!NPlugin_001_MTA(client, String(F("MAIL FROM:<")) + notificationsettings.Sender + ">", F("250 "))) break;" 

### System configuration

Hardware: NodeMCU esp8266

ESP Easy version: mega-20180927

closed 2024-05-01 by TD-er