letscontrolit/ESPEasy #1716: Feature request: Short serial command reference

s0170071, 2018-09-07:
similar to #1549 , there could also be a page showing implemented serial commands.

- The simple version is to store it in the ESP

- The memory saving option would be to <iframe> a wiki page but that would lead to wrong info when the page is updated but the ESP is not. 

- The cool option would be to still use the iframe and have the ESP "tag/tick" the commands on that page that it supports. This way users would be implicitly informed about updates ;-) 
Extra nerd bonus level: provide fallback option if no web access is available.

Grovkillen, 2018-09-07:
We are in the process of releasing a completely new GUI which will incorporate things like this.. We'll be releasing more information soon :)

s0170071, 2018-09-07:
Sounds like a lot of work... 
Has there been some discussion about it that I missed?

TD-er, 2018-09-07:
@s0170071 It is indeed a lot of work and it will be done in stages, after the next release.

In short, a lot of interactions will be done via JSON, which makes the frontend more detachable from the firmware itself. (and still capable of running on a single node)
For example the system variable page would have to be done twice, once generating HTML and one for JSON to be layout by code running mainly in the browser.
So in that sense I get @Grovkillen , why it may be easier to implement it when we're moving to the new web page generation.
But when you look at the current implementation of the sysvar page, you can imagine it would take about 10 - 20 lines of code to transfer it into JSON. So we're already working toward such a transition. :)

And this move will allow for a lot (!!!!!) new features which we will explain later.

s0170071, 2018-09-07:
I get the idea behind this. Sounds very promising. 
You could even move the webpage generation to one/any single node and have all others supply their JSON stuff.
Now if that single node is a raspi or the internet this becomes faster than lightning.

Grovkillen, 2018-09-08:
You get the point, spot on! 😎

tonhuisman, 2023-09-03:
This seems to be solved adding documentation in RTD, so can be closed.

closed 2023-09-04 by TD-er