letscontrolit/ESPEasy #1718: MQTT Retain Flag

flygrmn51, 2018-09-07:
Can we get the retain flag at the device level? The retain flag is useful mainly for status of switches, less so for periodic measurements such as temperature and humidity.

I am using mega 20180818 standard and dev versions on generic esp8266-12F.

Thanks

TD-er, 2018-09-07:
As far as I understood, there is some limit on max numbers of retained messages.
Not sure yet if it is one per login, or one per topic (for the same login?)

flygrmn51, 2018-09-07:
I believe there is one  retained message for each topic (the last one), however, there could be many messages in the queue when quality of service levels have been set and the connection is interrupted. So on the ESP device messages could build up if the broker was off line for a period of time. My main concern is for the broker to hang on to messages while the subscriber is off line. The broker hangs on to the messages for the subscriber until the connection is re-established. This also means that the ESP could be receiving several messages if the subscription qos level is set above 0 and the connection to the broker was lost. I am not sure if this presents a problem. The messages would still come in one at a time, just at a higher tempo if there are many. This would only happen if the broker was online but the esp was off line and unable to reconnect.

TD-er, 2018-09-07:
I guess the same applies as I just wrote on the MQTT QoS issue.

JeeCee1, 2020-02-07:
I would like to request the same, but for the reason mentioned here: https://www.letscontrolit.com/forum/viewtopic.php?t=6478

fnees, 2020-07-17:
I cant even find the global retain flag on the R120 (I installed my first R120 firmware just now), but, yes I would love to have it at device level!

tonhuisman, 2020-07-17:
> on the R120 (I installed my first R120 firmware just now)

You should be using a (far) more current release, to be obtained from here: https://github.com/letscontrolit/ESPEasy/releases

tonhuisman, 2022-05-07:
This seems to be outdated, so can probably be closed

TommoT1, 2024-02-09:
Do we now have a Retain Flag selectable in the Publish Command ?
So we can set it or not as per Topic in the Rules?

TD-er, 2024-02-09:
Nope, not yet.
But I think we can add a command version `publishRetain`

TD-er, 2024-02-09:
There is one thing that really needs to be checked first before adding this command...
Does MQTT support sending non-retained messages to a topic which had received retained messages?
And if the MQTT protocol does support it, what do MQTT brokers do when this happens?

TommoT1, 2024-02-09:
That would be great, so than we would have Publish and PublishR and this would be independent from the Controller Settings (there is a retained checkbox as well)?
I just played around yesterday and happened to publish retained to a topic and later non retained, the Broker is fine, but the retained message always comes back after the non-retained one (may be when it was delived to a subscriber once?). So I had to manually remove the retained one, it took only a while for me to understand where those numbers were comming from ;-). My Broker is a out-of-the-box mosquitto.

TD-er, 2024-02-09:
The retained flag in the controllers is a global flag.
Also there is a flag for having the LWT messages sent with a retain flag.

Not sure if there exists an "unpublish" command in the MQTT language/protocol.
And I think a retained message will be forwarded to any who subscribes to that topic, not just depending on the client.
Though that might be related to the QoS flag when publishing a message.

Maybe I should read this [excellent MQTT Essentials](https://www.hivemq.com/mqtt/) again to refresh my knowledge about MQTT.

TommoT1, 2024-02-09:
Exactly, it gets delivered to any new subscriber.
The way I know to delete a retained message is to send an empty retained message to that topic.
But I think those things don´t have to be handled by espeasy.
To have the option to send retained and non-retained should be enough.

TommoT1, 2024-02-09:
maybe someday somebody exactly needs to do both, publish retained an non-retained to the same topic for whatever reason.....

TD-er, 2024-02-09:
Well if it is that simple, publishing an empty message should also be an option of this new command :)

TommoT1, 2024-02-09:
````
PublisR  <topic>,""
````
Should do the job

tonhuisman, 2024-02-20:
@TommoT1 I've added the `PublishR` command that enforces the Retain flag on publish, can you please test by downloading and installing a binary from this [GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/7979299454)?

TommoT1, 2024-02-21:
Hi Ton!
I just tested it with the ESP_Easy_mega_20240220_normal_ESP8266_1M.bin and it works like expected!
Should then the checkbox "Publish Retain Flag:" in the controller settings be removed because it's not needed anymore and if checked Publish and PublishR is retained.
Thanks for implementing!

TD-er, 2024-02-21:
Nope, that should be kept there as there are units out in the field running rules which won't get automagically updated if someone updates ESPEasy firmware.

PR #4984 (2024-02-21): [MQTT] Add PublishR command: publish with will-retain
Resolves #1718 
Resolves #967 

Features:
- Add command `PublishR,<topic>,<value>` to send a value to a topic on the first enabled MQTT Controller
  _with the Will Retain flag enabled_
- Updated documentation
PR #5371 (2025-09-07): [MQTT] Add Retain option per task
Resolves #4611 
Resolves #1718 

Features:
- Add a Retained flag for each MQTT Controller per task

The Retained option sends data with `Retain` enabled for the task. this setting is overridden by the Controller **Publish Retain Flag** setting, then it will send the Retain flag for _each_ task's value sent.

TODO:
- [ ] Testing
- [x] Update documentation
closed 2024-02-21 by TD-er