letscontrolit/ESPEasy #1729: Release mega-20180910

Sasch600xt, 2018-09-10:
Hello all,

i came from build 20180908

Hardware is an NodeMCU Lolin.
i took the "normal,4069" image.

I need to send http commands, thats why i upgrated to 20180910.
After upgrade at least generic - system info Devices not working anymore like "rssi,uptime,system load" etc.

Please tell me if i did something wrong ?
I did also upload a blink sketch inbetween the first and second try...
no luck so far....

Sasch600xt, 2018-09-11:
and you are working with the release 20180910 ?

Heiend, 2018-09-11:
Yes. I'm just updating all my devices to 20180910.

Sasch600xt, 2018-09-12:
ok....what about if you try to send system infos too over domicz protcol for example ? .....thats what is not working here

TD-er, 2018-09-12:
My test setup at home is sending CPU load and mem free to Domoticz.
Which one are you using? http or mqtt?

Sasch600xt, 2018-09-12:
http.....can i somehow give you more informations ? Screenshots ? What do you need.......and dont come up with "vacation" or something like this :)

Sasch600xt, 2018-09-12:
so here would be a few screenshots between 20180908 and 20180910
i hope it will help to find the problem....maybe it´s me ?
Have a good one
Sascha
[issue_sascha.zip](https://github.com/letscontrolit/ESPEasy/files/2374772/issue_sascha.zip)

Oxyandy, 2018-09-12:
Serial logging via cable used to show (info set in settings) **uptime** updates every 30 seconds
now only seems to report when something is not as expected ?
Have not been keeping up with releases, first in a long time, 20180910 flashed today

```
INIT : Booting version: mega-20180910 (ESP82xx Core 00000000, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
101 : INIT : Warm boot #4 - Restart Reason: Software/System restart
103 : FS   : Mounting...
109 : FS   : Mount successful, used 75802 bytes of 113201
437 : CRC  : program checksum       ...OK
444 : CRC  : SecuritySettings CRC   ...OK 
463 : INIT : Free RAM:27832
464 : INIT : SPI not enabled
487 : INFO : Plugins: 46 [Normal] (ESP82xx Core 00000000, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
488 : EVENT: System#Wake
496 : WIFI : Set WiFi to STA
529 : WIFI : Connecting MAD_MOB attempt #0
542 : EVENT: System#Boot
18:16:18: 1879 : WD   : Uptime 0 ConnectFailures 0 FreeMem 25728
18:16:21: 4322 : WIFI : Connected! AP: MAD_MOB (18:90:D8:AC:0F:D8) Ch: 7 Duration: 3792 ms
4324 : EVENT: WiFi#ChangedAccesspoint
4359 : WIFI : DHCP IP: 192.168.0.103 (Strawberry-NEW-6) GW: 192.168.0.254 SN: 255.255.255.0   duration: 36 ms
4371 : EVENT: WiFi#Connected
4377 : Webserver: start
4471 : MQTT : Intentional reconnect
4529 : MQTT : Connected to broker with client ID: ESPClient_DC:4F:22:0C:C7:F0
4530 : Subscribed to: domoticz/out
4532 : EVENT: MQTT#Connected
18:16:22: 5424 : Current Time Zone: STD time start: 2018-04-01 01:00:00 offset: 600 min
5425 : EVENT: Time#Initialized
5431 : EVENT: Clock#Time=Wed,18:16
18:56:32: 2415475 : MQTT : Connected to broker with client ID: ESPClient_DC:4F:22:0C:C7:F0
2415476 : Subscribed to: domoticz/out
2415478 : EVENT: MQTT#Connected
20:22:17: 7559897 : MQTT : Failed to connect to broker
20:22:23: 7565943 : MQTT : Failed to connect to broker
20:22:24: 7566966 : WD   : Uptime 126 ConnectFailures 6 FreeMem 18560
20:22:51: 7594029 : WD   : Uptime 127 ConnectFailures 10 FreeMem 18560
20:23:19: 7621879 : WD   : Uptime 127 ConnectFailures 12 FreeMem 18560
20:24:08: 7671114 : MQTT : Failed to connect to broker
20:24:49: 7711879 : WD   : Uptime 129 ConnectFailures 18 FreeMem 18560
20:25:10: 7733182 : MQTT : Failed to connect to broker
20:25:19: 7741880 : WD   : Uptime 129 ConnectFailures 20 FreeMem 18560
20:25:49: 7771879 : WD   : Uptime 130 ConnectFailures 22 FreeMem 18560
20:26:19: 7801879 : WD   : Uptime 130 ConnectFailures 24 FreeMem 18560
20:26:49: 7832304 : WD   : Uptime 131 ConnectFailures 26 FreeMem 18560
20:27:19: 7861879 : WD   : Uptime 131 ConnectFailures 28 FreeMem 18432
20:27:49: 7891879 : WD   : Uptime 132 ConnectFailures 30 FreeMem 18432
20:28:19: 7921879 : WD   : Uptime 132 ConnectFailures 32 FreeMem 18432
20:28:49: 7951879 : WD   : Uptime 133 ConnectFailures 34 FreeMem 18432
20:29:19: 7982477 : WD   : Uptime 133 ConnectFailures 36 FreeMem 18432
20:29:50: 8013511 : WD   : Uptime 134 ConnectFailures 38 FreeMem 18560
20:30:21: 8044540 : WD   : Uptime 134 ConnectFailures 40 FreeMem 18432
20:30:52: 8075576 : WD   : Uptime 135 ConnectFailures 42 FreeMem 18432
20:31:02: 8085568 : EVENT: Clock#Time=Wed,20:31
20:31:23: 8106621 : WD   : Uptime 135 ConnectFailures 44 FreeMem 18328
20:31:54: 8137654 : WD   : Uptime 136 ConnectFailures 46 FreeMem 18432
20:32:01: 8144629 : EVENT: Clock#Time=Wed,20:32
20:32:24: 8167659 : WD   : Uptime 136 ConnectFailures 48 FreeMem 18328
20:32:55: 8198694 : MQTT : Failed to connect to broker
20:33:01: 8204724 : EVENT: Clock#Time=Wed,20:33
20:33:27: 8229778 : MQTT : Failed to connect to broker
20:33:53: 8255848 : MQTT : Connected to broker with client ID: ESPClient_DC:4F:22:0C:C7:F0
8255850 : Subscribed to: domoticz/out
8255851 : EVENT: MQTT#Connected
20:34:01: 8263782 : EVENT: Clock#Time=Wed,20:34
20:34:27: 8289850 : EVENT: Time#Set

```

Oxyandy, 2018-09-12:
The logs used to log the same as weblog shows here
[image]

Sasch600xt, 2018-09-12:
ok......so maybe this time it´s not me :)

Oxyandy, 2018-09-12:
Flashed mega-20180908 over the top, node refused to connect to wifi 
SO issued reset command via serial terminal connected via cable..
Expected text was seen
```
> reset
112086 : Command: reset
RESET: Resetting factory defaults...
22:19:00: RESET: Warm boot, reset count: 0
RESET: formatting...
22:19:02: RESET: formatting done...
22:19:06: 119227 : FILE : Saved config.dat
119549 : FILE : Saved security.dat
RESET: Succesful, rebooting. (you might need to press the reset button if you've justed flashed the firmware)

```
Still remembered secondary network.. huh ?
Reset command again.. this time seemed to have worked - but 
`WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1`
AP can never be found..
`>wifiscan`
Shows my local wifi networks OK
Yeah, yeah I know.. not directly related the original post - just saying it as I see it.

Sasch600xt, 2018-09-22:
so here are more informations:

- no plugin
- Domoticz HTTP Controller
- Getting uptime every 60 seconds by sending over controller.
- No data recieved / Uptime freeze at "0"

To reproduce, simple use attached config.
Hardware is a Lolin V3 Board.
I used "normal" 4Mb image.

At the moment i use the image from 20180826 without a problem.

Have a good one

[config.zip](https://github.com/letscontrolit/ESPEasy/files/2408057/config.zip)

TD-er, 2018-09-22:
You write "no plugin", but what should deliver the data then?
I will load your config to test.

TD-er, 2018-09-22:
I loaded your settings, but the LED on my node starts lighting up and the boot just stops.
So apparently it is not working well.

Could you please show some screenshots or something that is supposed to send the system information.
If you're using a controller, there must be some plugin (Generic sysinfo?) collecting the data.
Or maybe some rule to publish something.
So please give more info.

In the mean time, I have to wipe my test node with a blank file and start over. No idea what's going on with your config file.
It doesn't contain much info, since it is only 259 bytes compressed.

Sasch600xt, 2018-09-23:
by the way it is also not working for me by starting from scratch without a config file. So simple try to send Sysinfo Uptime with Domoticz HTTP Controller.

TD-er, 2018-09-23:
I just added it to my config and it just works.
Could you remove the Domoticz HTTP controller and add it again?

Sasch600xt, 2018-09-23:
when i did delete the controller i ended up in a bootloop.
Also after a reset of the hardware i stuck in a bootloop.

So i used 4Mb blank image to sart from scratch.
i did also try the new 20180923 image (normal,4096)

After i set up my local WIFI by setup, i added a contoller "Domoticz HTTP" and added a plugin (Sysinfo,uptime) From this moment on uptime freezed.

Do you have the same hardware i have ? Lolin V3 Board ?

i did try this at a few boards i have here, but all are Lolin V3 boards.

So the bug must comming from a much erlyer release......as i said, 20180826 is working fine.

Greetz
Sascha

Sasch600xt, 2018-09-23:
feels free to "see" it with your eyes by using teamviewer with me.....

TD-er, 2018-09-23:
I tested it here on a NodeMCU board I have been using for development for the last year (that flash chip must be worn out I guess ;) )
But I don't think it will be related to the hardware.

I did notice you are using a non-default port for the Domoticz HTTP controller.
There has been a change in setting ports (outgoing ports set at random) Maybe something is wrong here?
Do you use some kind of router to portforward your connections to the Domoticz server?

Sasch600xt, 2018-09-23:
well.....i use IP-Symcon for Houseautomation. In our forum i found a reciever-script for Domoticz messages.
So i have 3 nodes at the moment, every node has a different port (1024,1025,1026).

This is how a reciever Script looks like:
`<?

//IPS_LogMessage("Sender ", $_IPS['SENDER']);

if ($_IPS['SENDER'] == 'RegisterVariable'){
    $data = $_IPS['VALUE'];

    $Prefix1 = "&idx=";                                    // damit fängt der Datensatz an
    $Suffix1 = "&";
    $Prefix2 = "&svalue=";                                    // damit fängt der Datensatz an
    $Suffix2 = "HTTP/1.1";
    $Prefix3 = "&switchcmd=";                                    // damit fängt der Datensatz an
    $Suffix3 = "HTTP/1.1";

    //$Ende1nachZeichen = 2;                                       // und ist xx Zeichen lang
    $AnfangID = strlen($Prefix1)+(strpos($_IPS['VALUE'], $Prefix1));
    $EndeID = strpos($_IPS['VALUE'], $Suffix1, $AnfangID+1);
    $ID = substr($_IPS['VALUE'], $AnfangID, $EndeID-$AnfangID);

    $AnfangValue = strlen($Prefix2)+(strpos($_IPS['VALUE'], $Prefix2));
    $EndeValue = strpos($_IPS['VALUE'], $Suffix2, $AnfangValue);
    $Value = substr($_IPS['VALUE'], $AnfangValue, $EndeValue-$AnfangValue-1);

    $AnfangSw = strlen($Prefix3)+(strpos($_IPS['VALUE'], $Prefix3));
    $EndeSw = strpos($_IPS['VALUE'], $Suffix3, $AnfangSw);
    $Switch = substr($_IPS['VALUE'], $AnfangSw, $EndeSw-$AnfangSw-1);
    //IPS_LogMessage("Switch roh: ", $Switch);

    if($Switch == "On"){
       $Switch = true;
   }
    else{
         $Switch = false;
   }
	

    switch ($ID) {
	
		 case "1": // Status Relais 01
		 
		     if(strpos($Value,"On")!==false) {
			 SetValueBoolean(42939 /*[Entwicklungsbereich Sascha\HTTP Commands Relais\Programm/Scripte\ESP Easy Gartenhaus\Status Relais\Relais 01]*/,1);
			 }
			 if(strpos($Value,"Off")!==false) {
			 SetValueBoolean(42939 /*[Entwicklungsbereich Sascha\HTTP Commands Relais\Programm/Scripte\ESP Easy Gartenhaus\Status Relais\Relais 01]*/,0);
			 }
			 break;
		 
		 case "2": // Status Relais 02
		
		 	 if(strpos($Value,"On")!==false) {
			 SetValueBoolean(12309 /*[Entwicklungsbereich Sascha\HTTP Commands Relais\Programm/Scripte\ESP Easy Gartenhaus\Status Relais\Relais 02]*/,1);
			 }
			 
			 if(strpos($Value,"Off")!==false) {
			 SetValueBoolean(12309 /*[Entwicklungsbereich Sascha\HTTP Commands Relais\Programm/Scripte\ESP Easy Gartenhaus\Status Relais\Relais 02]*/,0);
			 }
			 break;
			 
		case "3": // Status Relais 03
		
		 	 if(strpos($Value,"On")!==false) {
			 SetValueBoolean(30789 /*[Entwicklungsbereich Sascha\HTTP Commands Relais\Programm/Scripte\ESP Easy Gartenhaus\Status Relais\Relais 03]*/,1);
			 }
			 
			 if(strpos($Value,"Off")!==false) {
			 SetValueBoolean(30789 /*[Entwicklungsbereich Sascha\HTTP Commands Relais\Programm/Scripte\ESP Easy Gartenhaus\Status Relais\Relais 03]*/,0);
			 }
			 break;
			 
		case "4": // Status Relais 04
		
		 	 if(strpos($Value,"On")!==false) {
			 SetValueBoolean(44012 /*[Entwicklungsbereich Sascha\HTTP Commands Relais\Programm/Scripte\ESP Easy Gartenhaus\Status Relais\Relais 04]*/,1);
			 }
			 
			 if(strpos($Value,"Off")!==false) {
			 SetValueBoolean(44012 /*[Entwicklungsbereich Sascha\HTTP Commands Relais\Programm/Scripte\ESP Easy Gartenhaus\Status Relais\Relais 04]*/,0);
			 }
			 break;
			 
		case "5": // Status Relais 05
		
		 	 if(strpos($Value,"On")!==false) {
			 SetValueBoolean(45432 /*[Entwicklungsbereich Sascha\HTTP Commands Relais\Programm/Scripte\ESP Easy Gartenhaus\Status Relais\Relais 05]*/,1);
			 }
			 
			 if(strpos($Value,"Off")!==false) {
			 SetValueBoolean(45432 /*[Entwicklungsbereich Sascha\HTTP Commands Relais\Programm/Scripte\ESP Easy Gartenhaus\Status Relais\Relais 05]*/,0);
			 }
			 break;
			 
		case "6": // Status Relais 06
		
		 	 if(strpos($Value,"On")!==false) {
			 SetValueBoolean(25636 /*[Entwicklungsbereich Sascha\HTTP Commands Relais\Programm/Scripte\ESP Easy Gartenhaus\Status Relais\Relais 06]*/,1);
			 }
			 
			 if(strpos($Value,"Off")!==false) {
			 SetValueBoolean(25636 /*[Entwicklungsbereich Sascha\HTTP Commands Relais\Programm/Scripte\ESP Easy Gartenhaus\Status Relais\Relais 06]*/,0);
			 }
			 break;
			 
		case "7": // Status Relais 07
		
		 	 if(strpos($Value,"On")!==false) {
			 SetValueBoolean(11700 /*[Entwicklungsbereich Sascha\HTTP Commands Relais\Programm/Scripte\ESP Easy Gartenhaus\Status Relais\Relais 07]*/,1);
			 }
			 
			 if(strpos($Value,"Off")!==false) {
			 SetValueBoolean(11700 /*[Entwicklungsbereich Sascha\HTTP Commands Relais\Programm/Scripte\ESP Easy Gartenhaus\Status Relais\Relais 07]*/,0);
			 }
			 break;
			 
		case "8": // Status Relais 08
		
		 	 if(strpos($Value,"On")!==false) {
			 SetValueBoolean(12562 /*[Entwicklungsbereich Sascha\HTTP Commands Relais\Programm/Scripte\ESP Easy Gartenhaus\Status Relais\Relais 08]*/,1);
			 }
			 
			 if(strpos($Value,"Off")!==false) {
			 SetValueBoolean(12562 /*[Entwicklungsbereich Sascha\HTTP Commands Relais\Programm/Scripte\ESP Easy Gartenhaus\Status Relais\Relais 08]*/,0);
			 }
			 break;
	
         case "101": // Uptime
             SetValueInteger(42387 /*[Entwicklungsbereich Sascha\HTTP Commands Relais\Programm/Scripte\ESP Easy Gartenhaus\Uptime in Minuten]*/, $Value);
         break;
         
    }
}

?>`

Sasch600xt, 2018-09-23:
that looks strange., but i did use "insert code" only once in my las post.
Half of the code is formatted, other half not.

Grovkillen, 2018-09-23:
Yes you have those code parts "too close to the left", i.e. without the code format that GitHub expects... only ad some spaces and it'll format properly.

TD-er, 2018-09-23:
If you stop trying to edit the post, I can fix the layout for you.
I tried for 4 times already, but it was changed before I could press the save button.

Sasch600xt, 2018-09-23:
i am sorry....same i did here :(

TD-er, 2018-09-23:
You just have to add three times the left tick quote on a single line at start and end of a multi-line code block.
The single left tick quote is for inline code blocks.

Sasch600xt, 2018-09-23:
ok, got it and thank you :)

TD-er, 2018-09-23:
```
GET /json.htm?type=command&param=udevice&idx=209&nvalue=0&svalue=719.00&rssi=8&battery=38 HTTP/1.1
Host: domoticz:8080
User-Agent: ESP Easy/20102/Sep 23 2018 00:44:42
Connection: close
```
As an example the output of such a request from one of my nodes.

A quick look at your script suggests the order of `&svalue` and `&nvalue` may have changed ?
Looks like you used [this post](https://www.symcon.de/forum/threads/31373-ESPeasy?s=c1a5f53b8629fe3e57535a9645f60343&p=294666#post294666) I guess?

And just curious, why do you run several instances of the same script on different ports?
To differentiate the sender, you can simply define some IDX value and grab that from the idx parameter in the GET string. (209 in my example)
Those IDX values can be set at the plugin where you check what controller to use.

Sasch600xt, 2018-09-23:
You are right, i will change the IDX values so i only have one recieving script running.
Thank you for that hint.

And second yes, i used this post to get it to work.

> A quick look at your script suggests the order of `&svalue` and `&nvalue` may have changed ?

I will try to fix this in my script, thank you very much for this hint

i will report my success or fail :)

TD-er, 2018-09-23:
I don't know PHP, but to me it seems parsing parameters from a GET url is one of the most common things for PHP to do.
So maybe [this example](https://stackoverflow.com/a/27689796/8708166) is a better approach.
Then you can simply address the values from an array.

```php
$parsed = parse_url($data);
$idx = $parsed['idx'];
$svalue = $parsed['svalue'];
```

And so on.
That's much better readable code and also doesn't care if the order of parameters in the URL changes or is left out.

Sasch600xt, 2018-09-23:
ok...thank you very much.
I am not a pro, but i try to change my reciever script.
I hope i will understand it :) (not your fault)

Sasch600xt, 2018-09-28:
ok.....me again (last chance for switching off your screen :))

Now i have a brand new receiver script.
But still no luck with receiving HTTP commands from the esp.

If i use commandline in "tool" and send 
```SendToHTTP` 192.168.123.10,8080,/json.htmtype=command&param=switchlight&idx=17&switchcmd=On```

i can receive once the message. After this short moment of glory teh esp freezes and i have to do an hard reset. I can repeat this and the behavior is always the same.

Thanks for help
Sascha

TD-er, 2018-10-02:
Since this seems to be a duplicate of https://github.com/letscontrolit/ESPEasy/issues/1830 I will close this one.

closed 2018-10-02 by TD-er