letscontrolit/ESPEasy #1735: P009

andrea-daru, 2018-09-11:
BUILD: **mega-20180910**
PLUGIN: **P009** (MCP)

While a pin is set to Input mode (By creating a device "_Switch input - MCP23017_") the status it's not correctly retrieved.
I'll try to explain the behaviour: 
_If after a power cycle the state on the pin it's 1, it doesn't update the status exposed in the device page and it seems that it doesn't recognize any other status change.
it work only if after a reboot the status on the pin is 0._
The expected behaviour it's that on System Boot the status it's updated with the real status on the pin.

any help will be appreciated
thank you in advance for the great job done in thi project!

andrea-daru, 2018-10-28:
Are there any news about this bug?
Now after a power cycle it always show state 1 on the pin configured as input.

TD-er, 2018-10-28:
What version are you now using?
Last week(s) a number of bugfixes are merged for these GPIO expanders.

giig1967g, 2018-10-28:
@andrea-daru 
I should have fixed this starting from version 20181027.

andrea-daru, 2018-10-29:
Thanks for your reply! Now I’m using version mega-20181028.
If I reset the esp while an mcp23017 input it’s in state 1 it doesn’t update the state anymore. I can’t figure out what’s the problem.
The only way to make it work correctly it’s to force the mcp23017 input to 0 (in my case it means that I’ve to switch off all the lights in my house) before reboot.

giig1967g, 2018-10-29:
Will check it

giig1967g, 2018-10-29:
@andrea-daru 
a) was it working correctly in the past?
b) what is the configuration of the MCP port? Normal or inversed?
c) did you set "send boot state"?
d) did you try to set `MCPGPIO,portnumber,2` in `System#Boot` event? (it sets the port as an inupt).

andrea-daru, 2018-10-29:
@giig1967g  
a) no it wasn't working
b) Normal
c) tried in both configuration
d) Yes I've tried also setting `MCPGPIO,portnumber,2 `in Rules without any luck :
```
on System#Boot do
MCPGPIO,9,2
MCPGPIO,10,2
MCPGPIO,11,2
MCPGPIO,12,2
endon
```

andrea-daru, 2018-10-29:
Basically I’ve a Double phase Step Relay (Finder 260280240000 1) which it’s driven by another Relay that send a 50ms pulse to make the Step Relay change it’s position.
I use the second Phase of the Step Relay to get back the status because I use also wall mounted button to toggle the Step Ralay.

I've seen that if the "status" pin it's high (1) during reboot it will stop to fire EVENT:
CORRECT (pin low after reboot)
```
EspEasy: MCP  : GPIO 6 Pulsed for 50 mS -> toggle step relay
EspEasy: MCP  : Port=10 State=1 Output value=1 -> port 10 it's the status pin
EspEasy: EVENT: stato_luce_cucina#Switch=1.00
EspEasy: EVENT: stato_luce_cucina#Switch=1.00 Processing time:66 milliSeconds
EspEasy: MQTT : /ESP_CORE/stato_luce_cucina/Switch 1
```
WRONG (pin high after reboot)
`
EspEasy: MCP  : GPIO 6 Pulsed for 50 mS  -> toggle step relay
`
No event fired

giig1967g, 2018-10-29:
ok. So the command that you use is mcppulse.
Can you post the rules file?

andrea-daru, 2018-10-29:
I use the MCPPULSE to toggle the step relay (let's say on MCP port 6).
Then I expect to get the status update on port  on port 10

This is the port "configuration"
light 1 MCPPULSE 50ms -> port 5
light 1 status/input -> port 9

light 2 MCPPULSE 50ms -> port 6
light 2 status7input -> port 10

light 3 MCPPULSE 50ms -> port 7
light 3 status/input -> port 11

light 4 MCPPULSE 50ms -> port 8
light 4 status/input -> port 12

This is the only rule that I've:
```
on System#Boot do
MCPGPIO,9,2
MCPGPIO,10,2
MCPGPIO,11,2
MCPGPIO,12,2
endon
```

uzi18, 2018-10-29:
try something like this:
MCPGPIO,9,0
MCPGPIO,9,2
MCPGPIO,10,0
MCPGPIO,10,2
...

giig1967g, 2018-10-29:
I don't understand where you call the `mcppulse` command from.

andrea-daru, 2018-10-29:
@uzi18 
Tried but no luck!

andrea-daru, 2018-10-29:
@giig1967g 
I call the `MCPPULSE` via `MQTT` command with an external agent (_OpenHAB_)

giig1967g, 2018-10-29:
can you post the full command please.
Otherwise it's difficult to help.

andrea-daru, 2018-10-29:
This is the OpenHAB item configuration:
```
Switch Cucina_Luce "Luce cucina [%s]" {mqtt=">[mosquitto:/ESP_CORE/cmd:command:ON:MCPPULSE,6,0,50], >[mosquitto:/ESP_CORE/cmd:command:OFF:MCPPULSE,6,0,50], <[mosquitto:/ESP_CORE/stato_luce_cucina/Switch:state:ON:1], <[mosquitto:/ESP_CORE/stato_luce_cucina/Switch:state:OFF:0"}
```
This is the relevant part of the configuration in ESPEasy WEB UI
[image]

Basically to switch ON and OFF the light i send the same command (PULSE) and to update the status i wait that ESPEasy fire the event when it detect the status change on  the pin

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er