letscontrolit/ESPEasy #1739: JSON output temperature

norimikan, 2018-09-13:
Hi

I use Sonoff RF and DHT22 for temperature.
ESP Easy Mega 0910

I try get temperature from Json use 
http://192.168.xxx.xxx/json?tasknr=3&view=sensorupdate

{
"TaskValues": [
{"ValueNumber":1,
"Name":"temperature",
"NrDecimals":2,
"Value":30.50
},
{"ValueNumber":2,
"Name":"humidity",
"NrDecimals":2,
"Value":70.80
}],
"TTL":60000,
"TaskEnabled":"true",
"TaskNumber":3
}

But I like get Json only temperature and humidity like this.
temperature: 30.50
humidity: 70.80
(Same as ESPEasyR120)

Please advice to me How to get Json output.

Thankyou

Grovkillen, 2018-09-13:
What you're suggesting is not proper JSON since that mean that the receiving end must know the JSON path. So I suggest you take this to the forum and explain your exact need and environment.

closed 2018-09-13 by Grovkillen

norimikan, 2018-09-13:
I'll apologize for my improper question.
(And my poor english)

I'm use ESPEasy mega and show temperature in my iphone.(iOS)
use homebridge and homebridge-http-temperature-humidity.

I'll make an effort in order to settle it by oneself.

TD-er, 2018-09-13:
@norimikan What @Grovkillen meant was to open an issue on the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) with these questions.
Questions like these are more likely to be about "how do I do this?" and not about "this is a bug and needs to be fixed".

norimikan, 2018-09-14:
Thankyou for advice.

I'm register and question in forum

Sorry for my poor english and misunderstand.