letscontrolit/ESPEasy #1743: no two decimal places are transferred to thingspeak

AxelMilb, 2018-09-14:
Since version 20180908, no two decimal places are transferred to thingspeak & emoncms or very rarely.
The values are correct in the ESP under Devices.

Sorry for my bad english

TD-er, 2018-09-14:
Did you also try to set it to 3 decimals, just to see if that also makes a difference? 
That makes it less likely it will be rounded to exactly .00 or .x0

TD-er, 2018-09-14:
The controllers both format the values at 2 decimals.
So I guess it is removing the trailing zeroes.
You could use the formula part of the plugin to add 0.01 to see if it will output 2 decimals again.

AxelMilb, 2018-09-15:
Unfortunately, the problem solves only sporadically with %value%+0.01

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er