letscontrolit/ESPEasy #1748: Boot loop with 20180915 release

giig1967g, 2018-09-15:
Hi,
I am getting boot loops with latest version.

```
Free Mem | 18984 (18072 - ruleMatch2)
-- | --
Boot | Manual reboot (17)
Reset Reason | Hardware Watchdog

```
Reverted to 20180914 and everything works fine again.

TD-er, 2018-09-15:
I added something to disable plugins/controllers one by one when it detects a bootloop.
So eventually it should stop looping. Unless ofcourse it is related to a rule. Maybe I should add to disable rules also?
Most merges are yours by the way :)

TD-er, 2018-09-15:
What plugins/controllers/rules are you using?

micropet, 2018-09-15:
The version mega-20180915 from this morning runs only a few seconds.
Then a reboot takes place.

Units without hardware will run for a few minutes.
The unit with attached hardware boot right away,

TD-er, 2018-09-15:
I have one with OLED and BME280 attached and it runs for > 1 hour now (just flashed it an hour ago)
The main thing changed in the last build is the handling of commands (no longer queued) and the rules related PR's that were ready.

TD-er, 2018-09-15:
Could you test by changing the `yield()` on line 2168 of Misc.ino into `backgroundtasks()` ?
Or else we should get back to an older core library, since these hardware watchdog issues are something that got worse in the last few weeks/months

micropet, 2018-09-15:
So, we were with the granddaughter at a city festival.

I have flashed the unit again with the version from 9.9.
This is always very cumbersome, because I have to unscrew the case to interrupt the RX GPIO3 for the MH-Z19 TX from the Wemos. Otherwise I can not flash by cable.

Yes, I have a MH-Z19 and a PMS5003 serial connected.
But the PMS5003 has never worked.

micropet, 2018-09-15:
>> Could you test by changing the yield() on line 2168 of Misc.ino into backgroundtasks() ?

I have done it und flasht 2 Units.

giig1967g, 2018-09-15:
Hi all,
I tried to compile the version myself without changing anything, and the resulting version does not reboot. So maybe it's a compile issue?

giig1967g, 2018-09-15:
Hi,
I just compiled a version removing my commit #1742, but the resulting firmware still reboot after few seconds.

micropet, 2018-09-15:
With backgroundtasks () it runs on two units already 5 hours.
The units have no hardware connected.

clumsy-stefan, 2018-09-15:
had similar issues see also my comment in #1720 ...

also I had some units with corrupted settings files after flashing/rebooting..

the sonoff-pow's I cant get to boot at all... trying to change the line 2168 now, we'll see if that changes anything..

TD-er, 2018-09-15:
If it really is this disastrous, maybe it is better to remove this build and revert the changes of yesterday to a separate Branch to see what's going on

micropet, 2018-09-15:
With backgroundtasks (), two units now run for 7 hours. (without hardware, but with rules)

giig1967g, 2018-09-15:
> If it really is this disastrous, maybe it is better to remove this build and revert the changes of yesterday to a separate Branch to see what's going on

I agree.
Let's revert and merge every single change one by one.

clumsy-stefan, 2018-09-16:
after fixing the config files as descibed above, all units run now since 10h+ except the sonoff-pow's which I can flash (via serial) but not get to boot... reset every 10sec. or so.. still trying to find what exactly causes this...

clumsy-stefan, 2018-09-16:
quick update: using version from 20180915 and reverting commit #38af765935a3584796f01fe55871b092632b21c2 ([Scheduler] Disable command scheduler) solves at least my issues (bootloop) with the sonoff_pow device! (and probably other strange issues)...
could be a start point to look into..

clumsy-stefan, 2018-09-21:
I guess that's solved with #1770 and could be cloesd?

TD-er, 2018-09-21:
Yep, forgot this one yesterday.

closed 2018-09-21 by TD-er