letscontrolit/ESPEasy #1755: Boor loop on 20180916 / Sonoff 4ch RF

norimikan, 2018-09-16:
My Sonoff 4ch RF is Boot loop in 20180916.
(20180915 is boot loop same)

After add this rule, system is boot loop

On System#Boot do
timerSet,1,30
endon

On Rules#Timer=1 do
if %systm_hm% > %sunrise% and %systm_hm% < %sunset%
event 12PowerOff
event 5PowerOff
event 4PowerOff
else
event 12PowerOn
event 5PowerOn
event 4PowerOn
endif
EndOn

Use system 20180914 is not boot loop.

Thankyou.

TD-er, 2018-09-16:
Can you give the output of %sunrise%?
See the new system variable page in tools

giig1967g, 2018-09-16:
Hi @TD-er: ,
Instead of reverting changes, is it possible to "restore" the code to the one dated 20180914?
The reason is because it could even be a github bug.... 
So restoring the code is safer that reverting the changes.

In order to find the root cause, my suggestion is to restore the previous version without changing anything (not even changing the yeald() to runbackgroundtasks()) and then restart from there.

Otherwise we risk to sum up mistakes...

What do you think?

TD-er, 2018-09-16:
I can also do that and I will do it, since I will be going away for a few days. (meeting Jimmy @Grovkillen in Sweden)

norimikan, 2018-09-16:
Dear TD-er

Sunrise time is ok.

%unixtime% | 1537087533 | 1537087533
%sunset% | 18:17 | 18%3a17
%sunset-1h% | 17:17 | 17%3a17
%sunrise% | 6:05 | 6%3a05
%sunrise+10m% | 6:15 | 6%3a15

In case use %sunset% in rule not boot loop.

On Clock#Time=All,%sunset-145m% do
inputswitchstate 4,1
inputswitchstate 5,1
inputswitchstate 6,1
gpio,12,1
gpio,5,1
gpio,4,1
EndOn

norimikan, 2018-09-16:
Dear TD-er

log for boot loop.
Sep 16 12:04:30 Sonoff 4ch Pro R2 EspEasy: EVENT: Rules#Timer=1
Sep 16 12:04:30 Sonoff 4ch Pro R2 EspEasy: ReplacementString SunTime: %sunset% offset: 0
Sep 16 12:04:30 Sonoff 4ch Pro R2 EspEasy: ReplacementString SunTime: %sunrise% offset: 0
Sep 16 12:04:30 Sonoff 4ch Pro R2 EspEasy: Lev.1: [if 12:04 > 6:05 and 12:04 < 18:17]=true
Sep 16 12:04:30 Sonoff 4ch Pro R2 EspEasy: ACT  : inputswitchstate 4,0
Sep 16 12:04:30 Sonoff 4ch Pro R2 EspEasy: ACT  : inputswitchstate 5,0
Sep 16 12:04:30 Sonoff 4ch Pro R2 EspEasy: ACT  : inputswitchstate 6,0
Sep 16 12:04:30 Sonoff 4ch Pro R2 EspEasy: ACT  : event 12PowerOff
Sep 16 12:04:30 Sonoff 4ch Pro R2 EspEasy: Command: event
Sep 16 12:04:30 Sonoff 4ch Pro R2 EspEasy: EVENT: 12PowerOff
Sep 16 12:04:43 Sonoff 4ch Pro R2 EspEasy: Webserver: start **<<boot**
Sep 16 12:04:43 Sonoff 4ch Pro R2 EspEasy: WIFI  : Arduino wifi status: WL_CONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_SERVICES_INITIALIZED
Sep 16 12:04:43 Sonoff 4ch Pro R2 EspEasy: NTP  : NTP host ntp.nict.jp (133.243.238.244) queried
Sep 16 12:04:43 Sonoff 4ch Pro R2 EspEasy: NTP  : NTP replied: 111 mSec
Sep 16 12:04:43 Sonoff 4ch Pro R2 EspEasy: Current Time Zone: STD time start: 2018-10-28 03:00:00 offset: 420 min
Sep 16 12:04:43 Sonoff 4ch Pro R2 EspEasy: EVENT: Time#Initialized
Sep 16 12:04:43 Sonoff 4ch Pro R2 EspEasy: ReplacementString SunTime: %sunset% offset: 0
Sep 16 12:04:43 Sonoff 4ch Pro R2 EspEasy: ReplacementString SunTime: %sunrise% offset: 0
Sep 16 12:04:43 Sonoff 4ch Pro R2 EspEasy: EVENT: Time#Initialized Processing time:75 milliSeconds
Sep 16 12:04:43 Sonoff 4ch Pro R2 EspEasy: EVENT: Clock#Time=Sun,12:04
Sep 16 12:04:43 Sonoff 4ch Pro R2 EspEasy: ReplacementString SunTime: %sunset% offset: 0
Sep 16 12:04:43 Sonoff 4ch Pro R2 EspEasy: ReplacementString SunTime: %sunrise% offset: 0
Sep 16 12:04:43 Sonoff 4ch Pro R2 EspEasy: EVENT: Clock#Time=Sun,12:04 Processing time:70 milliSeconds

TD-er, 2018-09-16:
OK, thanks, that is really helpful.

Sasch600xt, 2018-09-16:
> Hi @TD-er: ,
> Instead of reverting changes, is it possible to "restore" the code to the one dated 20180914?

At least for Dominicz HTTP Commands 20180914 has also bugs...... so maybe not a save way to restore it from there....

i had to go back to 20180826, and there is all working with Dominicz HTTP Commands......just to let you know.

Have a great Day and lot of fun in Sweden TDer :)

TD-er, 2018-09-16:
How do you use the Domoticz HTTP commands?
As controller or from rules?

Sasch600xt, 2018-09-16:
both......and both it is not working

norimikan, 2018-09-16:
Dear TD-er

Sorry many time comment.

I use button hardware(GPIO0, 9, 10, 14) and system is reboot.

Log reboot is 

ep 16 16:08:15 ESP_Easy EspEasy: SW   : Switch state 1 Output value 0
Sep 16 16:08:15 ESP_Easy EspEasy: Read settings: TaskSettings index: 4
Sep 16 16:08:15 ESP_Easy EspEasy: EVENT: Button1#State1=0.00
Sep 16 16:08:15 ESP_Easy EspEasy: ReplacementString SunTime: %sunset% offset: 0
Sep 16 16:08:15 ESP_Easy EspEasy: ReplacementString SunTime: %sunrise% offset: 0
Sep 16 16:08:15 ESP_Easy EspEasy: Read settings: TaskSettings index: 0
Sep 16 16:08:15 ESP_Easy EspEasy: Read settings: TaskSettings index: 1
Sep 16 16:08:15 ESP_Easy EspEasy: Read settings: TaskSettings index: 2
Sep 16 16:08:15 ESP_Easy EspEasy: Read settings: TaskSettings index: 3
Sep 16 16:08:15 ESP_Easy EspEasy: Read settings: TaskSettings index: 4
Sep 16 16:08:15 ESP_Easy EspEasy: DEBUG DEV: Parsed String='if 0'
Sep 16 16:08:15 ESP_Easy EspEasy: [if 0=0]=true
Sep 16 16:08:26 ESP_Easy EspEasy: Webserver: start  **<< Boot**
Sep 16 16:08:26 ESP_Easy EspEasy: WIFI  : Arduino wifi status: WL_CONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_SERVICES_INITIALIZED
Sep 16 16:08:26 ESP_Easy EspEasy: NTP  : NTP host 0.asia.pool.ntp.org (133.243.238.164) queried
Sep 16 16:08:26 ESP_Easy EspEasy: NTP  : NTP replied: 131 mSec
Sep 16 16:08:26 ESP_Easy EspEasy: Current Time Zone: STD time start: 2018-10-28 03:00:00 offset: 420 min
Sep 16 16:08:26 ESP_Easy EspEasy: EVENT: Time#Initialized
Sep 16 16:08:26 ESP_Easy EspEasy: ReplacementString SunTime: %sunset% offset: 0
Sep 16 16:08:26 ESP_Easy EspEasy: ReplacementString SunTime: %sunrise% offset: 0
Sep 16 16:08:26 ESP_Easy EspEasy: EVENT: Time#Initialized Processing time:63 milliSeconds
Sep 16 16:08:26 ESP_Easy EspEasy: EVENT: Clock#Time=Sun,16:08
Sep 16 16:08:26 ESP_Easy EspEasy: ReplacementString SunTime: %sunset% offset: 0
Sep 16 16:08:26 ESP_Easy EspEasy: ReplacementString SunTime: %sunrise% offset: 0
Sep 16 16:08:26 ESP_Easy EspEasy: EVENT: Clock#Time=Sun,16:08 Processing time:60 milliSeconds

Sasch600xt, 2018-09-16:
at least from 201800908 to newer versions.
20180826 is working fine.

giig1967g, 2018-09-16:
for me 20180914 is working fine and stable, apart the hardware watchdog problems generated from slow servers or specific routers...
Using MQTT with OpenHAB.

TD-er, 2018-09-16:
I think I found the issue with Sunrise/Sunset.

When offset != 0. it is using the internal member `tsRise` or `tsSet`
Otherwise it is using `sunRise` or `sunSet` .
And since the last issue started, these values may have gotten out of sync.

If someone could check this, that would be great.
I gotta fly now ;)

closed 2023-09-04 by TD-er