letscontrolit/ESPEasy #1759: lost updates of task-values

JohnMTu, 2018-09-16:
### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

Each rule on btn click-release is incrementing task value. But i observe lost updates.

### Expected behavior

I expect e.g. 77 on click, and 78 on release, then 79 on next click etc.

### Actual behavior

I get same value on press and subsequent release. E.g. 77 on click and 77 on release. Then 78 on next click and also on release. Then 79 on next click...

### Steps to reproduce

Configure rules as shown bellow, task to hold evnNr and longPress state and normal button

### System configuration

Hardware: ESP-12-F

ESP Easy version: mega-20180910

ESP Easy settings/screenshots: 

### Rules or log data

`on spalna3#btnState do
  TaskValueSet 10,1,[storage1#evtNr]+1

  if [spalna3#btnState] = 0.00
    TaskValueSet 10,4,0
    timerSet,3,1
    SendToUDP 10.99.0.12,514,btn=spalna3,evt=INITIAL_PRESS,seq=[storage1#evtNr]
  endif

  if [spalna3#btnState] = 1.00 AND [storage1#spalna3LongPress]=0
    timerSet,3,0
    SendToUDP 10.99.0.12,514,btn=spalna3,evt=SHORT_RELEASED,seq=[storage1#evtNr]
  endif

  if [spalna3#btnState] = 1.00 AND [storage1#spalna3LongPress]=1
    timerSet,3,0
    SendToUDP 10.99.0.12,514,btn=spalna3,evt=LONG_RELEASED,seq=[storage1#evtNr]
  endif
endon

on rules#timer=3 do
   timerSet,3,1
   TaskValueSet 10,1,[storage1#evtNr]+1
   TaskValueSet 10,4,1 // set it to long press

   SendToUDP 10.99.0.12,514,btn=spalna3,evt=HOLD,seq=[storage1#evtNr]
endon
`

uzi18, 2018-09-16:
please show us task 10 configuration, and button configuration

JohnMTu, 2018-09-16:
Here you are, nothing special inside i hope.

[image]

[image]

Thank you

giig1967g, 2018-09-17:
there is a known bug see #1712 that could affect your code.
You can not refer values modified in a rule until the rule is completed.

JohnMTu, 2018-09-19:
update: 
if i do just quick click-release, then both events are using same ID.
if don't release so quickly (but still under half second), then i get two different IDs.

TD-er, 2018-11-28:
Is this one now fixed?
Please test with latest build, which will be made tonight. (20181128)

JohnMTu, 2018-12-16:
will try to check it around xmas, Can i take any later build?

JohnMTu, 2019-01-26:
wanted to test this. Flashed ESPEasy_mega-20190121 to wemos nodemcu, but seems to have very unstable wifi in station mode. I was able to setup it as wifi station, while it was exposed as AP. But after it changed to client mode, i can usually ping it only once, until it disconnects.
Flashed back micropython to see if there is problem with board, but there it looks all fine.
Will take some month old flash to try again.

JohnMTu, 2019-01-26:
with ESP_Easy_mega-20181216_normal_ESP8266_4096.bin same result.

JohnMTu, 2019-01-26:
hm cannot edit old post, "edit" menu link not working. Also tried ESP_Easy_mega-20190121_minimal_ESP8266_1024_OTA.bin, same wifi issue.

TD-er, 2019-10-27:
Close this now, please re-open (or add a new one) if it is still an issue.

closed 2019-10-27 by TD-er