letscontrolit/ESPEasy #1761: Plugin P040_ID12 problem

rr10, 2018-09-17:
on this page: https://www.letscontrolit.com/wiki/index.php/Official_plugin_list
the rfid plugin P040_ID12 is among the normal plugins.
in the mega-20180914 release the plugin is not present neither in the normal release nor in the test version or developer

gelitoam, 2019-03-15:
Maybe I solved it... I noticed the file src/define_plugin_sets.h had no #define USES_P040 line. It jumped from 39 to 41:

#define USES_P039   // ID12
#define USES_P041   // NeoClock

And also _P039 is thermocouple, not ID12. So I edited the file this way:

#define USES_P039   //Thermocouple
#define USES_P040   // ID12

And now the plugin appears in the list.

Hope it helps. Regards.

TD-er, 2019-03-15:
OK, so there is an entry missing?
Will look at it.

TD-er, 2019-03-15:
Consider this one fixed.
@gelitoam Thanks for putting this one back on the attention radar.

gelitoam, 2019-03-15:
Glad to help :-)

closed 2019-10-10 by TD-er