letscontrolit/ESPEasy #1763: Rule domoticz/publish Battery MQTT Spam

jannnfe, 2018-09-18:
### Summarize of the problem/feature request

I would like to use rules to send fixed battery and RSSI values while changing the status of a relay to Domoticz.

Using this Rule:
`on Opener#Switch do
  Publish domoticz/in,{"idx":43,"nvalue":[Opener#Switch],"svalue":"","RSSI":7,"Battery":64}
endon`

### Expected behavior

Relay switches from 0 to 1 --> A rule triggers and an MQTT message is sent in which the battery and RSSI value is also transferred.
Same relay switches from 1 to 0 --> Another rule triggers and an MQTT message will be send in which the battery and RSSI value is also transferred.

### Actual behavior

The MQTT message is sent, but it is often sent in a second and it does not stop. It spammed the log of Domoticz. Actually, a message would have to be sent every time you change the ralais.

### Steps to reproduce

1. Create Switch input Device and Name it Opener
2. Set Idx in Device Settings to 43
3. Create Rule above

### System configuration

Hardware: Wemos D1 Mini

ESP Easy version: mega-20180818
Updating to the newest version is not an option because of the known MQTT Import Issue.

TD-er, 2018-09-18:
I don't think updating the firmware to a newer version will resolve these issues. (yet)
The problem is probably related to the commands using the scheduler.
I want to revert that, but judging other reported issues, that may cause other issues.

giig1967g, 2018-09-18:
Looking at the rule syntax, it seems to me that the rule should work fine.
The 1 second delay that you noticed is related to the asynch command handling.
The log spamming is not clear to me.
Please send an example.

giig1967g, 2018-09-18:
I suspect that the problem is in the interval of the switch.
Can you print the screenshot of the Lock button?

TD-er, 2018-09-18:
Maybe the delay of command handling (because it is handled from a queue) is causing issues?

jannnfe, 2018-09-18:
This are my settings:
[image]
[image]
[image]

giig1967g, 2018-09-18:
could it be that defining in two different devices the same GPIO-5 creates an endless loop?

jannnfe, 2018-09-18:
Unfortunately that makes no difference, because with all three rules, the same problem occurs. I think the error is on domoticz page but I can not reproduce it and I do not understand why that happens.

[image]

jannnfe, 2018-09-21:
Hey @giig1967g & @TD-er,
I have a little further informed and looked at other firmware, how it is regulated there. Is it possible that Tasmota changes the problem with a setting rule? See here: https://github.com/arendst/Sonoff-Tasmota/issues/2549#issuecomment-384510845
There, "Once": "ON" sends an MQTT message only once if the trigger matches. With "Once": "ON" the MQTT message is repeated.

On the ESPEasy Wiki is something about Subscribe Template and Publish Template.
Would not I come to the same goal as with the rules?
> Subscribe Template Used for MQTT subscription. Selecting a MQTT protocol will automatically fill this field
> Publish Template Used for MQTT publishing. Selecting a MQTT protocol will automatically fill this field It is also used for the Generic HTTP protocol. Create your custom template here.

Unfortunately the setting is missing in the last ESPEasy versions.
Thanks :)

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er