letscontrolit/ESPEasy #1766: No Power with Sonoff Pow R2 and ...hard_SONOFF_POW.bin

Spohky, 2018-09-19:
Hello everybody,

I've been trying to get my SONOFF Pow R2 up and running for a few weeks with different ESPEasy versions, unfortunately without success.

I tried it today with the version "ESP_Easy_mega-20180914_hard_SONOFF_POW.bin", unfortunately I do not get any values.

In addition, I found nothing on the Internet (Google) that has brought me further.

On the Pow R2 is a CSE77 ?? Unfortunately, the last two numbers are unreadable.
And a Winbond 25q32JVSIQ.

The Device: Energy (AC) - HLW8012 [TESTING]
and then the GPIO as they are selected below.

I hope you can help me, thank you!

Best regards
Spohky

TD-er, 2018-09-19:
The Pow R2 does not yet have a (tested) build.
I am not sure if the plugin needed for the POW R2 is included in the test or dev build.
The POW build was initially meant to be specific for the first version and due to lack of time I have not yet made that build to work with the R2.

Spohky, 2018-09-19:
Ok, I've tested the dev version (ESP_Easy_mega-20180914_dev_ESP8266_1024), here's the CSE7766 in it (Energy (AC) - CSE7766 [TESTING]).
I tested it once, seems to work.
I'll flash all my Pow R2 tomorrow or weekend and do an evaluation in FHEM.

I have a question about the value of "Pulse" of the CSE7766, if you can help me there, unfortunately, i found nothing.
Do 1000 pulses = 1Watt consumption?

Thank you for the help! 👍

TD-er, 2018-11-27:
It was already present for a while.
Now we have a (new) special purpose build for the Sonoff POW/POWr2. This build is called `hard_SONOFF_POW_4M`
Both setups are now included and all button/relay/rules to make some proper default are available in the new Factory Reset page as preset.

So I will close this issue.

closed 2018-11-27 by TD-er