letscontrolit/ESPEasy #1767: PCF8574 plugin and inversed logic doesn't work as expected in ESPEasy

MaliMrav, 2018-09-20:
I am using the ESPEasy 
Build | 20102  - Mega
Libraries | ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
with the MPRSx8 board (8 Ch. Relay board with PCF8574 control).

The PCF8574 Plugin (019) in normal (No Inversed Logic) reports 1 in the off state and 0 in the on state.
to switch it on I use "http://172.20.20.141/control?cmd=pcfgpio,69,0"
to switch it off I use "http://172.20.20.141/control?cmd=pcfgpio,69,1"

My assumption was if I use the "Inversed Logic" (and have it checked in the devices section i should be able to reverse the commands
i.e.
to switch it on I use "http://172.20.20.141/control?cmd=pcfgpio,69,1"
to switch it off I use "http://172.20.20.141/control?cmd=pcfgpio,69,0"

That isn't the case. Briefly the logic shows the correct "0" straight after a reboot of the device but when I issue the first command it goes back to normal logic.

Is this the expected behavior?

TD-er, 2018-09-20:
My initial guess would be "inverse logic" means the state of the observed **_electrical signal_** is inverted.
But I have to look into the code to see how it really works.

If all is inverted, it would be useless I guess, or at least really confusing.

uzi18, 2018-09-20:
how did you set inversed logic here?

sassod, 2018-09-20:
As I remember correctly this has been discussed before.

And if my memory is correct is that if the inverse selection is active only
applies to the pi signal itself (high/low), the displayed 1 or 0 only means
1= ON, 0=OFF, but the state of the pin is reversed.  But I can be wrong...
I newer used it so....

giig1967g, 2018-09-21:
@MaliMrav: I don't see any "Inverted option" in PCF8574 device settings.
Can you print your screenshot?

MaliMrav, 2018-09-21:
@giig1967g - The option is only there if you recompile after enabling the option in the _P019_PCF8574.ino file i.e. line 26: Device[deviceCount].InverseLogicOption = true; 

@sassod et al - The issue is that the inversed logic only works after a boot and before issuing a command to the port. After you issue a command to the port it reverts back to normal operation.

The reason the Inversed logic would be great here is that you want to issue a "1" to turn the relay on and a "0" to turn it off. but it doesn't seem to do it. This helps with MQTT, sending data to Domoticz and so on.
In fact if I turn on send data at boot, the switches go crazy in a loop of turning off and on (like nightrider)

TD-er, 2018-09-21:
So this is a new feature request, since it was never programmed to be used like this?

giig1967g, 2018-09-21:
@MaliMrav: Just enabling the option in the code doesn't activate any feature.

uzi18, 2018-09-21:
nice, not existing issue reached =)

MaliMrav, 2018-09-28:
> 
> 
> @MaliMrav: Just enabling the option in the code doesn't activate any feature.

@giig1967g - so I enabled the option and ticked it for each switch. What more do I need to do?
[image]

MaliMrav, 2018-09-28:
> 
> 
> So this is a new feature request, since it was never programmed to be used like this?
@TD-er - I'm not sure it's a feature at all. Utilising the PCF8574 driver (IO extender) has, it seems, inverts the states (it shows the state of the virtual port as "1" when it is off and "0" when it is on. this goes against all standards. It's workable but causes confusion and heartache when sending MQTT messages back to your home automation system (in my case, Domotisz)

meritumcom, 2018-09-30:
Inversed logic in my opinion it does not work. If I set inversed logic in esp, and in domoticz ON state as http://192.168.10.107/control?cmd=PCFGPIO,1,0 and OFF http://192.168.10.107/control?cmd=PCFGPIO,1,1 then I have
`61815: PCF : GPIO 1 Set to 0
61857: PCF : State 0
62140: PCF : GPIO 1 Set to 1
62157: PCF : State 1
62405: PCF : GPIO 1 Set to 0
62457: PCF : State 0
62708: PCF : GPIO 1 Set to 1
62757: PCF : State 1
63010: PCF : GPIO 1 Set to 0
63057: PCF : State 0
63316: PCF : GPIO 1 Set to 1
63357: PCF : State 1
63619: PCF : GPIO 1 Set to 0
63657: PCF : State 0
63921: PCF : GPIO 1 Set to 1
63957: PCF : State 1
64227: PCF : GPIO 1 Set to 0`
the effect is identical when the inversed logic option is not selected.
The difference only occurs when displaying the status of 1.0 while the type of signals sent from ESP to PCF8574 does not change

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er