letscontrolit/ESPEasy #1771: [Proposal] Splits rules in more file

workgroupengineering, 2018-09-21:
Splits rules in more file, for optimize l'utilizzo della cpu.

### Expected behavior
Exampe:
if raise System#Wake event , the parser will read file named System_Wake ecc ...

### Actual behavior
When raised event the rules parser read all file.

### System configuration

Hardware: All

ESP Easy version:  All

ESP Easy settings/screenshots:  none

TD-er, 2018-09-21:
We could also keep track of some of these event lines the first time a (changed) rule file is read.
For example event "bla" => file 1, line 23

giig1967g, 2018-09-21:
both solutions will dramatically increase the rule speed

uzi18, 2018-09-21:
more like: file number and byte number of line to seek faster

workgroupengineering, 2018-09-21:
Even I had thought of an equal solution, only that we would waste more memory, but associating the event with the file name there would be a direct access saving also memory.

Grovkillen, 2018-09-22:
The proposal suggested by @TD-er would work just as today but only add a dynamic start for parsing. That would probably be the first approach to test. Not having to add new error capturing functions. I.e. what to do when a file is referred to but not found. Etc. 

Not that I don't like the idea of using multiple files but let's leave that for the future.

TD-er, 2018-09-22:
Also I don't know how SPIFFS handles a lot of files and about characters used in filenames.

giig1967g, 2018-09-22:
And in case of rule:
on switch#state=0 do
Or
on rules#timer=1 do
Or 
On clock#time=all,%sunrise%

Your idea is nice, but maybe it''s more complex to read the rules file in several different files instead of a single file.
I like also the idea of Gijs even if it uses more memory.

For sure rules need to be speed up.

workgroupengineering, 2018-09-24:
@giig1967g 

there are two possibilities:

the first is to insert rules with the same root in the same file; example: switch#state = 0 and switch#state = 1 would be written in switch/state.txt.

The second, and to create files paths  based on the parameters,  examples: switch/state/0.txt or rules/timer/1.txt or clock/time/all/.sunrise.txt

uzi18, 2018-09-24:
my ideas are more generic:
conditional/unconditional rules execute like: 
on something execute file1.txt
or execute file2.txt

task specific:
task1.txt execute if file exists and task 1 is enabled

these ideas are very easy to code and will clean up rules.txt

closed 2018-12-03 by workgroupengineering